weewx/weewx #661: update debian (and redhat?) package installers for both rc and systemd

matthewwall, 2021-04-04:
need to make the debian rules understand both rc and systemd.

most immediately, need to make rules not break when there is no /etc/init.d/weewx file (see https://groups.google.com/g/weewx-user/c/RLCk9bO_wDg/m/YU1KwC-bCwAJ)

sandeen, 2023-08-23:
Hi Matthew! 

Just to chime in, was looking into weewx on a RHEL9 box, and used the el8 RPM. It did strike me as odd that it wasn't defaulting to the systemd service, so I manually switched to that, naming it "weewxd.service" because systemd had already tried to convert the rc init file to a service via a wrapper ...

I wonder if it might make sense to try to get weewx into Fedora / EPEL proper? I'd be happy to help with that, though I'm not an expert python packager. If you had any interest in lending a hand, though, getting that done might take that RPM package maintenance off your plate if that sounds good to you?

matthewwall, 2023-12-29:
as of weewx 5, the deb and rpm packages default to using a systemd unit.  they also include a systemd unit template, so the weewx-multi case is trivial.  the implementation happens in post, with a check for whether systemd is the init system.  this not only makes the packages safe for non-systemd variants of debian, but also ensures that the packages will not fail in a basic docker environment.

closed 2023-12-29 by matthewwall