weewx/weewx #676: weewx multi

18Scoobz, 2021-05-31:
Installed weewx 4.5.1 using setup.py
centos 7 minimal

tried to setup weewx multi with commands below:

sudo cp util/init.d/weewx-multi /etc/init.d/weewx
sudo update-rc.d weewx defaults
echo 'WEEWX_INSTANCES="vantage acurite"' | sudo tee /etc/default/weewx-multi
echo 'WEEWX_CFGDIR=/etc/weewx' | sudo tee -a /etc/default/weewx-multi

sudo update-rc.d weewx defaults
- gives this error sudo: update-rc.d: command not found

I used chkconfig weewx on that I used in another document in place of sudo update-rc.d weewx defaults 98

now running:
/etc/init.d/weewx status

I get the following:
/etc/init.d/weewx: line 58: /lib/init/vars.sh: No such file or directory

What do I need to do to get this working?

tkeffer, 2021-05-31:
The issues list is for reporting and tracking bugs. You should submit your query to [weewx-user](https://groups.google.com/forum/#!forum/weewx-user). That way, others can answer and learn.

But, offhand, I'd say the problem is that weewx-multi may not work on a Redhat-based system.

closed 2021-05-31 by tkeffer

18Scoobz, 2021-06-01:
How do I do that