weewx/weewx #678: PWSWeather Only Posting Once Per Day

techfixpros, 2021-06-15:
WeeWx 4.5.1 .deb
python --version
Python 3.5.3

Receiving the following error with debug = 1:
`weewx[13897] DEBUG weewx.restx: PWSWeather: Failed upload attempt 2: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL routines:SSL23_GET_SERVER_HELLO:unknown protocol>`

Per PWSWeather, the station is making 1 successful publish to that service once per day, around 3AM CST. All subsequent archive runs it returns the above error. I have regenerated the API key, and it is in quotes in the .conf.

gjr80, 2021-06-15:
Could you please provide:

- a larger log extract that shows the full error trace, preferably covering at least a few archive intervals
- a log extract covering the successful post, again covering a period of a few archive intervals
- a sanitised copy of the [StdRESTful] stanza from `weewx.conf`

Are you sure you are running WeeWX under python3? `python--version` only shows the version of python run when using the `python` command, this may or may not be the same version that WeeWX uses. You can confirm the python version used by WeeWX by looking at the WeeWX log during WeeWX startup, you will see something like this:

```
Jun 15 12:02:24 buster7 weewx[427] INFO __main__: Initializing weewx version 4.4.0
Jun 15 12:02:24 buster7 weewx[427] INFO __main__: Using Python 3.7.3 (default, Jan 22 2021, 20:04:44) #012[GCC 8.3.0]
Jun 15 12:02:24 buster7 weewx[427] INFO __main__: Platform Linux-4.19.0-16-amd64-x86_64-with-debian-10.9
Jun 15 12:02:24 buster7 weewx[427] INFO __main__: Locale is 'en_AU.UTF-8'
Jun 15 12:02:24 buster7 weewx[427] INFO __main__: PID file is /var/run/weewx.pid
Jun 15 12:02:24 buster7 weewx[366]: Starting weewx weather system: weewx.
```

techfixpros, 2021-06-16:
INFO __main__: Initializing weewx version 4.5.1
INFO __main__: Using Python 2.7.6 (default, Nov 13 2018, 12:45:42) #012[GCC 4.8.4]
INFO __main__: Platform Linux-4.4.0-148-generic-x86_64-with-LinuxMint-17.2-rafaela
INFO __main__: Locale is 'en_US.UTF-8'
INFO __main__: PID file is /var/run/weewx.pid
INFO __main__: Using configuration file /etc/weewx/weewx.conf
INFO __main__: Debug is 1

`[StdRESTful]

    [[StationRegistry]]
        # To register this weather station with weewx, set this to true
        register_this_station = true

    [[AWEKAS]]
        # This section is for configuring posts to AWEKAS.

        # If you wish to do this, set the option 'enable' to true,
        # and specify a username and password.
        # To guard against parsing errors, put the password in quotes.
        enable = true
        username = <not4u>
        password = <not4u>
        log_success = false

    [[CWOP]]
        # This section is for configuring posts to CWOP.

        # If you wish to do this, set the option 'enable' to true,
        # and specify the station ID (e.g., CW1234).
        enable = true
        station = <not4u>
        log_success = false

        # If this is an APRS (radio amateur) station, uncomment
        # the following and replace with a passcode (e.g., 12345).
        passcode = <not4u>

    [[PWSweather]]
        # This section is for configuring posts to PWSweather.com.

        # If you wish to do this, set the option 'enable' to true,
        # and specify a station and password.
        # To guard against parsing errors, put the password in quotes.
        enable = true
        station = <not4u>
        password = "<not4u>"
        log_failure = true
        log_success = true

    [[WOW]]
        # This section is for configuring posts to WOW.

        # If you wish to do this, set the option 'enable' to true,
        # and specify a station and password.
        # To guard against parsing errors, put the password in quotes.
        enable = true
        station = <not4u>
        password = <not4u>
        log_success = false

    [[Wunderground]]
        # This section is for configuring posts to the Weather Underground.

        # If you wish to do this, set the option 'enable' to true,
        # and specify a station (e.g., 'KORHOODR3') and password.
        # To guard against parsing errors, put the password in quotes.
        enable = true
        station = <not4u>
        password = <not4u>
[... 76 lines cut ...]
utines:SSL23_GET_SERVER_HELLO:unknown protocol>
Jun 16 17:00:25 john-PS583AA-ABA-a1020n weewx[29013] DEBUG weewx.restx: PWSWeather: Failed upload attempt 2: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL ro
utines:SSL23_GET_SERVER_HELLO:unknown protocol>
Jun 16 17:00:30 john-PS583AA-ABA-a1020n weewx[29013] DEBUG weewx.restx: PWSWeather: Failed upload attempt 3: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL ro
utines:SSL23_GET_SERVER_HELLO:unknown protocol>
Jun 16 17:00:30 john-PS583AA-ABA-a1020n weewx[29013] ERROR weewx.restx: PWSWeather: Failed to publish record 2021-06-16 17:00:00 CDT (1623880800): Failed upload after
3 tries
Jun 16 17:05:19 john-PS583AA-ABA-a1020n weewx[29013] DEBUG weewx.restx: PWSWeather: Failed upload attempt 1: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL ro
utines:SSL23_GET_SERVER_HELLO:unknown protocol>
Jun 16 17:05:24 john-PS583AA-ABA-a1020n weewx[29013] DEBUG weewx.restx: PWSWeather: Failed upload attempt 2: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL ro
utines:SSL23_GET_SERVER_HELLO:unknown protocol>
Jun 16 17:05:29 john-PS583AA-ABA-a1020n weewx[29013] DEBUG weewx.restx: PWSWeather: Failed upload attempt 3: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL ro
utines:SSL23_GET_SERVER_HELLO:unknown protocol>
Jun 16 17:05:29 john-PS583AA-ABA-a1020n weewx[29013] ERROR weewx.restx: PWSWeather: Failed to publish record 2021-06-16 17:05:00 CDT (1623881100): Failed upload after
3 tries
Jun 16 17:10:19 john-PS583AA-ABA-a1020n weewx[29013] DEBUG weewx.restx: PWSWeather: Failed upload attempt 1: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL ro
utines:SSL23_GET_SERVER_HELLO:unknown protocol>
Jun 16 17:10:24 john-PS583AA-ABA-a1020n weewx[29013] DEBUG weewx.restx: PWSWeather: Failed upload attempt 2: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL ro
utines:SSL23_GET_SERVER_HELLO:unknown protocol>
Jun 16 17:10:29 john-PS583AA-ABA-a1020n weewx[29013] DEBUG weewx.restx: PWSWeather: Failed upload attempt 3: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL ro
utines:SSL23_GET_SERVER_HELLO:unknown protocol>
Jun 16 17:10:29 john-PS583AA-ABA-a1020n weewx[29013] ERROR weewx.restx: PWSWeather: Failed to publish record 2021-06-16 17:10:00 CDT (1623881400): Failed upload after
3 tries`

Trying to capture a successful publish since it has not published at all today and my syslog dumps data several hours old.

techfixpros, 2021-06-17:
Captured two successful publishes today, 1:35a (second attempt) and 3:45a (third attempt):

`Jun 17 03:40:31 john-PS583AA-ABA-a1020n weewx[6940] DEBUG weewx.restx: PWSWeather: Failed upload attempt 3: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL rout
Jun 17 03:40:31 john-PS583AA-ABA-a1020n weewx[6940] ERROR weewx.restx: PWSWeather: Failed to publish record 2021-06-17 03:40:00 CDT (1623919200): Failed upload after 3
Jun 17 03:45:20 john-PS583AA-ABA-a1020n weewx[6940] DEBUG weewx.restx: PWSWeather: Failed upload attempt 1: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL rout
Jun 17 03:45:25 john-PS583AA-ABA-a1020n weewx[6940] INFO weewx.restx: PWSWeather: Published record 2021-06-17 03:45:00 CDT (1623919500)
Jun 17 03:50:19 john-PS583AA-ABA-a1020n weewx[6940] DEBUG weewx.restx: PWSWeather: Failed upload attempt 1: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL rout
Jun 17 03:50:24 john-PS583AA-ABA-a1020n weewx[6940] DEBUG weewx.restx: PWSWeather: Failed upload attempt 2: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL rout
Jun 17 03:50:29 john-PS583AA-ABA-a1020n weewx[6940] DEBUG weewx.restx: PWSWeather: Failed upload attempt 3: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL rout
Jun 17 03:50:29 john-PS583AA-ABA-a1020n weewx[6940] ERROR weewx.restx: PWSWeather: Failed to publish record 2021-06-17 03:50:00 CDT (1623919800): Failed upload after 3
`

and

`Jun 17 01:30:30 john-PS583AA-ABA-a1020n weewx[6940] DEBUG weewx.restx: PWSWeather: Failed upload attempt 3: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL rout
Jun 17 01:30:30 john-PS583AA-ABA-a1020n weewx[6940] ERROR weewx.restx: PWSWeather: Failed to publish record 2021-06-17 01:30:00 CDT (1623911400): Failed upload after 3
Jun 17 01:35:19 john-PS583AA-ABA-a1020n weewx[6940] DEBUG weewx.restx: PWSWeather: Failed upload attempt 1: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL rout
Jun 17 01:35:24 john-PS583AA-ABA-a1020n weewx[6940] DEBUG weewx.restx: PWSWeather: Failed upload attempt 2: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL rout
Jun 17 01:35:29 john-PS583AA-ABA-a1020n weewx[6940] INFO weewx.restx: PWSWeather: Published record 2021-06-17 01:35:00 CDT (1623911700)
Jun 17 01:40:19 john-PS583AA-ABA-a1020n weewx[6940] DEBUG weewx.restx: PWSWeather: Failed upload attempt 1: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL rout
Jun 17 01:40:24 john-PS583AA-ABA-a1020n weewx[6940] DEBUG weewx.restx: PWSWeather: Failed upload attempt 2: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL rout
Jun 17 01:40:29 john-PS583AA-ABA-a1020n weewx[6940] DEBUG weewx.restx: PWSWeather: Failed upload attempt 3: <urlopen error [Errno 1] _ssl.c:510: error:140770FC:SSL rout
Jun 17 01:40:29 john-PS583AA-ABA-a1020n weewx[6940] ERROR weewx.restx: PWSWeather: Failed to publish record 2021-06-17 01:40:00 CDT (1623912000): Failed upload after 3
`

tkeffer, 2021-06-17:
See if this helps.

https://stackoverflow.com/questions/34513784/urlopen-error-errno-1-ssl-c510-error14077417ssl

Try running a more modern version of Python.

techfixpros, 2021-06-18:
Thank you, updating python from 2.7.6 to 2.7.12 fixed it. I did try the python3 apt source, but am on 14.0.4.1 and was getting odd failures trying to force weewx to reinstall using the buster repos. Safer to use 2.7.12 backported.

closed 2021-06-18 by techfixpros