weewx/weewx #681: Wee_Device error with python-usb

nmarie276, 2021-06-19:
Hello.

I'm using WeeWX 4.5.1 Debian package on Raspbian Buster. Everything works fine with WeeWX, but I'm stuck with an error while using **wee_device**:
```
wee_device 
Using configuration file /etc/weewx/weewx.conf
Using FineOffsetUSB driver version 1.20 (weewx.drivers.fousb)
Traceback (most recent call last):
  File "/usr/share/weewx/weewx/drivers/fousb.py", line 1037, in openPort
    self.devh.claimInterface(self.usb_interface)
  File "/usr/lib/python3/dist-packages/usb/legacy.py", line 230, in claimInterface
    util.claim_interface(self.dev, interface)
  File "/usr/lib/python3/dist-packages/usb/util.py", line 205, in claim_interface
    device._ctx.managed_claim_interface(device, interface)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 811, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/share/weewx/wee_device", line 78, in <module>
    main()
  File "/usr/share/weewx/wee_device", line 74, in main
    device.configure(config_dict)
  File "/usr/share/weewx/weewx/drivers/__init__.py", line 67, in configure
    self.do_options(options, parser, config_dict, not options.noprompt)
  File "/usr/share/weewx/weewx/drivers/fousb.py", line 382, in do_options
    self.station = FineOffsetUSB(**config_dict[DRIVER_NAME])
  File "/usr/share/weewx/weewx/drivers/fousb.py", line 968, in __init__
    self.openPort()
  File "/usr/share/weewx/weewx/drivers/fousb.py", line 1041, in openPort
    raise weewx.WeeWxIOError(e)
weewx.WeeWxIOError: [Errno 16] Resource busy
```

In fact, this issue is quite old, but since I can upload my plots, I was quite lazy to open an issue until now. The history:
 * When I used Raspbian Jessie, the Python-USB package version was **0.4.3-1**: wee_device worked fine (so I displayed the station time, etc).
 * When I used Raspbian Stretch, the Python-USB package version was **1.0.0-1**: wee_device displayed the error mentionned above. My workaround was to use the package version from Jessie, which worked fine.
 * Now with Raspbian Buster, we switched to Python3-USB. The package version is **1.0.2-1**, and my workaround is not working.
 
 I suspect a link to the issue #178, which talk about a change in pyusb:
>all of the weewx drivers use pyusb 0.4. these should work fine on any system with pyusb 1.0 thanks to the pyusb legacy mode.

Again, this is a non-blocking issue. My use case is to check the station time and then force **--set-time** in case of clock drift.

Regards.

matthewwall, 2021-06-19:
this looks like a port contention issue.  when one process has an open connection to a usb port, no other process can open a connection.  so when `weewxd` has a connection open, doing `wee_device` will fail.

nmarie276, 2021-06-19:
@matthewwall you're right: just tried to disable the weewx daemon and I see a normal result:
```
wee_device 
Using configuration file /etc/weewx/weewx.conf
Using FineOffsetUSB driver version 1.20 (weewx.drivers.fousb)
Querying the station...
Fine Offset station settings:
                    local time: 2021.06.19 13:15:36 CEST
                  polling mode: PERIODIC

                  abs_pressure: 1001.9000000000001
                   current_pos: 28036
                  data_changed: 0
                    data_count: 3264
                     date_time: 2015-01-01 13:38
                 hum_in_offset: 14355
                hum_out_offset: 257
                            id: 8703
                 lux_wm2_coeff: 0
                       magic_1: 0x55
                       magic_2: 0xaa
                         model: 5890
                     rain_coef: 6150
                   read_period: 30
                  rel_pressure: 1013.4000000000001
                temp_in_offset: 5376
               temp_out_offset: 0
                      timezone: 0
                    unknown_01: 127
                    unknown_18: 0
                       version: 191
                     wind_coef: 4885
                     wind_mult: 0
```
But in this case, why I didn't need to disable weewx daemon with the previous python-usb library version? is this due to the "legacy mode" mentionned in the issue #178? Or is there another reason (for example, python-usb architecture change)?

closed 2021-06-19 by tkeffer