weewx/weewx #684: Display of evapotranspiration value in standard skin

michael-slx, 2021-07-20:
Hi,

since the ET (evapotranspiration) value is a rate and always quite low, the standard skin ("seasons") never displays a sensible value since the value is practically always rounded to 0.0.

Also looking at the skin's source files, I see there are some conditional checks that I don't really understand.

```html
#if $day.ET.has_data and $day.ET.sum.raw is not None and $day.ET.sum.raw > 0.0
      <tr>
        <td class="label">$obs.label.ET</td>
        <td class="data">$current.ET</td>
      </tr>
#end if
```

It seems to me as if the tag in the data column should read `$day.ET.sum` or the conditionals need to be removed or updated.

Either way, something should be changed in order to have a sensible ET value displayed in the default skin.

tkeffer, 2021-07-22:
Good point. 

Some weather stations (notably the Vantage series), emit ET only once an hour, on the hour. In that case `$current.ET` is typically high enough to show a number. So, perhaps what we want to display (and test for) is the hourly sum.

Let me think about this a bit. Right now, I'm traveling, so it might take a while.

michael-slx, 2021-07-28:
No worries, take your time.

I'd also like to add that some stations don't emit an ET value at all (e.g. the WeatherLink Live on it's LAN API). In that case, WeeWX will do the calculation and the value varies based on the `et_period` configuration option.

tkeffer, 2021-08-08:
Inclined to change the skin so it displays the total ET over the last hour. For example,

```
              #if $span(time_delta=3600).ET.has_data and $span(time_delta=3600).ET.sum.raw is not None and $span(time_delta=3600).ET.sum.raw > 0.0
              <tr>
                <td class="stats_label">$obs.label.ET</td>
                <td class="stats_data">$span(time_delta=3600).ET.sum</td>
              </tr>
              #end if
```

gjr80, 2021-08-09:
What about treating ET the same as we do with `rain`? ie, display 'Rain today' in the Current panel and in the High/Low panel display 'Rain today' and if a period other than 'Day' is selected (in the History panel) display the total rain over that period as well. Total ET over the last hour is a step in the right direction (we need to display a sum over a longer period rather than the 'sum' over a single archive interval) but I fear it will still be too short a period plus it will seem somewhat at odds with `rain` which will show 'whole' day values.

tkeffer, 2021-08-09:
I like that idea. I've always thought of ET as a kind of "anti-rain".

gjr80, 2021-08-09:
> I've always thought of ET as a kind of "anti-rain".

likewise.

tkeffer, 2021-08-11:
Question: what to do with the values under "Max/Min" in the _Seasons_ skin? Right now, the value displayed for rain is actually the total rain for the time period, not the max/min rain values seen.

For a heading of "Max/Min", that doesn't seem right. 

For an example, you can look at my website: https://www.threefools.org/weewx/

gjr80, 2021-08-12:
Presume you mean the 'High/Low panel'. Thinking of `rain` and then we can extrapolate to ET. To me keeping the intent of 'High/Low` would mean the logical display for periods greater than one day is the max daily rainfall. That leaves the question of what to display in the 'day' column? Maybe max hourly rain? The problem then is that you are displaying different things in the one row; maybe that is a problem, maybe not. Not sure I can see anything else that fits.

tkeffer, 2021-08-12:
I see three choices

1. Your suggestion. As you note, this means changing semantics across the columns. That has the potential to make things more confusing, not less.
2. Change the label to "Total Rain" (and, "Total ET"). Accept that this is inconsistent with the title 'Max/Min".
3. Add a new section, formatted similarly to the Max/Min panel, but title it "Totals".

I'm leaning towards option 3.

gjr80, 2021-08-12:
Agree re the three options. First thoughts were punt option one, useful data but the change in semantics kills it. I was leaning towards option two over option three but now that I have thought about it a bit more a 'Totals panel' makes sense. The modular nature of Seasons makes it dead easy to add and I don't think it will take up too much real estate.

gjr80, 2021-08-12:
OK, had a bit of a fiddle this arvo. Something like [this](https://weather.therodericks.id.au/weewx/index.html)?

A few points though:

- You will notice Evapotranspiration in full, using ET resulted in the label column being very narrow moving the data to the left and it just looked unbalanced, especially in conjunction with the High/Low panel (personally I prefer it in full). Perhaps someone with better HTML/CSS foo can fix that.
- Put the Totals panel above the High/Low panel so that today’s rainfall was as close as possible to the Current panel (from which today’s rain has been removed).
- Added ET plots, these would need to be commented out as not everyone has ET and we can’t selectively produce plots.
- Haven’t touched the Statistics page, didn’t see a need.
- Maybe a bit of re-ordering might be need with the plots, rather than tacking on ET as I did.

tkeffer, 2021-08-12:
Yes, that's exactly what I'm talking about. It looks good and solves the problem. Can you do a push with your changes?

I have no problem with using `Evapotranspiration` instead of `ET`. It will have to get translated to the different languages, but I can handle that once you're done.

gjr80, 2021-08-12:
I think that commit covers it. I noticed I was working from a pre-i18n version of Seasons, I don't believe I have undone any `$gettext`s but as you point out there will need to be some added. Will leave that with you now, late here.

tkeffer, 2021-09-08:
Can we close this one?

gjr80, 2021-09-08:
I'm happy to close. Implemented in development at commit 97ead1761870481f2671dbedbb3e5dfd736e4517.

michael-slx, 2021-09-08:
Skin change looks good, but I guess something should be changed on the calculations side of things too.

As it stands, ET is calculated by the StdCalculate service per archive record for the time period configured. This makes it difficult to aggregate the values (i.e. sum). I think it should calculate an "ET amount" per archive interval, just like rain works.

I understand this might be hard to implement though, as some devices provide ET in hardware

tkeffer, 2021-09-08:
> As it stands, ET is calculated by the StdCalculate service per archive record for the time period configured. This makes it difficult to aggregate the values (i.e. sum). I think it should calculate an "ET amount" per archive interval, just like rain works.

My apologies, but I'm not following.

michael-slx, 2021-09-08:
OK, forget my last post.

I just took a look at the source code of the [ET calculation](https://github.com/weewx/weewx/blob/644e794abd6b01fb458f757bf3c687694f48b56a/bin/weewx/wxxtypes.py#L142) in WeeWX and that cleared it up. I wrongly assumed that the ET was calculated for the `et_period` instead of the archive interval.

That would have made it difficult to display it correctly in the skin.

I guess we can close this one.

closed 2021-09-08 by michael-slx