weewx/weewx #685: Revision of light related fields and unit groups

gjr80, 2021-07-21:
The implementation and documentation of light related fields, units and unit groups is inconsistent.

Due in part to some historical implementation decisions we have the situation where the now default `wview_extended` schema includes field `luminosity` but no field `illuminance` yet the `obs_group_dict` supports type `illuminance` but not type `luminosity`. There was some off-line discussion on this via Slack #general back in January and February 2021 with a couple of ways ahead discussed and tentatively agreed but no implementation has occurred.

It's probably worth (at a minimum) reviewing the Slack discussion to see if the previously agreed solution is still acceptable.

This issue is related to some of the latter parts of issue #613 and [this weewx-user thread](https://groups.google.com/g/weewx-user/c/MaWgABMS0rE/m/YZfeAxvPDQAJ).

Nkawu, 2022-03-29:
Any movement on this? Just ran into this issue while using a VEML7700 Ambient light sensor. My solution in the meantime was to add this to `bin/user/extensions.py` (not sure if that's the correct location):
```python
import weewx.units
weewx.units.obs_group_dict['luminosity'] = 'group_illuminance'
```

mfraser, 2023-06-18:
Has there been any update on this as I've just come across this myself? Is it easier to change the schema than rename a field in the database?

closed 2024-06-04 by github-actions[bot]