weewx/weewx #686: Correct decode of Vantage LOOP2 windGust10

gjr80, 2021-08-01:
There appears to be a mistake in the Davis Serial Protocol Documentation in that the LOOP2 packet 10-min Wind Gust field is described as a 2 byte unsigned value in tenths of a mile per hour when in fact the data is presented in integer miles per hour. [This WXForum.net thread](https://www.wxforum.net/index.php?topic=42449.msg433430;topicseen#new) and links refer.

I have briefly perused the Vantage driver and I believe the driver interprets the LOOP2 packet 10-min Wind Gust field in tenths of a mile per hour. Needs further verification and a possible change to the driver.

tkeffer, 2021-08-07:
Is just 10 minute wind gust affected? How about the other wind fields? The documentation says,

> 10-Min Avg Wind Speed It is a two-byte unsigned value in 0.1mph resolution.
> 2-Min Avg Wind Speed It is a two-byte unsigned value in 0.1mph resolution.
> 10-Min Wind Gust It is a two-byte unsigned value in 0.1mph resoluation.
>

tkeffer, 2021-08-07:
Staring at the output of my VP2 for a while, it appears that just the 10-min wind gust is wrong. The others have sensible values. Does that match your experience?

gjr80, 2021-08-07:
I hadn't actually sat down to look at mine. The wind has picked up this morning so just switched to alternate between LOOP/LOOP2 and I saw this in my LOOP2:
```
LOOP:   2021-08-08 09:12:24 AEST (1628377944) altimeter: 1028.2083965933411, appTemp: 16.942858508455625, bar_calibration: -0.036, barometer: 1028.107445016697, bar_reduction: 0.002, cloudbase: 1128.9854001127387, dateTime: 1628377944, dayET: 0.011, dayRain: 0.0, dewpoint: 8.333333333333334, ET: None, heatindex: 16.666666666666668, hourRain: 0.0, humidex: 18.24509669646974, inDewpoint: 8.148606525574342, inHumidity: 55.0, inTemp: 17.27777777777778, maxSolarRad: 457.1672168823997, outHumidity: 56.0, outTemp: 17.555555555555557, outTempDay: 17.555555555555557, outTempNight: None, pressure: 1022.5332161237421, radiation: 552.0, rain: 0.0, rain15: 0.0, rain24: 0.0, rainRate: 0.0, stormRain: 0.0, THSW: 23.333333333333332, trendIcon: 20, usUnits: 16, UV: 0.0, wet_bulb: 12.88271234052837, windchill: 17.77777777777778, windDir: 196.0, windGust: 4.828032, windGust10: 0.7, windGustDir: 196.0, windGustDir10: 204.0, windrun: None, windSpeed: 1.609344, windSpeed10: 4.988966400000001, windSpeed2: 3.0
```
`windGust10` is definitely low and cannot be correct. `windSpeed10` is consistent with what is displayed on my VP console. I have a service running that (among other things) caches LOOP `windSpeed` readings and allows calculation of various aggregates over various periods. That service was giving me results that were consistent with LOOP2 `windSpeed2` and `windSpeed5`.

So to cut a long story short I think it is only `windGust10` that is incorrect/needs to be fixed.

tkeffer, 2021-08-07:
Fixed in commit https://github.com/weewx/weewx/commit/11b795ba457cd5603803438a21d5e2055e9f3937

closed 2021-08-07 by tkeffer