weewx/weewx #687: crash with bogus units for altitude

vinceskahan, 2021-08-09:
(from weewx-user) we have a recent [report](https://groups.google.com/g/weewx-user/c/DSuLT89IOC0) in the thread 'Problem with GW1000 driver' in which weewx crashes if a bogus unit (in this case 'feet' rather than the correct units of 'foot' is used for altitude, without much helpful information.

It would be helpful if weewx logs 'why' it is aborting so that the user knows what to fix.    Alternate solutions such as some kind of pre-check on weewx.conf for these kinds of syntax errors might be worth thinking about as well.

gjr80, 2021-08-09:
WeeWX reports such a config error just fine:
```
Aug 10 03:36:13 buster31 weewx[731]: Starting weewx weather system: weewx.
Aug 10 03:36:13 buster31 weewx[742] INFO __main__: Locale is 'en_AU.UTF-8'
Aug 10 03:36:13 buster31 weewx[742] INFO __main__: PID file is /var/run/weewx.pid
Aug 10 03:36:13 buster31 weewx[746] INFO __main__: Using configuration file /home/weewx/weewx.conf
Aug 10 03:36:13 buster31 weewx[746] INFO __main__: Debug is 1
Aug 10 03:36:13 buster31 weewx[746] DEBUG __main__: Initializing engine
Aug 10 03:36:13 buster31 weewx[746] INFO weewx.engine: Loading station type Simulator (weewx.drivers.simulator)
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.engine.StdTimeSynch
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.engine.StdTimeSynch
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.engine.StdConvert
Aug 10 03:36:13 buster31 weewx[746] INFO weewx.engine: StdConvert target unit is 0x1
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.engine.StdConvert
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.engine.StdCalibrate
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.engine.StdCalibrate
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.engine.StdQC
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.engine.StdQC
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.wxservices.StdWXCalculate
Aug 10 03:36:13 buster31 weewx[746] INFO weewx.wxservices: StdWXCalculate will use data binding wx_binding
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.manager: Daily summary version is 4.0
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.wxservices.StdWXCalculate
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdWXXTypes
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdWXXTypes
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdPressureCooker
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdPressureCooker
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdRainRater
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdRainRater
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdDelta
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdDelta
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.engine.StdArchive
Aug 10 03:36:13 buster31 weewx[746] INFO weewx.engine: Archive will use data binding wx_binding
Aug 10 03:36:13 buster31 weewx[746] INFO weewx.engine: Record generation will be attempted in 'hardware'
Aug 10 03:36:13 buster31 weewx[746] INFO weewx.engine: Using archive interval of 60 seconds (specified in weewx configuration)
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Use LOOP data in hi/low calculations: 1
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.engine.StdArchive
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.restx.StdStationRegistry
Aug 10 03:36:13 buster31 weewx[746] INFO weewx.restx: StationRegistry: Registration not requested.
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.restx.StdStationRegistry
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.restx.StdWunderground
Aug 10 03:36:13 buster31 weewx[746] INFO weewx.restx: Wunderground: Posting not enabled.
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.restx.StdWunderground
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.restx.StdPWSweather
Aug 10 03:36:13 buster31 weewx[746] INFO weewx.restx: PWSweather: Posting not enabled.
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.restx.StdPWSweather
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.restx.StdCWOP
Aug 10 03:36:13 buster31 weewx[746] INFO weewx.restx: CWOP: Posting not enabled.
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.restx.StdCWOP
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.restx.StdWOW
Aug 10 03:36:13 buster31 weewx[746] INFO weewx.restx: WOW: Posting not enabled.
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.restx.StdWOW
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.restx.StdAWEKAS
Aug 10 03:36:13 buster31 weewx[746] INFO weewx.restx: AWEKAS: Posting not enabled.
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.restx.StdAWEKAS
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.engine.StdPrint
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Finished loading service weewx.engine.StdPrint
Aug 10 03:36:13 buster31 weewx[746] DEBUG weewx.engine: Loading service weewx.engine.StdReport
[... cut ...]
Aug 10 03:36:16 buster31 weewx[746] CRITICAL __main__:     ****    File "/home/weewx/bin/weewx/units.py", line 1569, in convert
Aug 10 03:36:16 buster31 weewx[746] CRITICAL __main__:     ****      conversion_func = conversionDict[val_t[1]][target_unit]
Aug 10 03:36:16 buster31 weewx[746] CRITICAL __main__:     ****  KeyError: 'feet'
Aug 10 03:36:16 buster31 weewx[746] CRITICAL __main__:     ****  Exiting.
```

I suspect you will find the issue is the OP's system, for some reason it was not logging anything higher than INFO, it certainly was not logging what we would expect when `debug = 1`.

vinceskahan, 2021-08-09:
Agreed.   I'm good with canceling this one.

closed 2021-08-09 by vinceskahan