weewx/weewx #692: Vantage: wee_device --dump backfills timeout every 2 pages (10 records)

gashalot, 2021-08-22:
# Problem
Occasionally my station will suffer a multi-day power outage, and upon restart, does not properly backfill the `archive` with records on the Vantage console. To recover missing data I use the `wee_device --dump` command. In my setup, this backfill can take many minutes as the archive database (MySQL) is offsite across a high latency WAN connection.

When backfilling more than 2 pages of data, the command will consistently throw a `weewx.WeeWxIOError: Timeout in get_data_with_crc16` exception. This occurs when attempting to fetch the third page of "unseen" records. If you re-run the command, it will properly read through the recently inserted records, insert two more pages worth of records, then fail again.

I've tracked the issue down to the time it takes to handle a single page of data from the Vantage console. If handling of a full page of console data happens within a few seconds, the next page is fetched successfully. If handling a single page takes more than ~10 seconds, page fetches eventually fail. Because my archive fetches/reads can be slow, this happens consistently in my environment.

```Using configuration file /home/weewx/weewx.conf
Using Vantage driver version 3.2.2 (weewx.drivers.vantage)
Proceeding will dump all data in the logger.
Are you sure you want to proceed (y/n)? Starting dump ...
 Traceback (most recent call last):: 2021-08-13 23:50:00 PDT (1628923800)
  File "/home/weewx/./bin/wee_device", line 78, in <module>
    main()
  File "/home/weewx/./bin/wee_device", line 74, in main
    device.configure(config_dict)
  File "/home/weewx/bin/weewx/drivers/__init__.py", line 67, in configure
    self.do_options(options, parser, config_dict, not options.noprompt)
  File "/home/weewx/bin/weewx/drivers/vantage.py", line 2101, in do_options
    self.dump_logger(station, config_dict, options.noprompt)
  File "/home/weewx/bin/weewx/drivers/vantage.py", line 2703, in dump_logger
    for record in converted_generator:
  File "/home/weewx/bin/weewx/units.py", line 1682, in __next__
    _record = next(self.input_generator)
  File "/home/weewx/bin/weewx/drivers/vantage.py", line 697, in genArchiveDump
    _page = self.port.get_data_with_crc16(267, prompt=_ack, max_tries=self.max_tries)
  File "/home/weewx/bin/weewx/drivers/vantage.py", line 233, in get_data_with_crc16
    raise weewx.WeeWxIOError("Timeout in get_data_with_crc16")
weewx.WeeWxIOError: Timeout in get_data_with_crc16
```

*NOTE: The reason this backfill is necessary is a byproduct of another local issue that prevents weewx from properly backfilling data when restarted. I've explained it below.*

# Verification
After a failure (which occurs quickly when starting the backfill), you can re-start the command, and it will successfully fetch N+2 pages, insert the 10 new archive records, then timeout. You can repeat this process to fully backfill the archive. There is no limit to how many pages of already-inserted data can be read.

There are no documented timeouts in the Vantage serial console documentation.

# Solution
The fix appears to be simple: fetch the entire Vantage console database before attempting to insert into the archive. This is simple as changing L2703 in `vantage.py` to `for record in list(converted_generator):` . By wrapping the generator in a list, Python fetches the entire console (max of 2650 records),  I have tested locally, and the change worked: the entire console was dumped in a few seconds, then the inserts handled slowly later.

While the generator approach is elegant if the archive is large, the Vantage console can only store 2560 records, each of which is less than 100 bytes of data (hard limits already implemented in the driver), so the risk to memory exhaustion with this change is low.

https://github.com/weewx/weewx/blob/master/bin/weewx/drivers/vantage.py#L2703 

# Why does the backfill not work?
If your station operates without a battery-backed clock and weewx is restarted before you have properly resynchronized the clock, weewx's backfill procedures discards valid records that appear in the "future." When the Vantage generates its next archive record, weewx will store it correctly, leaving a gap in the archive.

Subsequent restarts of weewx with a properly set clock will not trigger a new backfill, as WeeWx looks for records after the last known record in the archive (specifically it issues a `DMPAFT` to the Vantage to fetch only records after the last archive record.)

Using `wee_device --dump` is the only way to backfill the records.

tkeffer, 2021-08-23:
Looks like you dug deep into the Vantage driver! That's good.

I'll have to experiment with my vantage console. Unfortunately, I'm away from it right now, so it may be a while.

I will say that you have a very unusual configuration that will lend itself to these kinds of timeout errors. I'm not keen on changing the code if it has side-effects for other users.

Thanks for the PR.

gashalot, 2021-09-05:
After some reflection, I've modified the PR to provide a new mode for the Vantage driver, allowing the current `--dump` to operate without modification.

tkeffer, 2021-09-24:
Addressed by PR #693.

closed 2021-09-24 by tkeffer