weewx/weewx #722: FR - Weewx foreground

Drealine, 2021-12-21:
Hi,

I make a weewx container wich included supervisord to keep service running when fail.
At the moment, I use the command /usr/share/weewx/weewxd /etc/weewx/conf/weewx.conf to start weewx as a deamon.

It is possible in the next release to implement a function to launch weewx in foreground ? Because when I use this command, if I execut "service weewx status", the status is down and I can't use wee_reports, etc ...

Thank you so much!

-Drealine

tkeffer, 2021-12-21:
Hello,

I am not following what you are asking. Can you be more specific about what you are trying to do? Are you using something other than systemd or initrc?

I will say that `weewxd` runs as a daemon only if you pass it a `--daemon` flag. This is normally done in the service file `weewx.service`.

Drealine, 2021-12-21:
The weewx.service is located in my container in /etc/init.d/weewx.
At the moment, if I try to keep weewx foreground, I need to use this command : 
`/usr/share/weewx/weewxd /etc/weewx/conf/weewx.conf`

For example, to start apache2 in foreground, I need to set like this : 

`command=/usr/sbin/apache2ctl -DFOREGROUND`

I would like to have the same option for weewx.service instead of weewxd which is not connected to weewx.service in /etc/init.d/

tkeffer, 2021-12-21:
Don't know what you mean by "container". Are you using something other than systemd or initrc? Are you using Docker?

Have you tried changing the run command in `weewx.service` so that the `--daemon` flag is not used? For example, change this:

    ExecStart=/home/weewx/bin/weewxd --daemon --pidfile=/var/run/weewx.pid /home/weewx/weewx.conf

to

    ExecStart=/home/weewx/bin/weewxd  --pidfile=/var/run/weewx.pid /home/weewx/weewx.conf

Drealine, 2021-12-21:
I'm using docker yes. 

The only file that I've is in /etc/init.d/weewx. I don't have weewx.service because I use the debian package so it's already created and I can't modify that. 

I already use `/usr/share/weewx/weewxd  /etc/weewx/conf/weewx.conf` but when I start weewx like this, when I want to make a "service weewx status", it show me that is not start.

The only thing that I see is to remove  `--daemon` on the weewx file in this line : 

`DAEMON_ARGS="--pidfile=$WEEWX_PID $WEEWX_CFG"`

tkeffer, 2021-12-21:
Using Docker: that was an important detail that you left out!

Another detail: `DAEMON_ARGS` is in the init.d file. Is that what you are using? Did you try removing `--daemon`? What happened? Were the results what you expected?

We don't support Docker, but if you can give me the specific change that you would need, I would be happy to look into implementing it. Please be specific: what file do you wish to be modified, and what should be its contents afterwards? Under what environment?

Drealine, 2021-12-21:
Yes at this line : https://github.com/weewx/weewx/blob/02e7dc99ce9519d430d06ecbe9d3f0ab05a40b26/util/init.d/weewx.debian#L46

If I remove --daemon, it work but weewx started multiple times, this is normal because the script a new pid. 

So the idea is to have a parameter on the /etc/init.d/weewx file wich keep the service in forgeground and not in background and can iteract with wee_reports, wee_utils etc and can do service weewx status which is not possible by using `/usr/share/weewx/weewxd /etc/weewx/conf/weewx.conf`

For example the best way to do that should be like this : 

`[supervisord]
nodaemon=true

[program:cron]
command=/usr/sbin/cron -f
autostart=true
autorestart=true
stdout_logfile=/var/log/cron.log
stderr_logfile=/var/log/cron.log
priority=2

[program:rsyslog]
command=/usr/sbin/rsyslogd -n -f /etc/rsyslog.conf
autostart=true
autorestart=true
priority=1

[program:weewx]
command=/usr/share/weewx/weewxd /etc/weewx/conf/weewx.conf
autorestart=true
redirect_stderr=true
stdout_logfile=/dev/stdout
stdout_logfile_maxbytes = 0
priority=3`

Drealine, 2021-12-21:
Okay, I found :) 

`python3 /usr/share/weewx/weewxd --pidfile=/var/run/weewx.pid /etc/weewx/weewx.conf`

But I would be great to implement a parameter to make weewx in foreground like say in the last message

matthewwall, 2021-12-21:
@Drealine i am confused.  you say you want to invoke weewxd directly, i.e., not as a daemon.  so just do that, as you said you have been doing:
```
weewxd /etc/weewx/weewx.conf
```
if you want to invoke weewx as a daemon, then use the rc script:
```
/etc/init.d/weewx start
```
you gave an example of using `apache2ctl`, but weewx is not like that.  we do not need (or want) to introduce another entire layer of complexity as they have done in apache.

Drealine, 2021-12-21:
> @Drealine i am confused. you say you want to invoke weewxd directly, i.e., not as a daemon. so just do that, as you said you have been doing:
>
> ```
> weewxd /etc/weewx/weewx.conf
> ```
>
> if you want to [...]

The idea is to start weewx in foreground and use weewx command directly on /bin/bash, not work if I start weewxd directly because the weewx.pid is not created.

Drealine, 2021-12-21:
Yes sorry ahah ...

matthewwall, 2021-12-21:
so what i hear you saying is:

- you want to use supervisord to start/stop weewxd
- you want weewxd to run in the foreground, not with the --daemon arg
- you want to be able to use 'service status weewx' to see the status of weewxd, whether or not you started weewxd using the init script

is that correct?  what else is there?

Drealine, 2021-12-21:
> so what i hear you saying is:
>
> - you want to use supervisord to start/stop weewxd
> - you want weewxd to run in the foreground, not with the --daemon arg
> - you want to be able to use 'service status [...]

Exactly!

At the moment, using this command on supervisord work : python3 /usr/share/weewx/weewxd --pidfile=/var/run/weewx.pid /etc/weewx/weewx.conf

But when service weewx stop told me that the service is not running but it's normal because this is not the script which created the pid.

matthewwall, 2021-12-21:
then you have a few options:

option a: make your supervisord configuration do everything that initrc (using 'service') expects.  you already figured out how to specify the pid.  now you need to delete the pid after weewxd exits.  when you do 'service status weewx', that simply uses the existence of the pid to determine whether weewxd is running.  the weewx init script is a bit smarter than that - it not only looks for the pid file, but it checks any running weewxd processes with that pid to be sure the pid file is not stale.

option b: modify your copy of /etc/init.d/weewx so that it contains whatever options you need to make supervisord work with it, without breaking 'service'.  this might be complicated, depending on how supervisord handles stdout/stderr, forking of child processes, etc.

and probably other options.

Drealine, 2021-12-21:
There is no way to do that because supervisord need to have a pid to check if the task is run or not. Using weewx file with remove --daemon, it's create multiple pid and weewx tasks so weewx say that is launch multiple times.
The only command which is work with service weewx status is this : 
`python3 /usr/share/weewx/weewxd --pidfile=/var/run/weewx.pid /etc/weewx/weewx.conf`

The user have to restart the container to restart weewx. This is also the idea of using docker so it's good to do that.

tkeffer, 2021-12-21:
I guess I still don't understand what changes / additions you need in WeeWX. If nothing, then let's close this issue.

Drealine, 2021-12-21:
It could be good to have a parameter that launches weewx in the foreground but it seems complex as explained before. We will stay at this solution, You have no future projets of creating a docker image for Weewx later?

tkeffer, 2021-12-21:
You see, I don't know what you mean by "have a parameter that launches weewx in the foreground." A parameter to what? 

No plans to support a Docker image, but many people have done this. For example, see https://github.com/vinceskahan/weewx-docker

Drealine, 2021-12-21:
To start weewx in foreground and interact directly witch service weewx status, service weewx reload.
It's complicated to explain and my english is not too good :/ 

Thank you for this example, I don't see that.

I close this issue.

closed 2021-12-21 by Drealine