weewx/weewx #724: timediff.py craps out when comparing a list and int

matthewwall, 2021-12-30:
latest development branch as of 30dec2021.  weewx-brultech driver as of 30dec2021.

weewx craps out with this exception:
```
Traceback (most recent call last):
  File "/opt/weewx/bin/weewxd", line 248, in <module>
    main()
  File "/opt/weewx/bin/weewxd", line 153, in main
    engine.run()
  File "/opt/weewx/bin/weewx/engine.py", line 210, in run
    self.dispatchEvent(weewx.Event(weewx.NEW_LOOP_PACKET, packet=packet))
  File "/opt/weewx/bin/weewx/engine.py", line 245, in dispatchEvent
    callback(event)
  File "/opt/weewx/bin/user/brultech.py", line 911, in new_loop_packet
    self.bt_extends.add_power_to_packet(event.packet)
  File "/opt/weewx/bin/user/brultech.py", line 781, in add_power_to_packet
    deriv = self.derivatives[obs_type].add_record(packet)
  File "/opt/weewx/bin/weeutil/timediff.py", line 44, in add_record
    and ['dateTime'] - self.old_timestamp <= self.stale_age:
TypeError: unsupported operand type(s) for -: 'list' and 'int'
```

weewx was invoked directly like this:
```
sudo /opt/weewx/bin/weewxd /etc/weewx/gem1.conf
```

here is the weewx config:
```
debug = 0
WEEWX_ROOT = /opt/weewx
log_success = True
log_failure = True
socket_timeout = 20
version = 4.6.0b7

[Station]
    location = "laurel"
    latitude = 0.00
    longitude = 0.00
    altitude = 30, foot
    station_type = Brultech
    #station_url = http://www.example.com
    rain_year_start = 1
    week_start = 6

[Brultech]
    driver = user.brultech
    stale = 1800
    poll_interval = 10
    max_channels = 32
    packet_type = GEMBin48NetTime
    connection = socket
    [[socket]]
        host = 192.168.11.51
        port = 8000
						    
[StdRESTful]
    [[StationRegistry]]
        register_this_station = false
    [[Influx]]
        log_success = False
	log_failure = False
        server_url = http://192.168.11.60/influx
        database = data
        measurement = power.gem1
        line_format = multi-line-dotted
        binding = loop, archive
        tags= host=data, device=gem1

[StdReport]
    SKIN_ROOT = skins
    HTML_ROOT = /var/www/html/weewx-gem1
    data_binding = bt_binding
    log_success = True
    log_failure = True
    
    [[SeasonsReport]]
        skin = Seasons
        enable = true

    [[Defaults]]
        lang = en
        unit_system = metric
        
        [[[Units]]]
            [[[[TimeFormats]]]]
                day        = %H:%M
                week       = %H:%M on %A
                month      = %Y.%m.%d %H:%M
                year       = %Y.%m.%d %H:%M
                rainyear   = %Y.%m.%d %H:%M
                current    = %Y.%m.%d %H:%M
                ephem_day  = %H:%M
                ephem_year = %Y.%m.%d %H:%M
        
        [[[Labels]]]
            [[[[Generic]]]]
                extraTemp1     = Temperature1
                extraTemp2     = Temperature2
                extraTemp3     = Temperature3
                extraTemp4     = Temperature4
                extraTemp5     = Temperature5
		extraHumid1    = Humidity1
                extraHumid2    = Humidity2
                extraHumid3    = Humidity3
                extraHumid4    = Humidity4
                extraHumid5    = Humidity5

[StdConvert]
    target_unit = US

[StdCalibrate]
    [[Corrections]]
        foo = foo + 0.2

[StdQC]
    [[MinMax]]
        barometer = 26, 32.5, inHg

[StdTimeSynch]
    clock_check = 14400
    max_drift = 5

[StdArchive]
    archive_interval = 300
    record_generation = hardware
    loop_hilo = True
    data_binding = bt_binding

[DataBindings]
    [[bt_binding]]
        database = archive_sqlite
        table_name = archive
        manager = weewx.manager.DaySummaryManager
        schema = schemas.wview_extended.schema

[Databases]
    [[archive_sqlite]]
        database_name = gem1.sdb
        database_type = SQLite

[DatabaseTypes]
    [[SQLite]]
        driver = weedb.sqlite
        SQLITE_ROOT = /var/lib/weewx

[Engine]
    [[Services]]
        prep_services = weewx.engine.StdTimeSynch
        data_services = user.brultech.BrultechService
        process_services = weewx.engine.StdConvert, weewx.engine.StdCalibrate, weewx.engine.StdQC
        xtype_services = weewx.wxxtypes.StdWXXTypes, weewx.wxxtypes.StdDelta
        archive_services = weewx.engine.StdArchive
        restful_services = weewx.restx.StdStationRegistry, user.influx.Influx
        report_services = weewx.engine.StdPrint, weewx.engine.StdReport
```

looks like i got one loop packet then the death happened on the next one?
```
LOOP:   2021-12-30 13:03:46 EST (1640887426) ch10_a_energy2: 2327584, ch10_amp: 0.24, ch10_a_power: None, ch10_p_energy2: 0, ch10_p_power: None, ch11_a_energy2: 26592, ch11_amp: 0.0, ch11_a_power: None, ch11_p_energy2: 0, ch11_p_power: None, ch12_a_energy2: 116825, ch12_amp: 0.0, ch12_a_power: None, ch12_p_energy2: 0, ch12_p_power: None, ch13_a_energy2: 24869, ch13_amp: 0.0, ch13_a_power: None, ch13_p_energy2: 0, ch13_p_power: None, ch14_a_energy2: 5234, ch14_amp: 0.0, ch14_a_power: None, ch14_p_energy2: 0, ch14_p_power: None, ch15_a_energy2: 26199, ch15_amp: 0.0, ch15_a_power: None, ch15_p_energy2: 0, ch15_p_power: None, ch16_a_energy2: 1519, ch16_amp: 0.0, ch16_a_power: None, ch16_p_energy2: 0, ch16_p_power: None, ch17_a_energy2: 18157, ch17_amp: 0.0, ch17_a_power: None, ch17_p_energy2: 0, ch17_p_power: None, ch18_a_energy2: 476, ch18_amp: 0.0, ch18_a_power: None, ch18_p_energy2: 0, ch18_p_power: None, ch19_a_energy2: 32787, ch19_amp: 0.0, ch19_a_power: None, ch19_p_energy2: 0, ch19_p_power: None, ch1_a_energy2: 249360, ch1_amp: 0.0, ch1_a_power: None, ch1_count: 0, ch1_p_energy2: 0, ch1_p_power: None, ch1_temperature: 32.0, ch1_volt: 133.9, ch20_a_energy2: 5831, ch20_amp: 0.0, ch20_a_power: None, ch20_p_energy2: 0, ch20_p_power: None, ch21_a_energy2: 25800, ch21_amp: 0.0, ch21_a_power: None, ch21_p_energy2: 0, ch21_p_power: None, ch22_a_energy2: 757, ch22_amp: 0.0, ch22_a_power: None, ch22_p_energy2: 0, ch22_p_power: None, ch23_a_energy2: 26574, ch23_amp: 0.0, ch23_a_power: None, ch23_p_energy2: 0, ch23_p_power: None, ch24_a_energy2: 1573, ch24_amp: 0.0, ch24_a_power: None, ch24_p_energy2: 0, ch24_p_power: None, ch25_a_energy2: 24792, ch25_amp: 0.0, ch25_a_power: None, ch25_p_energy2: 0, ch25_p_power: None, ch26_a_energy2: 767, ch26_amp: 0.0, ch26_a_power: None, ch26_p_energy2: 0, ch26_p_power: None, ch27_a_energy2: 26884, ch27_amp: 0.0, ch27_a_power: None, ch27_p_energy2: 0, ch27_p_power: None, ch28_a_energy2: 2351, ch28_amp: 0.0, ch28_a_power: None, ch28_p_energy2: 0, ch28_p_power: None, ch29_a_energy2: 14154, ch29_amp: 0.0, ch29_a_power: None, ch29_p_energy2: 0, ch29_p_power: None, ch2_a_energy2: 9136, ch2_amp: 0.0, ch2_a_power: None, ch2_count: 0, ch2_p_energy2: 0, ch2_p_power: None, ch2_temperature: 32.0, ch30_a_energy2: 748, ch30_amp: 0.0, ch30_a_power: None, ch30_p_energy2: 0, ch30_p_power: None, ch31_a_energy2: 17018, ch31_amp: 0.0, ch31_a_power: None, ch31_p_energy2: 0, ch31_p_power: None, ch32_a_energy2: 12379696590, ch32_amp: 0.28, ch32_a_power: None, ch32_p_energy2: 0, ch32_p_power: None, ch3_a_energy2: 13015, ch3_amp: 0.0, ch3_a_power: None, ch3_count: 0, ch3_p_energy2: 0, ch3_p_power: None, ch3_temperature: 32.0, ch4_a_energy2: 1522, ch4_amp: 0.0, ch4_a_power: None, ch4_count: 0, ch4_p_energy2: 0, ch4_p_power: None, ch4_temperature: 32.0, ch5_a_energy2: 24631, ch5_amp: 0.0, ch5_a_power: None, ch5_p_energy2: 0, ch5_p_power: None, ch5_temperature: 32.0, ch6_a_energy2: 1646415, ch6_amp: 0.0, ch6_a_power: None, ch6_p_energy2: 0, ch6_p_power: None, ch6_temperature: 32.0, ch7_a_energy2: 141327, ch7_amp: 0.08, ch7_a_power: None, ch7_p_energy2: 0, ch7_p_power: None, ch7_temperature: 32.0, ch8_a_energy2: 6779, ch8_amp: 0.08, ch8_a_power: None, ch8_p_energy2: 0, ch8_p_power: None, ch8_temperature: 32.0, ch9_a_energy2: 66325, ch9_amp: 0.0, ch9_a_power: None, ch9_p_energy2: 0, ch9_p_power: None, dateTime: 1640887426, secs: 9544826, serial: YYYYYYYY, ser_no: XXXXX, time_created: 1640887426, unit_id: 11, usUnits: 1
```

matthewwall, 2021-12-30:
i guess this might be a problem with weewx-brultech, not timediff.py?

matthewwall, 2021-12-30:
btw, we probably should fix the sorter used to sort the observations in a packet.  the current one puts 'ch10' ahead of 'ch1'

gjr80, 2021-12-30:
> btw, we probably should fix the sorter used to sort the observations in a packet.  the current one puts 'ch10' ahead of 'ch1'

I ran into the same issue with gw1000 drivers; lots of 'chx' and 'chxy'. I implemented a natural sort (rather than strict ascii) sort in the driver and it seems to work fine. Will pull in the code later this morning.

tkeffer, 2021-12-30:
Stupid typo. Fixed in https://github.com/weewx/weewx/commit/414794aa3f46f2830a51eefbddda111a1d6126fc

closed 2021-12-30 by tkeffer

tkeffer, 2021-12-30:
As for collation of `ch1` vs `ch10`, it would probably be easiest to just add a leading zero, so `ch1` becomes `ch01`. I can do that but, unfortunately, it would break old databases.

gjr80, 2022-01-08:
> I ran into the same issue with gw1000 drivers; lots of 'chx' and 'chxy'. I implemented a natural sort (rather than strict ascii) sort in the driver and it seems to work fine. Will pull in the code [...]

'Natural' sorting of fields in string representations of packets/records implemented vide PR #728.