weewx/weewx #730: Evapotranspiration is set for grass only

lordratner, 2022-01-20:
https://github.com/weewx/weewx/blob/abb8320465646a67f4233edd580b707f85869e96/bin/weewx/wxformulas.py#L835-L838

These variables are locked in the code. At least albedo could be referenced from the weewx.conf file to allow for different usages.

tkeffer, 2022-01-20:
Good point. Is this something you are using?

lordratner, 2022-01-21:
Yeah, I'm going to attempt to use WeeWX ET calculation for modifying the watering routine for a fruit tree orchard.

tkeffer, 2022-07-30:
You can now specify something like
```ini
[StdWXCalculate]
  [[WXXTypes]]
    [[[ET]]]
      et_period = 3600
      wind_height = 2.0
      albedo = 0.23
      cn = 37
      cd = 0.34
```
See the User's Guide for details.

Closed.

closed 2022-07-30 by tkeffer