weewx/weewx #731: performance of per-sensor last-contact feature in 4.6b Seasons skin

matthewwall, 2022-01-21:
here are the timings for various configurations.  

this is on an rpi2 with 1G memory.  2 weewx instances: SDR and SDS011.  SDR uses wview_extended schema, SDS011 uses schema with only pm2_5 and pm10_0 columns (i started running this before the wview_extended schema was available).  the only skin is Seasons, and it is the latest from the development branch as of 20jan2021, plus the '$recent' changes to control the timespan used to determine whether to show an observation, instead of just '$year'.
```
#set $recent=$year
8 files in 6-10 seconds

#set $recent=$span($day_delta=30)
8 files in 55-60 seconds

#set $recent=$span($day_delta=120)
8 files in 180-190 seconds
```
the '$recent' pattern appears in these files:
* hilo.inc (1)
* index.html.tmpl (1)
* sensors.inc (9)
* statistics.inc (1)
the number in parentheses indicates the number of times that the `getattr($recent, $x)` pattern appears in each file.

note that this was run on 20-21 jan 2021, so the `$year` contains about 20 days of data.  so i guess the (significant) difference in timings between `$recent=$year` and `$recent=$span($day_delta=30)` is due to having yearly tables in the database?

the biggest slowdown comes from sensors.inc, probably because the getattr is invoked so many times (more specifically, because of the has_data and `lasttime.raw` that is invoked on the return value from `getattr`).

##  cheetah quirks:

the following two statements are apparently not equivalent:
```
#if getattr($year, '%s.has_data' % $x)

#if getattr($year, $x).has_data
```
the first seems to evaluate to true always.  the second behaves as expected, i.e., the same as this expression:
```
#if $getVar('year.%s.has_data' % $x)
```

## cheetah quirks noted in slack discussion jan2021:

this is a syntax error:
```
#set $data = getattr($recent, $x . '.lasttime.raw')
```
but this returns an `ObservationBinder`:
```
#set $data = getattr($recent, '%s.lasttime.raw' % $x)
```
but this returns a float:
```
$getVar('year.%s.lasttime.raw' % $x)
```

tkeffer, 2022-01-21:
I'm surprised it's so slow.

The `has_data` suffix tests whether there is a non-null value in the time range. There's an optimization to use the daily summaries if the time range falls on midnight-midnight, or midnight-to-now, boundaries. Perhaps we are not hitting that optimization?

Look for an instrumented version of xtypes.py.

matthewwall, 2022-01-22:
instrumentation showed that the slowness was due to the span operating on a boundary matching the time of the query, so summaries were not used.

at commit a006505 @tkeffer added an option to force the span calculation to a 'midnight' boundary.  using that resulted in significantly faster performance - the timings day_delta=30 are similar to those for year - about 6-12 seconds.  when day_delta=120 the times increase to 10-12 seconds.  when day_delta=300 the times stay around 11-12 seconds.

this is the comparison between 4.5.1 and 4.6:
```
this is typical with Seasons from the latest 4.6:

Jan 21 21:05:26 7up weewx-sdr[26970] INFO weewx.cheetahgenerator: Generated 8 files for report Seasons in 6.29 seconds
Jan 21 21:05:29 7up weewx-sdr[26970] INFO weewx.imagegenerator: Generated 7 images for report Seasons in 3.12 seconds
Jan 21 21:05:29 7up weewx-sdr[26970] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx-sdr
...
Jan 21 21:25:41 7up weewx-sdr[26970] INFO weewx.cheetahgenerator: Generated 8 files for report Seasons in 6.01 seconds
Jan 21 21:25:45 7up weewx-sdr[26970] INFO weewx.imagegenerator: Generated 7 images for report Seasons in 3.34 seconds
Jan 21 21:25:45 7up weewx-sdr[26970] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx-sdr

this is typical with Seasons from 4.5.1:

Jan 21 21:40:36 7up weewx-sdr[27457] INFO weewx.cheetahgenerator: Generated 8 files for report Seasons in 4.48 seconds
Jan 21 21:40:39 7up weewx-sdr[27457] INFO weewx.imagegenerator: Generated 15 images for report Seasons in 2.57 seconds
Jan 21 21:40:39 7up weewx-sdr[27457] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx-sdr-45
...
Jan 21 22:10:35 7up weewx-sdr[27457] INFO weewx.cheetahgenerator: Generated 8 files for report Seasons in 4.42 seconds
Jan 21 22:10:38 7up weewx-sdr[27457] INFO weewx.imagegenerator: Generated 15 images for report Seasons in 2.58 seconds
Jan 21 22:10:38 7up weewx-sdr[27457] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx-sdr-45
```

matthewwall, 2022-01-22:
after changes to rss template, imagegenerator (xtypes exception handling), and the 'midnight' option, these are the timings:

```
# SDR with 5 T/H sensors and wview_extended schema
Jan 22 16:00:34 7up weewx-sdr[30508] INFO weewx.cheetahgenerator: Generated 8 files for report Seasons in 6.31 seconds
Jan 22 16:00:42 7up weewx-sdr[30508] INFO weewx.imagegenerator: Generated 14 images for report Seasons in 8.16 seconds
Jan 22 16:00:42 7up weewx-sdr[30508] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx-sdr
```

```
# SDS011 with custom schema - pm only
Jan 22 16:00:24 7up weewx-sds011[30367] INFO weewx.cheetahgenerator: Generated 8 files for report Seasons in 1.01 seconds
Jan 22 16:00:27 7up weewx-sds011[30367] INFO weewx.imagegenerator: Generated 2 images for report Seasons in 3.15 seconds
Jan 22 16:00:27 7up weewx-sds011[30367] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx-sds011
```

matthewwall, 2022-01-22:
these are timings from an rpi called `pitest` that is running SDR with 12 T/H sensors and a weather station, and Brultech with one gem and 32 channels.  both are using Seasons skin.

```
Jan 22 16:20:33 pitest weewx-sdr[17907] INFO weewx.cheetahgenerator: Generated 8 files for report SeasonsReport in 11.10 seconds
Jan 22 16:20:36 pitest weewx-sdr[17907] INFO weewx.imagegenerator: Generated 14 images for report SeasonsReport in 3.44 seconds
Jan 22 16:20:37 pitest weewx-sdr[17907] INFO weewx.reportengine: Copied 5 files to /var/www/html/weewx-sdr
Jan 22 16:25:29 pitest weewx-sdr[17907] INFO weewx.cheetahgenerator: Generated 8 files for report SeasonsReport in 6.66 seconds
Jan 22 16:25:33 pitest weewx-sdr[17907] INFO weewx.imagegenerator: Generated 14 images for report SeasonsReport in 3.39 seconds
```

```
# GEM
Jan 22 16:20:26 pitest weewx-gem[17962] INFO weewx.cheetahgenerator: Generated 8 files for report SeasonsReport in 4.68 seconds
Jan 22 16:20:28 pitest weewx-gem[17962] INFO weewx.imagegenerator: Generated 0 images for report SeasonsReport in 1.75 seconds
Jan 22 16:20:28 pitest weewx-gem[17962] INFO weewx.reportengine: Copied 5 files to /var/www/html/weewx-gem
Jan 22 16:25:22 pitest weewx-gem[17962] INFO weewx.cheetahgenerator: Generated 8 files for report SeasonsReport in 0.65 seconds
Jan 22 16:25:24 pitest weewx-gem[17962] INFO weewx.imagegenerator: Generated 0 images for report SeasonsReport in 1.85 seconds
```

tkeffer, 2022-01-22:
I wonder why the GEM version is so much faster at generating files?

matthewwall, 2022-01-23:
i think it is because the observations are not being processed.  for some reason the skin is not finding the `*_p_power` fields in each `$current`.

matthewwall, 2022-01-23:
here are three examples of the Seasons skin being used with different drivers and numbers of sensors, and 'non-standard' schema.  these have no changes to the Seasons skin.  the skin is customized only by using parameters in DisplayOptions.

rpi2 with SDS011 driver, minimal schema
[image]

rpi2 with SDR driver and 12 T/H sensors
[image]

rpi2 with SDR driver and 6 sensors
[image]

tkeffer, 2022-01-23:
That's impressive.

We really need a section about `[DisplayOptions]` in the Customizing Guide.

tkeffer, 2022-05-06:
Was this issue just an FYI, or is there something we should be working on? Wondering if we can close it.

matthewwall, 2022-05-06:
i raised this issue as a place to document and comment on any performance issues in the parameterization of the Seasons skin.

closed 2022-05-06 by matthewwall