weewx/weewx #736: missing translations

matthewwall, 2022-01-25:
here is the output from `i18n-report` for each of the skins in the development branch as of 25jan2022.  looks like there are a few missing items.

```
% for x in skins/Ftp skins/Mobile skins/Rsync skins/Seasons skins/Smartphone skins/Standard; do ./util/i18n/i18n-report --skin $x; done
```
```
i18n translation report for skin: skins/Ftp
No language directory found at skins/Ftp/lang
i18n translation report for skin: skins/Mobile
skins/Mobile/lang/de.conf
  found only in skins/Mobile/lang/en.conf:
  found only in skins/Mobile/lang/de.conf:
skins/Mobile/lang/nl.conf
  found only in skins/Mobile/lang/en.conf:
  found only in skins/Mobile/lang/nl.conf:
listed in en.conf but not found in the skin:
  at
  Language
not listed in en.conf:
  Rain (hourly total)
```

```
i18n translation report for skin: skins/Rsync
No language directory found at skins/Rsync/lang
i18n translation report for skin: skins/Seasons
skins/Seasons/lang/cz.conf
  found only in skins/Seasons/lang/en.conf:
  found only in skins/Seasons/lang/cz.conf:
skins/Seasons/lang/de.conf
  found only in skins/Seasons/lang/en.conf:
  found only in skins/Seasons/lang/de.conf:
skins/Seasons/lang/es.conf
  found only in skins/Seasons/lang/en.conf:
  found only in skins/Seasons/lang/es.conf:
skins/Seasons/lang/fr.conf
  found only in skins/Seasons/lang/en.conf:
  found only in skins/Seasons/lang/fr.conf:
skins/Seasons/lang/gr.conf
  found only in skins/Seasons/lang/en.conf:
    Units.Labels.meter=[' meter', ' meters']
  found only in skins/Seasons/lang/gr.conf:
skins/Seasons/lang/it.conf
  found only in skins/Seasons/lang/en.conf:
  found only in skins/Seasons/lang/it.conf:
skins/Seasons/lang/nl.conf
  found only in skins/Seasons/lang/en.conf:
  found only in skins/Seasons/lang/nl.conf:
skins/Seasons/lang/th.conf
  found only in skins/Seasons/lang/en.conf:
    Units.Labels.meter=[' meter', ' meters']
  found only in skins/Seasons/lang/th.conf:
listed in en.conf but not found in the skin:
  Monthly summary as of
  Daily summary as of
  Yearly summary as of
  Language
not listed in en.conf:
  Max
  Skin
  Min
```

```
i18n translation report for skin: skins/Smartphone
skins/Smartphone/lang/de.conf
  found only in skins/Smartphone/lang/en.conf:
  found only in skins/Smartphone/lang/de.conf:
skins/Smartphone/lang/nl.conf
  found only in skins/Smartphone/lang/en.conf:
  found only in skins/Smartphone/lang/nl.conf:
listed in en.conf but not found in the skin:
  24h
  at
  7-day
  Language
not listed in en.conf:
  Rain (hourly total)
  Rain (daily total)
```

```
i18n translation report for skin: skins/Standard
skins/Standard/lang/de.conf
  found only in skins/Standard/lang/en.conf:
  found only in skins/Standard/lang/de.conf:
skins/Standard/lang/fr.conf
  found only in skins/Standard/lang/en.conf:
  found only in skins/Standard/lang/fr.conf:
skins/Standard/lang/nl.conf
  found only in skins/Standard/lang/en.conf:
  found only in skins/Standard/lang/nl.conf:
listed in en.conf but not found in the skin:
  Today
  Language
  Mobile formatted
  Always down
  Always up
not listed in en.conf:
  High
  max
```

matthewwall, 2022-01-25:
simplified to report only when there are exceptions:

```
for x in skins/Ftp skins/Mobile skins/Rsync skins/Seasons skins/Smartphone skins/Standard; do ./util/i18n/i18n-report --skin $x; done
```

```
i18n translation report for skin: skins/Ftp
No language directory found at skins/Ftp/lang
i18n translation report for skin: skins/Mobile
listed in en.conf but not found in the skin:
  at
  Language
not listed in en.conf:
  Rain (hourly total)
```

```
i18n translation report for skin: skins/Rsync
No language directory found at skins/Rsync/lang
i18n translation report for skin: skins/Seasons
skins/Seasons/lang/gr.conf
  found only in skins/Seasons/lang/en.conf:
    Units.Labels.meter=[' meter', ' meters']
skins/Seasons/lang/th.conf
  found only in skins/Seasons/lang/en.conf:
    Units.Labels.meter=[' meter', ' meters']
listed in en.conf but not found in the skin:
  Yearly summary as of
  Monthly summary as of
  Daily summary as of
  Language
not listed in en.conf:
  Max
  Min
  Skin
```

```
i18n translation report for skin: skins/Smartphone
listed in en.conf but not found in the skin:
  at
  7-day
  Language
  24h
not listed in en.conf:
  Rain (hourly total)
  Rain (daily total)
```

```
i18n translation report for skin: skins/Standard
listed in en.conf but not found in the skin:
  Mobile formatted
  Language
  Today
  Always up
  Always down
not listed in en.conf:
  max
  High
```

tkeffer, 2022-01-30:
```
strings in en.conf but not found in the skin:
  Monthly summary as of
  Language
  Yearly summary as of
  Daily summary as of
```
Lines 1, 3, and 4 are from parameterization efforts in `rss.xml.tmpl`. Line 2 is expected to be in all locale files. 

I think we're OK, at least with *Seasons*.

closed 2022-02-04 by matthewwall