weewx/weewx #737: `windrun` always zero

tkeffer, 2022-02-06:
A user [reported](https://groups.google.com/g/weewx-user/c/uPp6eEA83Gk) that `windrun` is always zero.  He has specified `prefer_hardware` for windrun, but StdWXCalculate always return zero for the calculated value.

LOOP packets all have `windrun` set to None (as they should), so no values are added to the `windrun` accumulator. When it comes time to calculate `windrun` for the archive record, the `sum` extractor is used. Because no values have been added to the `windrun` accumulator, the `sum` extractor pulls out zero.

StdWXCalculate sees the zero value for `windrun`, and assumes all is well, so it doesn’t bother calculating it. The archive record retains the zero value.

The workaround is to use `software` instead of `prefer_hardware`, but that’s just a band-aid.

The obvious solution is that the `sum` extractor should not just blindly return `sum` but, instead, should first check the `count`. If it’s zero, return `None`, otherwise return `sum`.

However, I’m worried about side effects. Can anyone think of any calculation that uses the `sum` extractor, but which depends on the returned value being zero, even if no data has contributed to it? I think something like `rain` will be OK, because `rain` is nearly always present in LOOP packets, and nearly always is non-None. I suppose one could imagine a weather station that doesn’t bother putting rain in a LOOP packet unless some rain fell, but in that case, extracting None, rather than zero, is probably the right thing to do.

c0by85, 2022-02-06:
Hello, thanks for opening an issue for this problem / bug. The same problem exists with ET after updating to 4.6.0.

When ET is used with `prefer_hardware` a 0 or 0.0 is written to the DB, with `software` this does not happen.

tkeffer, 2022-02-06:
That's good to know. Thanks for posting! 

If you're comfortable with git, there is a fix in the [`new_sum` branch](https://github.com/weewx/weewx/tree/new_sum).

c0by85, 2022-02-06:
Unfortunately not, I wait for the official version and update via apt.

tkeffer, 2022-02-09:
Fixed in commit https://github.com/weewx/weewx/commit/ed8b9f9223590f7dc1bf7b5d76436f839730967b

closed 2022-02-09 by tkeffer