weewx/weewx #740: Copy the weewx documentation to HTML_ROOT, via a skin.

glennmckechnie, 2022-02-08:
I usually copy the docs directory to HTML_ROOT manually but it occurs to me that it could be set up via an optional skin which would have the benefit of automating the process and hopefully getting the FineM out there.

It's added using the **copy_once** skin feature of the (attached) skin and using the reports **enable=true/false** switch then it's definitely configurable by the user.  Enabling it means that an accessible local copy exists (no digging around in the file system) of the docs that match the current installation.
Adding a link to the (Seasons) skin puts it just a click away, and giving that link the ability to be hidden, using the logic of  **$show_rss** and **$show_reports** found in 4.6.0's _titlebar.inc_ gives another level of configuration, for the RTFM hesitant, or obsessed.

The downside is that rsync could pick it up and transfer it to an offsite host, but is that a downside? Especially for what should be essential reading!

I'll attach a minimal skin that goes through the process. It can be installed via **wee_extension** and other than manually replacing _Seasons/titlebar.inc_ and _Seasons/skin.conf_ with the provided substitutes (_Seasons/titlebar-docs.inc_ and _Seasons/skin-docs.conf_) then it's right to go.

[weewx_documentation_concept.tar.gz](https://github.com/weewx/weewx/files/8023417/weewx_documentation_concept.tar.gz)

The DOC (Documentation) link will end up on the titlebar. The intention was to put it alongside the RSS one but my css-fu is not up to par so some imagination is required.

BTW, this wee_extension skin example is only for the concept. I'm thinking that this configuration would come pre-installed with weewx. The docs are already in the installation, they can just live in a friendlier location; with the option of moving to a prime location.

tkeffer, 2022-02-08:
One problem with this is that it either requires duplicating the entire documentation under `SKIN_ROOT`, or some serious hacking of `skin.conf` by the installer. I'm not keen on either. 

Including documentation under `/usr/share/doc` is expected, at least under Debian.

matthewwall, 2022-02-08:
glenn is right about making the docs easier to access.  neither /usr/share/weewx/docs or /home/weewx/docs are accessible to a web server by default.  on the other hand, skipping the server and using file:/// works perfectly well.

one option might be to create a 'docs' skin - a standalone skin whose sole purpose is to copy the weewx docs from the source/install tree into the HTML_ROOT area (or wherever).  that would then be accessible via a simple link to *every* skin, not just seasons.

i kinda cringe at stuff the docs into the seasons skin (or any other skin, for that matter)

tkeffer, 2022-02-08:
How would a "docs" skin work without maintaining a copy of the docs in it? With symlinks? If so, the symlinks would be dependent on the installation method. 

It just seems like a lot of complexity for marginal utility. Personally, I use either the `file:///` approach, or just browse to weewx.com. Both are easy.

glennmckechnie, 2022-02-08:
I use the setup.py method and I forgot about the packaging requirements of the distros. I agree that you don't want to be managing two versions, or including symlinks.
I wanted to steer clear of the symlinks because it would require webserver configuration and I felt that was just complicating things way more than intended. Just a quick, easy way to install the local copy of the docs, in particular for the new, novice users (and disorganized existing users!)

`file:///` and weewx.com
Yep, agreed that they are options and after a few visits I usually bookmark them (again), but then find that the bookmark cycles off the bookmark bar and I'm back to a click frenzy again. Rinse repeat.

Maybe the concept could be reduced to just point that `$show_doc` Seasons link to https://weewx.com/docs/usersguide.htm and do away with the `[StdReport] [[Documentation]]` skin part. Sadly Firefox won't follow a page link to `file:///home/weewx/docs/usersguide.htm` and as I recall, that is deliberate.
That would only require the `Seasons/skins.conf` , `Seasons/titlebar.inc` and a (yet to be created) `seasons.css` from that  `weewx_documentation_concept.tar.gz` file above. I'm not familiar with the mechanisms for other skins.

That would fulfill the original intent of having the Docs linked to/with the product, and hopefully increasing their visibility and use. (No, I'm not in marketing but for me it's a logical fit).

One final tweak to this revised suggestion would be to make that Docs URL configurable within `skin.conf`. That would satisfy my preference to use the current local copy, after I manually transfer it (for an advanced economy, we still have more outages than you'd expect).

----

Assuming the simplified version doesn't carry, and getting back to the original, separate skin idea, then I'm happy enough to turn that `weewx_documentation_concept.tar.gz` example into a more flexible example (as per mathews option), maintain it and add it to the `weewx/wiki` . I'd have to explore what can be automated considering all the variations for the document locations, versions. It would involve manual steps for the `skin.conf` configurations anyway so clear instructions for copying the docs locally wouldn't be out of character for it.

And a question. I'm under the impression that a particular version of weewx should access the docs that came with it, rather than the updated site docs. Correct?

tkeffer, 2022-02-08:
> And a question. I'm under the impression that a particular version of weewx should access the docs that came with it, rather than the updated site docs. Correct?

Yes, but. We try very hard to maintain backwards compatibility of everything. So, if it's in the docs, it should work for later versions. That's why I often refer to https://www.weewx.com/docs/

I think a separate skin is the way to go. You could probably hack its install script to set up the symlinks, if that's the way you decide to go.

tkeffer, 2022-02-09:
Closing this issue.

closed 2022-02-09 by tkeffer