weewx/weewx #750: Is it possible to modularly install drivers and utils of a specific station?

saveriogzz, 2022-02-16:
Hello! Thanks for the nice project :) it is the first issue that I opens here

I am working on a project that needs to query and send data from various Davis Vantage Pro 2. I went through the very well designed vantage's driver in this project and I was wondering whether it would be possible to install it alone, without the need to run weewx as a whole (something like `pip install 'weewx[vantage]'`).
The reason I'm asking this is because I am looking into a way to automate the deployment of python microservices (Docker containers) that serve as data-loggers for the weather stations and weexw's vantage module capabilities of setting up values on the console and querying data would be extremely useful.

I'm of course willing to send contributions towards this feature whether the community would find it suitable!
Thanks a lot in advance for your help

tkeffer, 2022-02-16:
I don't see why the Vantage driver cannot be used without using the rest of WeeWX. After all, it's possible to run the driver as a standalone program, so I don't see why other programs couldn't use it.

However, be aware that it has a number of dependencies on the rest of the WeeWX API.

We use the "issues" list for bug tracking, not for general queries. If you have follow-on questions, could you direct them to [weewx-development](https://groups.google.com/g/weewx-development)? That way, others can see the questions and, perhaps, help you answer them.

closed 2022-02-16 by tkeffer

saveriogzz, 2022-02-17:
Thanks @tkeffer !