weewx/weewx #756: ws23xx driver does not process 'rain' correctly

tkeffer, 2022-03-01:
See this thread: https://groups.google.com/g/weewx-user/c/8BEGeD7gAs0

In the ws23xx driver, `rain` is calculated as the difference between values of `totalRain`.

When weewxd first starts up, it goes into a "catch up" mode, where it downloads all archive records available in the console. In this case, there may be many records, so the differences between values of `totalRain` can be easily calculated, so `rain` can be calculated.

However, once catch up mode is over, only single archive records are retrieved from the console. Unfortunately, the driver does not save the old value of `totalRain`, so it has to start anew every time, resulting in 'rain' always being set to null.

tkeffer, 2022-08-12:
@matthewwall Have you had a chance to look at this?

mendikant, 2023-01-12:
Very happy to see that it is a bug. So I don't have to search long where my problem is. 
For me this occurs with a GW2000

gjr80, 2023-01-13:
@mendikant I'm not sure how this problem helps with your GW2000; maybe the same symptoms but almost certainly a different cause given you must be using a different driver. If you have an issue with the GW2000 try posting in [weewx-user](https://groups.google.com/g/weewx-user).

mendikant, 2023-01-13:
@gjr80 thank you, i posted there too

closed 2024-06-04 by github-actions[bot]