weewx/weewx #758: Allow user-selectable image formats

tkeffer, 2022-03-04:
Right now, everything is hardwired for .PNG files. It would be nice to allow users to select any format supported by PIL or Pillow.

N1OTX, 2024-05-18:
# What has to be done to allow user selection of an image format

1. Create image_format variable 
2. Create image_options variable as a list
3. Replace hard coded .extension in skins

Places that I am aware of where image_format variable would be used.
 src/weewx/imagegenerator.py

 Currently to coax imagegenerator to produce WebP images I only change 2 lines
 Line 93 from
`    img_file = os.path.join(image_root, '%s.png' % plotname)`
To
`    img_file = os.path.join(image_root, '%s.webp' % plotname`
So image_format would replace hard coded extension here.

Line 118 from
`    image.save(img_file)`
To
`    image.save(img_file,lossless = True)`
So 
image_options would be used here
There may well be other things that should be done in imagegenerator.py that I am unaware of, my python skills are weak at best.

The other place that image_option would be used is to replace hard coded .extension in skins.
Some skins such as Seasons require very few changes.
index.html 
Line 68
`    [image]`

telemetry.html
line 60
`    [image]`

Others such as Standard require multiple changes in multiple files
index.html.tmpl
month.html.tmpl
week.html.tmpl
year.html.tmpl
All these files contain have lots of hard coded file extensions such as
`    [image]`
That would need to be replaced.

### **WHY ALLOW USER TO PICK A Format other than the tried and try PNG ?**

This issue was opened because 2 years ago in March 2022 I slightly modified imagegenerator.py to generate images in the WebP format. WeeWx plot images in WebP format tend to be 59% smaller, with no noticeable difference in quality.

### **Warning not all browsers support WebP.**
As of May 2024 the web site https://caniuse.com/webp estimates that globally WebP is supported in 96.29% of browsers in use. In practical use having run a WeeWx WebP image only website for 2 years no one has noted unviewable images, with the exception of @vinceskahan who noted in May 2022, while testing my site,no images shown in Safari 15.3 on an Intel MacBook Air, how ever chrome on the same device was fine. The issue was Safari 14.0 – 15.6 has full support of WebP, but requires macOS 11 Big Sur or later.

### **Testing of image formats**
Pillow supports many image formats, those applicable for web usage with WeeWx seem to be JPG, PNG, GIF, and WebP. 
Test were run for each image format with weectl report, with the same date and time, to compare file size and image quality.
All image formats were tested with there Pillow default settings, with the exception of WebP.
Pillow defaults to saving WebP as a lossy image with a quality setting of 75, which is not ideal at all for the images that WeeWx produces. In fact a WebP lossy file size will be only slightly small then png and slightly lower in image quality.
So for WebP the losseless = True option was set. This produces a much smaller file of higher quality.
This seemed strange to me so I dug into it and found this explanation from Google the inventors of WebP.
If the source image is in lossless ARGB format, the spatial downsampling to YUV420 will introduce new colors that are harder to compress than the original ones. This situation can typically happen when the source is in PNG format with few colors: converting to lossy WebP (or, similarly to a lossy JPEG) will potentially result in a larger file size.
**Bottom Line to produce great quality very small WebP images with WeexWx you must add a lossless = True option.** 

python SSIM-PIL was used to compare the .png and .webp images.
SSIM-PIL allows comparison of two images using the structural similarity algorithm (SSIM).
The resulting value varies between 1.0 for identical images and 0.0 for completely different images.
SSIM-PIL produced a SIMM = 1.0 when comparing WeeWx .png and .webp images, it declares there identical.

One randomly picked image example:
**JPG**
  **16K** /var/www/html/imagetest/daytempdew.jpg
  SIMM Comparing JPG to PNG = 0.9449346505161323
  Visual comparison no perceived differences.
[... 54 lines cut ...]
weekrx                   | 6912              | 2616               | \-62.15         
weektempdew              | 11504             | 4852               | \-57.82         
weektempdiff             | 9097              | 3486               | \-61.68         
weektempfeel             | 9689              | 3942               | \-59.31         
weektempin               | 8090              | 3326               | \-58.89         
weekuv                   | 5787              | 2558               | \-55.80         
weekvolt                 | 8760              | 2898               | \-66.92         
weekwind                 | 11241             | 4630               | \-58.81         
yearbarometer            | 8594              | 3406               | \-60.37         
yearhilowmonth           | 7693              | 3402               | \-55.78         
yearhilow                | 10570             | 4572               | \-56.75         
yearhum                  | 8602              | 3636               | \-57.73         
yearradiation            | 8545              | 3408               | \-60.12         
yearrain                 | 7383              | 2950               | \-60.04         
yearrx                   | 6408              | 2644               | \-58.74         
yeartempdew              | 10896             | 4828               | \-55.69         
yeartempdiff             | 8705              | 3614               | \-58.48         
yeartempfeel             | 9661              | 4100               | \-57.56         
yeartempin               | 8831              | 3492               | \-60.46         
yearuv                   | 8679              | 3436               | \-60.41         
yearvolt                 | 8466              | 3080               | \-63.62         
yearwind                 | 10655             | 4858               | \-54.41         
**Total Size All Images**| **434172**        | **177062**         | **\-59.22**

Thanks Paul

tkeffer, 2024-05-18:
Thanks for your detailed analysis, Paul!

When I first created this issue I thought, "Oh, this will be easy." But then, when I investigated more closely, I realized that replacing the hard-coded extensions in the skins is actually pretty hard. After an hour or two of messing around, I gave up.

closed 2024-12-13 by github-actions[bot]