weewx/weewx #760: Update Seasons skin seasons.css file to favor woff2 font

N1OTX, 2022-03-08:
@font-face CSS at-rule follows precedence,first font hit that system can use will be downloaded.
Since Seasons skin was released in WeeWX 3.9.0 02/05/2019 woff was listed before woff2.
woff2 supports better compression so it's a smaller download. Any current user using unmodified seasons.css will never use woff2.  
This would fix issue:

/* use the local fonts */
@font-face {
    font-family: 'Open Sans';
    src: url('font/OpenSans.woff2') format('woff2'),
    url('font/OpenSans.woff') format('woff');
}

Verifed by reading:
CSS Fonts Module Level 3 at:
https://www.w3.org/TR/css-fonts-3/#font-face-rule

tkeffer, 2022-03-08:
I have no problem with this change, but if I swap woff for woff2, I'm seeing no difference in size in the .png files.

Maybe this is peculiar to my system... I don't know.

N1OTX, 2022-03-08:
Sorry not explaining issue clearly. All about Font File OpenSans.woff VS OpenSans.woff2  nothing to do with a .png file.
Online check than shows what request are made on a website
 https://tools.keycdn.com/speed

My site: Note It's using OpenSans.woff2
OpenSans.woff2 
https://pauland.net/font
STATUS	  SIZE	 TIME
200  	10.1 KB	 3.42 ms

Your Site Note It's using OpenSans.woff
OpenSans.woff
https://www.threefools.org/weewx/font
STATUS	SIZE	 TIME
200	        19.8 KB	  198.87 ms

Trying to save 9.7KB  seems silly I known.

tkeffer, 2022-03-08:
<smacks forehead> 
Of course. Silly me.

It's only a 10kB improvement, but why not!

Fixed in commit https://github.com/weewx/weewx/commit/7f424bebd2325e49675bae11d747d6d2262a7be4

closed 2022-03-08 by tkeffer

N1OTX, 2022-03-08:
Thank you Tom. You never disappoint!