weewx/weewx #763: Bullseye version of Debian cannot update WeeWX

tkeffer, 2022-03-21:
Updating WeeWX from Bullseye gives the following error:

```
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Ign:3 http://weewx.com/apt/python3 bullseye InRelease
Err:4 http://weewx.com/apt/python3 bullseye Release
  404  Not Found [IP: 34.218.218.142 80]
Reading package lists... Done
E: The repository 'http://weewx.com/apt/python3 bullseye Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

vinceskahan, 2022-04-05:
I can't reproduce this one.   Steps followed on a clean latest bullseye Raspian pi4:

```
wget -qO - https://weewx.com/keys.html | sudo gpg --dearmor --output /etc/apt/trusted.gpg.d/weewx.gpg
wget -qO - https://weewx.com/apt/weewx-python3.list | sudo tee /etc/apt/sources.list.d/weewx.list
sudo apt-get update

apt search weewx --all-releases

# install a specified version
sudo apt-get install weewx=4.6.2-1 -V
(verified it's running the desired version)

# upgrade to the latest version
sudo apt-get install weewx
(verified it's now running the latest version)

# clean it all up
sudo apt-get remove weewx
sudo apt autoremove
```

tkeffer, 2022-04-06:
It seems the user has deleted the original post, but here's the thread that led to the issue: https://groups.google.com/g/weewx-user/c/7T53ym4UdI8/m/MiEKNzU3AQAJ

tkeffer, 2022-08-12:
@vinceskahan Any reason not to close this?

vinceskahan, 2022-08-12:
No reason I can think of.  It worked fine for me back then.  How I tested was in my earlier followup.

closed 2022-08-12 by tkeffer