weewx/weewx #769: wgust history error rtgd version 5.5.0

kskenyon, 2022-04-26:
windGust data should be coming into the buffer but...

INFO user.rtgd: rtgdthread: **** Traceback (most recent call last):
INFO user.rtgd: rtgdthread: ****   File "/usr/share/weewx/user/rtgd.py", line 1889, in process_packet
INFO user.rtgd: rtgdthread: ****     data = self.calculate(cached_packet)
INFO user.rtgd: rtgdthread: ****   File "/usr/share/weewx/user/rtgd.py", line 2157, in calculate
INFO user.rtgd: rtgdthread: ****     wgust = self.buffer['windGust'].history_max(ts, age=600).value
INFO user.rtgd: rtgdthread: **** AttributeError: 'NoneType' object has no attribute 'value'

tkeffer, 2022-04-26:
I don't know what extension you are using, but it is not part of WeeWX. You'll have to ask the extension author. The issues list is for tracking bugs in WeeWX itself.

closed 2022-04-26 by tkeffer

kskenyon, 2022-04-26:
Oops. Sorry.