weewx/weewx #770: WeeWx v4.8.0 - Error in Seasons Skin

mr-bryn, 2022-05-05:
I’m running weewx from the weewx repository on a debian 11 box hooked up to a Oregon Scientific WMR200.

I’ve just upgraded weewx from v4.7.0 to v4.8.0

After the upgrade the Seasons skin is throwing errors. There are 3 lines of code contained within sensors.inc that seem to be the source of the error.

The error message I’m seeing is:
```
May  5 15:38:19 wyse weewx[36608] ERROR weewx.cheetahgenerator: Evaluation of template /etc/weewx/skins/Seasons/index.html.tmpl failed.
May  5 15:38:19 wyse weewx[36608] ERROR weewx.cheetahgenerator: **** Ignoring template /etc/weewx/skins/Seasons/index.html.tmpl
May  5 15:38:19 wyse weewx[36608] ERROR weewx.cheetahgenerator: **** Reason: cannot find 'raw' while searching for 'lasttime.raw'
May  5 15:38:19 wyse weewx[36608] ERROR weewx.cheetahgenerator: **** To debug, try inserting '#errorCatcher Echo' at top of template
May  5 15:38:19 wyse weewx[36608] ERROR weewx.cheetahgenerator: Evaluation of template /etc/weewx/skins/Seasons/telemetry.html.tmpl failed.
May  5 15:38:19 wyse weewx[36608] ERROR weewx.cheetahgenerator: **** Ignoring template /etc/weewx/skins/Seasons/telemetry.html.tmpl
May  5 15:38:19 wyse weewx[36608] ERROR weewx.cheetahgenerator: **** Reason: cannot find 'raw' while searching for 'lasttime.raw'
May  5 15:38:19 wyse weewx[36608] ERROR weewx.cheetahgenerator: **** To debug, try inserting '#errorCatcher Echo' at top of template
```
and the specific lines of code all look like:

`#set $lasttime = $getattr($recent, $x).lasttime.raw`

The suggested errorcatcher line is already in the template, so I'm not sure how to debug this further. 

I’ve made a temporarily work around by commenting out this line and the following line (which references $lasttime), but would appreciate some guidance on fixing it permanently.

tkeffer, 2022-05-05:
I've tried mightily, but am unable to replicate this error. 

Have you made any modifications to Seasons? In other words, is the version you are using identical to what is being shipped with WeeWX?

mr-bryn, 2022-05-06:
I haven't made any modifications to Seasons as I never even looked at that code until now!

I've taken a look at the current sensors.inc file (https://github.com/weewx/weewx/blob/master/skins/Seasons/sensors.inc) and I can see the lines in question are there too (line 99, line 113, line 127). To be sure I could copy/paste the content of sensors.inc from github to my install?

I'm using the driver for the WMR200 from https://github.com/weewx/weewx-wmr200 - perhaps that's related?

mr-bryn, 2022-05-06:
Could this change be related to my issue?
https://github.com/weewx/weewx/commit/c0ddc0d94add5a889cffd70c1769f31b1856d65d

tkeffer, 2022-05-06:
To answer your questions:
1. Yes, the problem is in `sensors.inc`, which is then included in `index.html.tmpl` and `telemetry.html.tmpl`. 
2. It's probably related to the WMR200 in that it seems to be using some sensors that are causing the problems.
3. I doubt that commit is involved.

I'd like to see what sensor type is causing  the problem. Can you replace your copy of `sensors.inc` with copy below? In place of the last time seen, it will give the name of the sensor. That can help us narrow down the problem.

```
## sensors module for weewx skins
## Copyright Tom Keffer, Matthew Wall
## See LICENSE.txt for your rights
#errorCatcher Echo
#encoding UTF-8

## this is a conditional display of sensor data, including connectivity,
## battery status, and various voltages.  if there are no data available,
## then this degenerates to nothing displayed.

## remember the current time - we will use it more than once.
#import time
#set $now = time.time()

## use this span to determine whether there are any data to consider.
#set $recent=$span($day_delta=30, boundary='midnight')

#def get_battery_status($x)
#if $x is None
<span class="status_unknown">$gettext('UNKNOWN')</span>
#elif $x == 1
<span class="status_low">$gettext('LOW')</span>
#else
<span class="status_ok">$gettext('OK')</span>
#end if
#end def

## provide an indication of how much time has passed since the last sensor
## reading.
#def get_time_delta($last_ts, $now)
  #if $last_ts
    #set $delta = int($now - $last_ts)
    #if $delta < 60

#elif $delta < 3600
      #set $minutes = int($delta / 60)
$minutes $gettext('minutes ago')
    #elif $delta < 86400
      #set $hours = int($delta / 3600)
$hours $gettext('hours ago')
    #else
      #set $days = int($delta / 86400)
$days $gettext('days ago')
    #end if
  #else
$gettext('never')
  #end if
#end def

## Get the list of sensor observations from the configuration file, otherwise
## fallback to a very rudimentary set.
#set $sensor_connections = $to_list($DisplayOptions.get('sensor_connections', ['rxCheckPercent']))
#set $sensor_batteries = $to_list($DisplayOptions.get('sensor_batteries', ['outTempBatteryStatus', 'inTempBatteryStatus', 'rainBatteryStatus', 'windBatteryStatus', 'uvBatteryStatus', 'txBatteryStatus']))
#set $sensor_voltages = $to_list($DisplayOptions.get('sensor_voltages', ['consBatteryVoltage', 'heatingVoltage', 'supplyVoltage', 'referenceVoltage']))

## first see what sensor data are available

#set $have_conn = 0
#for $x in $sensor_connections
  #if $getattr($recent, $x).has_data
    #set $have_conn = 1
  #end if
#end for

#set $have_battery = 0
#for $x in $sensor_batteries
  #if $getattr($recent, $x).has_data
    #set $have_battery = 1
  #end if
#end for

#set $have_voltage = 0
#for $x in $sensor_voltages
  #if $getattr($recent, $x).has_data
    #set $have_voltage = 1
  #end if
#end for

## now display the available data only

#if $have_conn or $have_battery or $have_voltage
<div id='sensors_widget' class="widget">
  <div class="widget_title">
    <a href="telemetry.html">$gettext("Sensor Status")</a>
    <a class="widget_control" onclick="toggle_widget('sensors')">&diams;</a>
  </div>
  <div class="widget_contents">
  <table>

#if $have_conn
    <tr><th>$gettext("Connectivity")</th><th></th></tr>
#for $x in $sensor_connections
  #if $getattr($recent, $x).has_data
    <tr>
      <td class="label">$obs.label[$x]</td>
      <td class="data">$getVar('current.' + $x)</td>
      <td class="timestamp">$x</td>
    </tr>
  #end if
#end for
#end if

#if $have_battery
    <tr><th>$gettext("Battery Status")</th><th></th></tr>
#for $x in $sensor_batteries
  #if $getattr($recent, $x).has_data
    <tr>
      <td class="label">$obs.label[$x]</td>
      <td class="data">$get_battery_status($getVar('current.%s.raw' % $x))</td>
      <td class="timestamp">$x</td>
    </tr>
  #end if
#end for
#end if

#if $have_voltage
    <tr><th>$gettext("Voltage")</th><th></th></tr>
#for $x in $sensor_voltages
  #if $getattr($recent, $x).has_data
    <tr>
      <td class="label">$obs.label[$x]</td>
      <td class="data">$getVar('current.' + $x)</td>
      <td class="timestamp">$x</td>
    </tr>
  #end if
#end for
#end if

  </table>
  </div>

</div>
#end if
```

mr-bryn, 2022-05-10:
Thanks Tom, I've replaced sensors.inc with the code supplied.
This didn't cause any errors to appear in the log file.
Screenshot of the sensors area on the webpage attached.
[image]

tkeffer, 2022-05-11:
Thanks for that. So, we know that the problem is being caused by one of those four status fields.

Before we go chasing possible bugs in the code, I want to elimate the possibility that the daily summaries in the database got out of sync with the main archive table somehow.

1. Drop and rebuild the daily summaries
    ```
    wee_database --drop-daily
    wee_database --rebuild-daily
    ```
2. Restore your original `sensors.inc`
3. Stop `weewxd`
4. Run `wee_reports`. See if the reports continue to crash.

mr-bryn, 2022-05-12:
Thanks Tom. It took quite a while to rebuild the daily summaries as I've got 7 years of data!

After the rebuild and replacing sensors.inc I had no problems running the wee_reports command. The output is below, note the 51 minutes was about how long it took to rebuild the summaries, and I had stopped weewx for this process.
[image]

However, after I restarted the weewx service I got the same errors in the logs.

For now I've commented these lines out again, but I noticed the UV sensor wasn't listed in the above screenshot, so perhaps this is the one causing problems?

With the lines commented, the output looks like this:
[image]

tkeffer, 2022-05-12:
Curious that it worked while using `wee_reports`, but not while using `weewxd`. It's possible one of the types is not in your database, so `weewxd` is using the value in the current, "live" archive record (coming off the station), which would not be available to `wee_reports`.

I suspect `uvBatteryStatus` as well, but with what follows, it's worth checking `outTempBatteryStatus`, `rainBatteryStatus`, and `windBatteryStatus` as well.

1. What kind of database? SQLite?
2. Check the schema and post here. 
   ```
   echo ".schema" | sqlite3 /var/lib/weewx.sdb
   ```
3. Run `weewxd` directly off the command line and watch the print out, in particular the archive records. They will be marked with `REC`. Do the archive records include all 4 types? Compare to the schema.
4. Another thing to try: go into `skin.conf` and remove `uvBatteryStatus` from the list `sensor_batteries` (under `[DisplayOptions]`). Then run `weewxd` again. Does this change anything? Try for the other battery statuses as well.
5. Finally, if all else fails, see what's in `uvBatteryStatus` for the last 30 days:
   ```
   select datetime(dateTime,'unixepoch','localtime'), uvBatteryStatus from archive where dateTime>unixepoch('now','-30 day') and uvBatteryStatus is not null order by dateTime desc limit 20;
   ```
   This will print the last 20 non-null records.

mr-bryn, 2022-05-14:
Hi Tom,

1. Yes I'm using an SQLite database - as I mentioned I've been running weewx for several years with this same database. Also, the UV sensor was added only in the last couple of years.
2. Schema attached - [weewx schema.txt](https://github.com/weewx/weewx/files/8691922/weewx.schema.txt)
3. It looks like the battery status isn't always reported, but I spied all 4 in one record - attached output from weewxd [weewxd_output.txt](https://github.com/weewx/weewx/files/8691945/weewxd_output.txt)
4. I skipped this for now because...
5. ...running the SQL said "Error: no such column: uvBatteryStatus"

I'm assuming at this point I need to add the uvBatteryStatus column before going any further. How can I do that?

tkeffer, 2022-05-14:
Somehow you ended up with an archive table with only a handful of types, but daily summaries that have lots of types. It's little wonder that `sensors.inc` was getting confused! Did you drop some types from the archive table, but not the daily summaries?

When you rebuilt the daily summaries, did you drop them first? 

I'm very curious how your database got into such a distorted state. First time I've seen this.

mr-bryn, 2022-05-14:
I have no idea! The data has been kicking along for 8 years. It began in a MySQL database and then I ported to SQLite - perhaps it happened somehow along the way?

What's my best option for getting it back in shape?

tkeffer, 2022-05-14:
Well, if it's been working for you, probably the easiest would be to add the missing `uvBatteryStatus` then call it good.
```
cd /var/lib/weewx
# Make a backup first:
cp weewx.sdb weewx.sdb.backup
# Add the missing column
echo "ALTER TABLE archive ADD COLUMN \`uvBatteryStatus\` REAL;" | sqlite3 weewx.sdb
```

The backslash before the backtick (\') is to escape the backtick. Otherwise, the shell will think it's a command you want to run. Strictly speaking the backticks are not necessary at all, but the other types have them, so this keeps things consistent.

mr-bryn, 2022-05-16:
That's sorted it, although I don't get a last time seen in the output. Honestly I don't really care, so unless this is another issue that needs to look into I'll close the issue.

closed 2022-05-16 by tkeffer