weewx/weewx #773: weewx.manager INFO logs should be debug messages

mfraser, 2022-05-29:
I'm seeing a lot of lines like these:
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'
May 29 16:15:48 weathercam weewx[5631] INFO weewx.manager: Added record 2022-05-29 16:15:48 BST (1653837348) to database 'forecast.sdb'

In the syslog even though I have log_success set to False in /etc/weewx/weewx.conf.

tkeffer, 2022-05-29:
Apparently, the forecast module is not honoring the `log_success` flag. It needs to pass it down to the `Manager.addRecord()` method.

There's a multiple of forks of the forecast module out there now. Which one are you using?

mfraser, 2022-05-29:
> Apparently, the forecast module is not honoring the `log_success` flag. It needs to pass it down to the `Manager.addRecord()` method.
> 
> There's a multiple of forks of the forecast module out there now. Which one are you using?

This one https://github.com/chaunceygardiner/weewx-forecast
Would also explain these lines:
May 29 19:15:53 weathercam weewx[5631] INFO user.forecast: WUThread: WU: download forecast from '
May 29 19:15:53 weathercam weewx[5631] INFO user.forecast: OWMThread: OWM: download forecast from

tkeffer, 2022-05-29:
I was going to suggest creating an issue, but I see that it's not available for that fork. You could try contacting the maintainer of the fork directly.

closed 2022-05-29 by tkeffer