weewx/weewx #779: Uncaught exception "Attempt to add out-of-interval record" causes weewx to crash

bwarden, 2022-07-01:
My backend driver (user.MQTTSubscribe) occasionally provides a record that's timestamped right between midnight and 00:00:01. When that happens, weewx will throw an exception, which of course is not handled, so it dies. It would be nice if this exception could be caught and logged instead.

```
Jun 28 00:00:01 snafu python2[236016]: weewx[236016] INFO weewx.manager: Added record 2022-06-28 00:00:00 PDT (1656399600) to database 'weewx'
Jun 28 00:00:01 snafu python2[236016]: weewx[236016] INFO weewx.engine: Main loop exiting. Shutting engine down.
Jun 28 00:00:01 snafu python2[236016]: weewx[236016] INFO weewx.engine: Shutting down StdReport thread
Jun 28 00:00:01 snafu python2[236016]: weewx[236016] INFO user.MQTTSubscribe: (Driver) Disconnected with result code 0
Jun 28 00:00:01 snafu python2[236016]: weewx[236016] CRITICAL __main__: Caught unrecoverable exception:
Jun 28 00:00:01 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****  Attempt to add out-of-interval record (1656399600.85) to timespan ([2022-06-27 00:00:00 PDT (1656313200) -> 2022-06-28 00:00:00 PDT (1656399600)])
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****  Traceback (most recent call last):
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****    File "/usr/share/weewx/weewxd", line 153, in main
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****      engine.run()
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 210, in run
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****      self.dispatchEvent(weewx.Event(weewx.NEW_LOOP_PACKET, packet=packet))
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 245, in dispatchEvent
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****      callback(event)
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****    File "/usr/share/weewx/user/raw.py", line 131, in newLoopPacket
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****      self.dbm.addRecord(packet)
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/manager.py", line 304, in addRecord
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****      self._addSingleRecord(record, cursor, log_success, log_failure)
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/manager.py", line 1017, in _addSingleRecord
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****      _day_summary.addRecord(record, weight=_weight)
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/accum.py", line 435, in addRecord
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****      % (record['dateTime'], self.timespan))
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****  OutOfSpan: Attempt to add out-of-interval record (1656399600.85) to timespan ([2022-06-27 00:00:00 PDT (1656313200) -> 2022-06-28 00:00:00 PDT (1656399600)])
Jun 28 00:00:02 snafu python2[236016]: weewx[236016] CRITICAL __main__:     ****  Exiting.
Jun 28 07:59:52 snafu systemd[1]: Stopping LSB: weewx weather system...
```

bwarden, 2022-07-01:
Oh, now I see the latest development has reworked _addSingleRecord in manager.py to avoid this. I'll try to update.

tkeffer, 2022-07-01:
I suspect the bigger problem is that your LOOP packet has a floating point timestamp (which is not an unreasonable thing). Unfortunately, because of its heritage from wview, WeeWX expects integers.

Could you modify your driver to emit only integers?

bwarden, 2022-07-01:
Yeah, that's part of it. Unfortunately, the underlying device provides partial packets, so it's quite possible to get two different packets within the same integer timestamp, causing a different problem.

tkeffer, 2022-07-01:
Let me try to figure out an easy way to get timestamps that fall within that one second interval to be assigned to the correct day...

tkeffer, 2022-07-02:
I came up with something. No idea if it works because I don't have an easy way of generating packets with such close timestamps.

Replace your copy of `weeutil.weeutil.py` with this one: https://raw.githubusercontent.com/weewx/weewx/startOfArchiveDay/bin/weeutil/weeutil.py

bwarden, 2022-07-05:
Thanks, I'll give it a try and report back in a few days.

bwarden, 2022-07-08:
So far no issues; I'll let it continue over the weekend and report again on Monday.

bwarden, 2022-07-11:
Looks like the new version handled it gracefully:
```
Jul 10 23:59:55 snafu python2[2342683]: weewx[2342683] INFO weewx.restx: MQTT: Published record 2022-07-10 23:59:51 PDT (1657522791)
Jul 10 23:59:57 snafu python2[2342683]: weewx[2342683] INFO weewx.manager: Added record 2022-07-10 23:59:56 PDT (1657522796) to database 'weewx'
Jul 10 23:59:57 snafu python2[2342683]: weewx[2342683] INFO weewx.manager: Added record 2022-07-10 23:59:56 PDT (1657522796) to daily summary in 'weewx'
Jul 10 23:59:57 snafu python2[2342683]: weewx[2342683] INFO weewx.restx: MQTT: Published record 2022-07-10 23:59:56 PDT (1657522796)
Jul 11 00:00:01 snafu python2[2342683]: weewx[2342683] INFO weewx.manager: Added record 2022-07-11 00:00:00 PDT (1657522800) to database 'weewx'
Jul 11 00:00:03 snafu python2[2342683]: weewx[2342683] INFO weewx.manager: Added record 2022-07-11 00:00:00 PDT (1657522800) to daily summary in 'weewx'
Jul 11 00:00:03 snafu python2[2342683]: weewx[2342683] ERROR user.MQTTSubscribe: (Driver) Ignoring record because 1657522500 archival start is before previous archive start 1657522800: 'dateTime': '1657522799.83', 'usUnits': '16', 'windSpeed': '0.0'
Jul 11 00:00:05 snafu python2[2342683]: weewx[2342683] INFO weewx.manager: Added record 2022-07-11 00:00:04 PDT (1657522804) to database 'weewx'
Jul 11 00:00:05 snafu python2[2342683]: weewx[2342683] INFO weewx.manager: Added record 2022-07-11 00:00:04 PDT (1657522804) to daily summary in 'weewx'
Jul 11 00:00:08 snafu python2[2342683]: weewx[2342683] INFO weewx.restx: MQTT: Published record 2022-07-11 00:00:00 PDT (1657522800)
Jul 11 00:00:08 snafu python2[2342683]: weewx[2342683] INFO weewx.restx: MQTT: Published record 2022-07-11 00:00:04 PDT (1657522804)
Jul 11 00:00:10 snafu python2[2342683]: weewx[2342683] INFO weewx.manager: Added record 2022-07-11 00:00:08 PDT (1657522808) to database 'weewx'
Jul 11 00:00:11 snafu python2[2342683]: weewx[2342683] INFO weewx.manager: Added record 2022-07-11 00:00:08 PDT (1657522808) to daily summary in 'weewx'
Jul 11 00:00:12 snafu python2[2342683]: weewx[2342683] INFO weewx.restx: MQTT: Published record 2022-07-11 00:00:08 PDT (1657522808)
```

tkeffer, 2022-07-16:
Wrote some more tests.

closed 2022-07-16 by tkeffer