weewx/weewx #780: ftpupload   'utf-8' codec can't decode byte 0x94 in position 41: invalid start byte

seanford, 2022-07-16:
Ran into an error for the first time...

Jul 16 12:12:58 raspberrypi wee_reports[2688] ERROR weewx.reportengine: Caught unrecoverable exception in generator 'weewx.reportengine.FtpGenerator'
Jul 16 12:12:58 raspberrypi wee_reports[2688] ERROR weewx.reportengine:         ****  'utf-8' codec can't decode byte 0x94 in position 41: invalid start byte

I believe the last upload file was corrupted, which led to the error.
1
The fix was to add UnicodeDecodeError, to the list of catchable errors in line 244 of ftpload.py.

On the next pass it uploaded all of the files and re-wrote the timestamp file.

I think this should be added to the base code.

tkeffer, 2022-07-16:
Before we simply suppress this error, I'd like to know more about why it's occurring. From your description, it sounds like the relevant code is:

```python
        try:
            with open(timestamp_file_path, "rb") as f:
                timestamp = cPickle.load(f)
                fileset = cPickle.load(f)
                hashdict = cPickle.load(f)
        except (IOError, EOFError, cPickle.PickleError, AttributeError):
```

The file is being opened in binary mode ("rb"), so no decoding should be applied to it. Any thoughts why a decode error would be occurring?

A few other questions:
- Could you show more of the log? In particular, the full stack log? Let's make sure we're looking at what we think we're looking at.
- Which version of Python?
- Which version of WeeWX?
- Does this occur with every report? Or, just once? Occasionally?

seanford, 2022-07-18:
Wish I had kept that info.  Here are the answers I can provide...

Python 3.9.2
WeeWx 4.8.0

This was a perfectly working setup....I had been doing some testing on a new service I wrote to pull canal data from a JSON feed, implementing MQTTSubscribe to read Lorawan sensor for lake levels, and upgrading from a Pi2 to a Pi 4 I just received.  I wish I could say exactly where it stopped working.  It may have been an improper shutdown.  But the reports were generating fine...I know this as I have apache installed in the pi and they were up to date.  It was just failing at the ftpupload.

I looked at the syslog error and traced it back to the #FTP.last file.  It was trying to read that file and throwing the error I mentioned in my OP.  When I saw the code and the error type in the log, I figured I'd try and add the type to the except line as the worst thing that would happen would be for all files to be uploaded.

On the next run, that is exactly what happened....and then it has worked as before ever since.  That is why I suggested the update to the script.

tkeffer, 2022-07-20:
Until we can be sure we're chasing the right error, I'm going to close this. If it happens again, perhaps you could post the full stack log?

closed 2022-07-20 by tkeffer