weewx/weewx #794: Rain values in FineOffset station WH1080 are triggered by strong winds

aslak47, 2022-09-11:
I have a WH1080 station. In strong winds the rain sensor is reporting wrong values due to the construction of the sensor. It vibrates in strong winds and the vibration triggers the rain sensor.
Would it be possible to compensate for this setting the rain value to 0 when outdoor humidity is below a limit and wind or windgust is above another limit?

gjr80, 2022-09-11:
Probably better asked in [weewx-user](https://groups.google.com/g/weewx-user) rather than this list.

Nonetheless, you should be able to do what you seek using a correction under the `StdCalibrate` service. For example, to set `rain` to 0.0 when outside humidity > 30% and wind speed > 10 or wind gust > 15 add something like the following to `weewx.conf` (untested):
```
[StdCalibrate]
    [[Corrections]]
        rain = 0.0 if outHumidity > 30 and (windSpeed > 10 or windGust > 15) else rain
```
Points to note:
- the wind speed/gust units used will be the same as the units used in the database (unless you changed the database unit system that will be miles/hour)
- both loop packets and archive records will have the correction applied, this may have some unexpected/undesirable effects
- `rain` will be left unchanged if either `outHumidity`, `windSpeed` or `windGust` is `None` or absent (this may be an issue for partial packet stations rather than the WH1080)

aslak47, 2022-09-11:
Thanks a lot.
Regards, Aslak

closed 2022-09-11 by aslak47