weewx/weewx #795: import from WU returns only 1 row from actual date

Sebeck81, 2022-09-27:
Hello everyone,
I would like to retrieve the data from WU via wee_import with API. At the moment I only have the data set from the night of 00:04:52. With the command: wee_import --import-config=/etc/weewx/import/wu-example.conf --from=2022-09-20T22:30 --to=2022-09-27T06:00 I get the entire records. But the last one is from the night 00:04. Is it even possible to call up current data at intervals of about 10 minutes or only from the previous day? Thank you

[image]
[image]
[import.txt](https://github.com/weewx/weewx/files/9656703/import.txt)
[log.txt](https://github.com/weewx/weewx/files/9656704/log.txt)

gjr80, 2022-12-13:
My apologies @Sebeck81 I had not noticed this issue.

If I understand the issue when you run `wee_import` to import from WU using an import period of a number of days that includes today you get all expected records. However, if you run `wee_import` with no `--date`, `--from` or `--to` (ie you just import todays records) then all you see is a single record timestamped four minutes after midnight. Is this correct?

I seem to recall similar behaviour in the past whereby interrogating the WU API for the current days data did not always return the complete set of data for today.

Could you please confirm if this issue is still ongoing and if my understanding is correct and I will look at it further.

Sebeck81, 2022-12-13:
Hello,

Yes, that's right, I'm currently using the following API import url = "https://api.weather.com/v2/pws/observations/all/1day?"\             "stationId=%s&format=json&units=e&numericPrecision=decimal&apiKey=%s" \% (self.station_id, self.api_key)

I'm getting data every 10-15 minutes. I would like to use these settings to get the most recent data, but unfortunately it doesn't work so far  url = "https://api.weather.com/v2/pws/observations/current?"\stationId=%s&format=json&units=e&numericPrecision=decimal&apiKey=%s" \% (self.station_id, self.api_key)

closed 2025-02-14 by github-actions[bot]