weewx/weewx #798: Average of wind direction is incorrect

jkumeboshi, 2022-10-10:
Wind direction average should be calculated differently from other scalar averaging:

```
from math import *
from statistics import *

WD = [355, 5, 15]

u_east = mean([sin(x * pi/180) for x in WD])
u_north = mean([cos(x * pi/180) for x in WD])
unit_WD = atan2(u_east, u_north) * 180/pi
unit_WD = (360 + unit_WD) % 360

print("should be 5 (correct):", unit_WD)
print("should be 125 (wrong):", mean(WD))
```

this code returns 5°, the expected average of wind direction.
a normal scalar average returns wrong value of 125°

tkeffer, 2022-10-10:
Are you sure you are using the right tag?

Something like `$month.windDir.avg` will return the scalar average, which, as you point out, is incorrect.

However, did you try `$month.wind.vecdir`? It calculates the vector average. All of the skins use `wind.vecdir`, not `windDir.avg`.

jkumeboshi, 2022-10-10:
@tkeffer I'm looking at the vecdir variable, although it is a vector and it should contain the wind direction and one could ignore wind speed, I'm not sure is exactly what I meant.
Anyway the skin I'm using (poblabs/weewx-belchertown) should be able to extract the avg wind direction for an arbitrary date ranges, not just from day/month summaries the skin allows to configure charts containing aggregated data of arbitrary granularity and length (such avg aggregate all wind direction every 30 minutes for last 24h).

tkeffer, 2022-10-10:
Not sure what bug report you are submitting. Can you give me an example of what is not working?

If the skin uses the function `get_aggregate()` to calculate the wind vector average, it will work for any time range.

jkumeboshi, 2022-10-11:
@tkeffer looking at the skin code, it's clear that the belchertown skin doesn't use any weewx code to access data. Instead it uses SQL queries to directly access data sets.
Closing the issue since it's not a WeeWX problem

closed 2022-10-11 by jkumeboshi