weewx/weewx #802: "UNKNOWN" for Transmitter battery and console battery when vantage set to "both"

staze, 2022-10-24:
When weewx is set to query both LOOP and LOOP2, the transmitter battery and console battery both report as UNKNOWN and N/A regularly. It seems to be based on which LOOP packet was last received before weewx generated the web interface as the values often say "UNKNOWN 15 minutes ago" (or similar). 

If I recall, these battery numbers are only available in one of the two LOOP packets (which is why I have it set to both). Is there any reason we couldn't store these values in the DB and just grab the last "valid" packet if within X minutes of web generation?

Example here: https://www.staze.org/weewx/

[image]

tkeffer, 2022-10-24:
Hmmm, that's not what I would expect, but then I haven't tried mixing packet types other than in brief tests.

If you're using both LOOP and LOOP2 packets, you'll get either `0` or an `OR` mix of bad battery statuses for the LOOP packets, and `None` for the LOOP2 packets. These will then get added up by the accumulators, which ignore `None` values. The result should be either zero, or some non-zero value signalling a bad battery, which would then get stored in the database under type `txBatteryStatus`. 

I'd be curious what you have in your database. Try the following query. It will show the 20 most recent battery events.

```sql
select datetime(dateTime,'unixepoch','localtime'),txBatteryStatus from archive where txBatteryStatus > 0 order by dateTime desc limit 20;
```

staze, 2022-10-25:
no result from that query. I assume that means the column doesn't exist, and that certainly would explain the weird behavior. Did I forget to run some command with wee_* after changing station type to initialize that column in DB?

fwiw, I do "both" because there's some piece that's missing from loop2 that's in loop if I recall. Maybe signal strength? I swear I switched to just LOOP2 and noticed a missing value.

staze, 2022-10-25:
Hey Tom,

Sorry, no, I mean, it returns nothing. 

sqlite> select datetime(dateTime,'unixepoch','localtime'),txBatteryStatus from archive where txBatteryStatus > 0 order by dateTime desc limit 20;
sqlite> 

Doing just a 
select datetime(dateTime,'unixepoch','localtime'),txBatteryStatus from archive where txBatteryStatus != ''; 
I get entires like

2022-10-24 05:10:00|0.0
2022-10-24 06:20:00|0.0
2022-10-24 06:45:00|0.0
2022-10-24 06:55:00|0.0
2022-10-24 07:05:00|0.0
2022-10-24 08:05:00|0.0
2022-10-24 08:30:00|0.0
2022-10-24 08:40:00|0.0
2022-10-24 10:15:00|0.0
2022-10-24 10:40:00|0.0
2022-10-24 11:15:00|0.0
2022-10-24 11:20:00|0.0
2022-10-24 11:55:00|0.0
2022-10-24 12:30:00|0.0
2022-10-24 13:05:00|0.0
2022-10-24 15:15:00|0.0
2022-10-24 16:40:00|0.0
2022-10-24 17:05:00|0.0
2022-10-24 17:10:00|0.0
2022-10-24 18:45:00|0.0
2022-10-24 19:25:00|0.0

So it would seem like no values are being stored that aren't NULL or 0. =/ 

consBatteryVoltage seems to have values. So 
sqlite> select datetime(dateTime,'unixepoch','localtime'),consBatteryVoltage from archive where consBatteryVoltage > 0 order by dateTime desc limit 20;
2022-10-24 19:35:00|3.97
2022-10-24 19:25:00|3.97
2022-10-24 18:45:00|3.97
2022-10-24 17:10:00|3.97
2022-10-24 17:05:00|3.97
2022-10-24 16:40:00|3.97
2022-10-24 15:15:00|3.97
2022-10-24 13:05:00|3.97
2022-10-24 12:30:00|3.97
2022-10-24 11:55:00|3.97
2022-10-24 11:20:00|3.97
2022-10-24 11:15:00|3.97
2022-10-24 10:40:00|3.97
2022-10-24 10:15:00|3.97
2022-10-24 08:40:00|3.97
2022-10-24 08:30:00|3.97
2022-10-24 08:05:00|3.97
2022-10-24 07:05:00|3.97
2022-10-24 06:55:00|3.97
2022-10-24 06:45:00|3.97

looking at the page right now, I can see valid info:
[image]

is it just inferring from conBatteryVoltage that "Transmitter Battery" is "OK"? This is a Vantage Vue, talking to a WeatherEnvoy, with a WeatherlinkIP.

staze, 2022-10-25:
next refresh, battery status is back to unknown. 

Here are the last two columns in archive (select * from archive order by datetime desc limit 2;):

1666665900|1|5|30.2401883549342|52.8682354377445||30.244|||||||||1111.181153093||||49.9480029263908||||||||||||||||||||||52.01||||55.076894996819||49.3476172585756|46.0|71.2|||||||||||0.0|||||90.0|52.8|||||29.7359590832472||0.0||0.0||83.7083333333333||||||||||||||||||||||||||||52.8||2.0|0.0|0.0|0.0
1666665600|1|5|30.2371981849792|52.8682354377445||30.241|||||||||1111.181153093|||3.97|49.9480029263908||||||||||||||||||||||52.01||||55.076894996819||49.1658906510365|46.0|71.0|||||||||||0.0|||||90.0|52.8|||||29.7330094774659||0.0||0.0||84.5625||||||||||||||||||||||||0.0||||52.8||2.0|157.5|0.0|0.0

tkeffer, 2022-10-25:
Sorry, I cannot decode your last post without the database schema. Try a SELECT with explicit fields:

```sql
select dateTime,datetime(dateTime,'unixepoch','localtime'),txBatteryStatus,consBatteryVoltage from archive order by dateTime desc limit 20;
```

One other thing to note: because these fields are only in the LOOP packets, any records downloaded from the logger's memory will not contain them. So, any "catchups" will have null values.

staze, 2022-10-25:
Makes sense on the logger piece. Here's the output from that select.

1666710000|2022-10-25 08:00:00|0.0|3.97
1666709700|2022-10-25 07:55:00||
1666709400|2022-10-25 07:50:00||
1666709100|2022-10-25 07:45:00||
1666708800|2022-10-25 07:40:00||
1666708500|2022-10-25 07:35:00|0.0|3.97
1666708200|2022-10-25 07:30:00||
1666707900|2022-10-25 07:25:00||
1666707600|2022-10-25 07:20:00||
1666707300|2022-10-25 07:15:00||
1666707000|2022-10-25 07:10:00||
1666706700|2022-10-25 07:05:00||
1666706400|2022-10-25 07:00:00||
1666706100|2022-10-25 06:55:00||
1666705800|2022-10-25 06:50:00||
1666705500|2022-10-25 06:45:00||
1666705200|2022-10-25 06:40:00|0.0|3.97
1666704900|2022-10-25 06:35:00|0.0|3.97
1666704600|2022-10-25 06:30:00||
1666704300|2022-10-25 06:25:00||

tkeffer, 2022-10-25:
> Assuming that's the case, run weewxd from the command line and watch what it's printing for values of txBatteryStatus in the LOOP packets. You should be getting 0 (zero), not None.

staze, 2022-10-25:
okay, ran that for a good 5 minutes. consBatteryVoltage was steady at 3.97. txBatteryStatus looks like it's 0 every time there's the proper LOOP. the other LOOP packet (LOOP2?) doesn't include those values. 

`LOOP:   2022-10-25 12:01:04 PDT (1666724464) 'altimeter': '30.237', 'appTemp': '54.181054442798896', 'bar_calibration': '-0.019', 'bar_offset': '0.084', 'bar_reduction': '2', 'barometer': '30.243', 'beaufort': '0', 'cloudbase': '1611.3301439249644', 'dateTime': '1666724464', 'dayET': '0.0', 'dayRain': '0.01', 'dewpoint': '49.0', 'heatindex': '54.0', 'hourRain': '0.0', 'humidex': '56.257705080274974', 'inDewpoint': '48.98765246333121', 'inHumidity': '43.0', 'inTemp': '72.8', 'maxSolarRad': '490.69583276043966', 'outHumidity': '83.0', 'outTemp': '54.3', 'pressure': '29.736', 'pressure_raw': '29.652', 'rain': '0.0', 'rain15': '0.0', 'rain24': '0.03', 'rainRate': '0.0', 'stormRain': '0.08', 'stormStart': '1666594800', 'trendIcon': '-60', 'usUnits': '1', 'windchill': '54.0', 'windDir': 'None', 'windGust': '0.0', 'windGust10': '3.0', 'windGustDir': 'None', 'windGustDir10': '202.0', 'windrun': 'None', 'windSpeed': '0.0', 'windSpeed2': '0.0', 'windSpeed10': '0.7'
LOOP:   2022-10-25 12:01:06 PDT (1666724466) 'altimeter': '30.23912900445008', 'appTemp': '54.181054442798896', 'barometer': '30.243', 'beaufort': '0', 'cloudbase': '1611.3301439249644', 'consBatteryVoltage': '3.97', 'dateTime': '1666724466', 'dayET': '0.0', 'dayRain': '0.01', 'dewpoint': '49.24734736673015', 'extraAlarm1': '0', 'extraAlarm2': '0', 'extraAlarm3': '0', 'extraAlarm4': '0', 'extraAlarm5': '0', 'extraAlarm6': '0', 'extraAlarm7': '0', 'extraAlarm8': '0', 'forecastIcon': '6', 'forecastRule': '12', 'heatindex': '53.331', 'humidex': '56.257705080274974', 'inDewpoint': '48.98765246333121', 'inHumidity': '43.0', 'insideAlarm': '0', 'inTemp': '72.8', 'maxSolarRad': '490.7270534665849', 'monthET': '0.0', 'monthRain': '0.53', 'outHumidity': '83.0', 'outsideAlarm1': '0', 'outsideAlarm2': '0', 'outTemp': '54.3', 'pressure': '29.734914103659314', 'rain': '0.0', 'rainAlarm': '0', 'rainRate': '0.0', 'soilLeafAlarm1': '0', 'soilLeafAlarm2': '0', 'soilLeafAlarm3': '0', 'soilLeafAlarm4': '0', 'stormRain': '0.08', 'stormStart': '1666594800', 'sunrise': '1666708800', 'sunset': '1666746720', 'txBatteryStatus': '0', 'usUnits': '1', 'windchill': '54.3', 'windDir': 'None', 'windGust': '0.0', 'windGustDir': 'None', 'windrun': 'None', 'windSpeed': '0.0', 'windSpeed10': '1.0', 'yearET': '0.0', 'yearRain': '0.53'`

Is it just that when it goes to write the archive it's just doing so based on the "last LOOP" it received, and "50%" of the time that LOOP is the "wrong" one?

tkeffer, 2022-10-26:
You are on to something. I was able to replicate the problem.

When the driver pulls an archive record out of the logger, it then procedes to include the last value for `txBatteryStatus` in it. This "feature" dates waaay back (over 9 years ago), from when WeeWX did not have accumulators. It is no longer necessary.

Fixed in commit https://github.com/weewx/weewx/commit/dcfe5fcbed52f54318e458b13376be9c3866abb8

If you want a fix, replace your copy of the vantage driver with this one:
https://raw.githubusercontent.com/weewx/weewx/dcfe5fcbed52f54318e458b13376be9c3866abb8/bin/weewx/drivers/vantage.py

closed 2022-10-26 by tkeffer

staze, 2022-10-26:
Looking good, thank you Tom!