weewx/weewx #806: Keys `config_path` and `entry_path` are written `weewx.conf` when upgrading

tkeffer, 2022-11-01:
A glitch. Because this implementation depends on adding keys `config_path` and `entry_path` to `config_dict`, when it comes time to do an upgrade  they are written out to `weewx.conf`.

You end up with the head of the file looking like:

```ini
...
# Do not modify this. It is used when installing and updating weewx.
version = 4.9.2a1
config_path = /home/weewx/weewx.conf
entry_path = /home/weewx/bin/wee_config
...
```

_Originally posted by @tkeffer in https://github.com/weewx/weewx/issues/705#issuecomment-1298770323_

tkeffer, 2022-11-02:
Fixed in commit https://github.com/weewx/weewx/commit/5fa664890f90cd23c33f83b7fecbb3f2a704de82

closed 2022-11-02 by tkeffer

KruseLuds, 2022-11-14:
Hi Tom, the work you do is amazing! I am using weewx to send data all over the place, but have suddently run into two issues and was wondering if you or someone can point me in the right direction? 

1. This happened shortly after I upgraded to 4.9.1 - only one of the places where I send data - WindGuru - keeps getting empty restful calls - but none of the other places I send data! I even spoke with the WindGur folks they did see my logging in with no issue but all my calles to them are miaaing any data (weird): https://groups.google.com/g/weewx-user/c/UR52ZAlBmNc/m/PnAbk4q8BQAJ

2. I keep getting errors from WOW quite often when I send them data - the return message from them is that it is being sent too often. I thionk they hgave changed their side so they want data only every 10 minutes instead of 5. Where do I make that adjustment?

Thank you and my apologies if there is not the correct place -

chaunceygardiner, 2022-11-14:
The WOW interval could be changed to 10m with a post_interval = 600, but WOW’s doc says the following (note: AWS is Automatic Weather Station in this context):Photoswow.metoffice.gov.ukHow frequently should I upload observations from my AWS?It is recommended that the interval between automatic observations from your AWS is at least 5 minutes.

KruseLuds, 2022-11-14:
> Kruse, you are not going to get a faster response by starting lots of threads, nor posting to an issues thread that is completely unrelated to your problem. You're just going to confuse and frustrate people.
>
> Please stick to one thread and have some patience. It may take some time. We are all volunteers here.

Will do, sorry

 

-Kruse

📱(M):  <tel:+1.201.925.4410> +1.201.925.4410

📞(O):  <tel:+1.201.857.8307> +1.862.308.7040

📞(H):  <tel:+1.201.857.8307> +1.201.857.8307

📠(F):  <tel:+1.201.857.7718> +1.201.857.7188
Current Weather and Forecast <https://www.windy.com/40.980/-74.109?40.743,-74.107,9,i:pressure,m:eNsad5E>