weewx/weewx #808: Invalid default format strings for contexts other than 'current'

roe-dl, 2022-11-05:
In `units.py` from line 748 on it says:
```
        elif val_t[2] == "group_deltatime":
            # Get a delta-time format string. Use a default if the user did not supply one:
            if useThisFormat is None:
                # For group_deltatime formatting, the default context cannot be 'current'.
                # Change it to something sensible.
                if context == 'current':
                    context = 'delta_time'
                format_string = self.time_format_dict.get(context, DEFAULT_DELTATIME_FORMAT)
            else:
                format_string = useThisFormat
            # Now format the delta time, using the function delta_secs_to_string:
            val_str = self.delta_secs_to_string(val_t[0], format_string)
            addLabel = False
```

`time_format_dict` contains format strings like `%X` or `%x %X` and the like for the different contexts. Those format strings are not appropriate for the `delta_secs_to_string()` function, which requires format strings like `%(hour)d %(hour_label)s, %(minute)d %(minute_label)s`.

tkeffer, 2022-11-05:
The group `group_deltatime` uses a different set of contexts.

```python
        brief_delta = "%(minute)d%(minute_label)s, %(second)d%(second_label)s"
        short_delta = "%(hour)d%(hour_label)s, %(minute)d%(minute_label)s, %(second)d%(second_label)s"
        long_delta  = "%(day)d%(day_label)s, %(hour)d%(hour_label)s, %(minute)d%(minute_label)s"
        delta_time  = "%(day)d%(day_label)s, %(hour)d%(hour_label)s, %(minute)d%(minute_label)s"
```

See `[[TimeFormats]]` in `defaults.py`.
https://github.com/weewx/weewx/blob/2639c6ca462b280e3da907d23d32c75d3f23dd11/bin/weewx/defaults.py#L213

roe-dl, 2022-11-06:
That ist an example:
```
<p>$day.sunshineDur.sum</p>
```

The resulting syslog messages are:
```
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: Evaluation of template /etc/weewx/skins/Seasons-de/test.html.tmpl failed with exception '<class 'TypeError'>'
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: **** Ignoring template /etc/weewx/skins/Seasons-de/test.html.tmpl
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: **** Reason: %X format: an integer is required, not dict
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****  Traceback (most recent call last):
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****    File "/usr/share/weewx/weewx/cheetahgenerator.py", line 348, in generate
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****      unicode_string = compiled_template.respond()
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****    File "cheetah__etc_weewx_skins_Seasons_de_test_html_tmpl_1667721018_940427_49359.py", line 96, in respond
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****    File "/usr/share/weewx/weewx/cheetahgenerator.py", line 823, in filter
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****      filtered = six.text_type(val)
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****    File "/usr/share/weewx/weewx/tags.py", line 461, in __str__
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****      return str(vh)
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****    File "/usr/share/weewx/weewx/units.py", line 1016, in __str__
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****      s = self.toString()
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****    File "/usr/share/weewx/weewx/units.py", line 1009, in toString
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****      s = self.formatter.toString(self.value_t, self.context, addLabel=addLabel,
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****    File "/usr/share/weewx/weewx/units.py", line 695, in toString
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****      s = self._to_string(val_t, context, addLabel, useThisFormat, None_string, localize)
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****    File "/usr/share/weewx/weewx/units.py", line 759, in _to_string
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****      val_str = self.delta_secs_to_string(val_t[0], format_string)
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****    File "/usr/share/weewx/weewx/units.py", line 807, in delta_secs_to_string
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****      ans = locale.format_string(label_format, etime_dict)
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****    File "/usr/lib/python3.8/locale.py", line 223, in format_string
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****      new_val.append(_format(perc.group(), val, grouping, monetary))
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****    File "/usr/lib/python3.8/locale.py", line 187, in _format
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****      formatted = percent % value
Nov  6 08:55:18 LokalWiki weewx[497241] ERROR weewx.cheetahgenerator: ****  TypeError: %X format: an integer is required, not dict
```

So I guess, in that case the context is 'day'.

tkeffer, 2022-11-06:
Thanks for the example.

This is going to take a rethink of how `ValueHelper` works.

tkeffer, 2022-11-06:
Fixing this required changing how delta times are handled. Your example now works.

See commit https://github.com/weewx/weewx/commit/209fd340267f5b3af8fb50ddfe8d844fa362cfdc

Be sure to read the Upgrade Guide. This involved a modest breaking change.

closed 2022-11-06 by tkeffer