weewx/weewx #819: Drivers support on Hardware Comparison charts

nmarie276, 2022-12-28:
Hello.

WeeWX is able to deal with dozens of weather station models, but the driver for each hardware type is not at the same support level. For my personal example, I used a FineOffset WH3080 for many years. The *foshk* driver is **perfectly stable** (probably because it's one of the most used weather station with WeeWX). When it failed, I bought a WH2315 weather station, mainly because I was able to use the new weather sensors with the previous receiver. Before buying it, I looked for a *supported* weather station from https://www.weewx.com/hwcmp.html.
But I tried to use the new screen, but the *wh23xx* driver is a **3rd party supported driver**, Moreover, it doesn't support all the features I used from foshk driver (for example the memory backfill during startup).
To help developers to improve the driver quality, why not add a new **driver status** field in the hardware comparison webpage? Moreover, If we want to merge this driver to the mainline Weewx tree, it can make sense to define the acceptance criteria (after all, some drivers has already been deprecated due to lack of maintenance / support / etc).
Maybe these criteria are written in a documentation I didn't know, in this case, I apologize. But there is not a support level for each driver like:
* Mainline.
* 3rd party.
* Need maintenance.
* Deprecated.

What do you think of this documentation enhancement?

Cheers.

tkeffer, 2022-12-28:
Sounds like a great idea, but who is going to do it? It would require direct or indirect access to a lot of different hardware types, plus a lot of management time to keep all the statuses up to date.

Without a volunteer, the idea is unlikely to go anywhere.

nmarie276, 2022-12-29:
@tkeffer, this software is supported by its community so, individually, each concerned member may take some minutes (really) to update a markdown table.
For the driver support status, it can be quite easy:
|Hardware|Driver|Support status|
|---|---|---|
|FineOffset|fousb|supported (since it's in WeeWX repo)|
|WH2315|wh23xx|3rd-party (since it's not in repo)|
|LaCrosse WS23xx|xxx|deprecated (following the 4.2.0 release note)|

I agree the "need maintenance" status may be in the grey zone, but we can determine that, since these users are still using a quite old WeeWX version, this may be due to a non-working weather station if they upgrade.

I seen the "hardware comparison" chart is maintained using a Google group discussion. I think that, if I took the responsability to open a feature request or a bug to the driver maintainer, I'm responsible to follow the issue, so I can fill the data into the Google Group channel, or a wiki page within the Github project (even if I think having several data sources is never a good idea).

Moreover, following the Showcase webpage, we are hundreds of users who register our station, we have the relevant hardware, we use it from day to day. So we have in mind what we expect from our hardware (for example, the chart allow us to compare the hardware accuracy). So, like above, to my part:
## Hardware
Driver
- Status
- [x] Data Reception
- [x] Memory Backfill
- [x] Time Sync
- [ ] Pairing Reset
- [ ] whatever

## FineOffset
Driver: fousb
- Status: Mainline
- [x] Data Reception
- [x] Memory Backfill
- [x] Time Sync
- [ ] Pairing Reset
- [ ] whatever

## WH2315
Driver: wh23xx
- Status: 3rd-party
- [x] Data Reception
- [ ] Memory Backfill
- [ ] Time Sync
- [ ] Pairing Reset

## LaCrosse WS23xx
Driver: xxx
- Status: Deprecated (Since 4.2.0)
- [x] Data Reception
- [x] Memory Backfill
- [] Time Sync
- [ ] Pairing Reset
- [ ] whatever

Etc.

This can be our non-dev 2-cent contribution to help new user to choose their hardware.

tkeffer, 2022-12-29:
Looks good. Want to format a wiki page and get started?

matthewwall, 2022-12-29:
imho, there is no such thing as deprecated.  if you have the hardware, you need the driver.

for someone buying new hardware, a list of hardware (and driver) capabilities is, indeed, helpful.  i tried to do that with the hardware comparison page:

http://weewx.com/hwcmp.html

another approach would be to bring the information in that page up to date, and add some sub-rows to the 'weewx driver' row.  those rows would enumerate the features available in weewx, such as backfill and time sync.

we wrote the drivers to take advantage of as many hardware features as possible, but there are cases (e.g., acurite 5in1) where a feature was too buggy or poorly implemented in hardware to enable (in the case of the 5in1, there is a logger, but it has few records and is extremely difficult to extract data from it).

matthewwall, 2022-12-29:
regarding the wh23xx driver, i'm pretty sure i implemented backfill in that driver.  is it not working?

in my case, the fine offset sensors died *much* more frequently than the consoles.  i typically had to replace wind cups at least once per year (they would crack off after uv exposure) and the anemometer bearings at least every other year (they sieze up).  and the range was not as good for the console as with the sdr usb dongle.

generally speaking, drivers do not need much work once the basic features are in place - only to keep up with weewx api changes, or quirks in hardware behavior that is apparent only after many users and many years of usage.

nmarie276, 2022-12-30:
Hello @matthewwall.

Your "bullet-points" approach in a in new row looks very conveniant, and avoid to create new "source page". In this case, I will try to join the Google group, since it's the reference to update the Hardware Comparison chart. My other point was how to "mainline" a driver: for example, you created the *wh23xx* and *weewx-sdr* drivers, but didn't merged them. Why:
* You're not satisfied about quality, feature?
* There is not much interest regarding how many users need it?
* Any other reason?
 
Regarding the *wh23xx* driver, since I use Python3, I used a fork, not your one (I seen some pull request on your repo, but not merged). I will make some new tests later, or switch to the *weewx-sdr* driver. Anyway, I don't want to drift from the original subject.

Regards.

matthewwall, 2022-12-31:
> My other point was how to "mainline" a driver: for example, you created the 
> wh23xx and weewx-sdr drivers, but didn't merged them into the weewx repo.

i think tom's intent is to remove as many things as possible from the weewx core, and make it easy to install extensions.  the problem with pushing drivers into the core is that then the core must be updated as the drivers change.  when there are many firmware changes (e.g., gw1000 driver) or protocol changes (e.g., sdr driver), then the driver changes much more often than the weewx core changes, so keeping those drivers as extensions makes sense.

at the other extreme is hardware that never changes, so the driver never has to change.  this is the case for currently manufactured hardware, as well as products that are no longer manufactured, but still very common.

we are adding a feature to wee_extension that makes it possible to install an extension directly from a git (or any) repository, or an arbitrary web site.  this will make it easier to manage extensions.

> Regarding the wh23xx driver, since I use Python3, I used a fork, not your one 
> (I seen some pull request on your repo, but not yet merged). I will make some 
> new tests later, or switch to the weewx-sdr driver.

my apologies - i am rather behind on updates to my weewx contributions.

nmarie276, 2023-01-15:
Hello.

OK. Following what your explain, not integrating the drivers within the WeeWX core actually makes sense. And your idea to install extension from an URL is a real enhancement to this point.
As I'll start a discussion into Google group to add the "features" row into the hwcmp webpage.

Regards.

closed 2023-01-15 by nmarie276