weewx/weewx #833: for aggregation of `wind` the reading of `windGust` is used for `max` but `windSpeed` for `maxtime`

roe-dl, 2023-01-19:
In `xtypes.py` the derived observation type `wind` is replaced by `windGust` for the aggregation type `max` and by `windSpeed` for other aggregation types (see line 311). This means `...wind.max` is based on observation type `windGust` but `...wind.maxtime` is based on `windSpeed`. Is that intended?

tkeffer, 2023-01-19:
You have an extraordinary eye for catching these things. I guess we want

``` python
        if obs_type == 'wind':
            sql_type = 'windGust' if aggregate_type in {'max', 'maxtime'} else 'windSpeed'
        else:
            sql_type = obs_type
```

roe-dl, 2023-01-19:
> You have an extraordinary eye for catching these things. I guess we want

Sometimes people called me autistic. May be they are right a little bit. But in this case it was not me but @hoetzgit who found out.

roe-dl, 2023-01-20:
I thought about other aggregation types that may be affected, but I found none. So I guess I send a PR about this.

closed 2023-01-20 by tkeffer