weewx/weewx #836: Default value for MinMax barometer/pressure can lead to missing data

jocelynj, 2023-01-22:
I've setup a weather station in the mountains, at 1000 meter of elevation, and found out that some pression values are not imported. In fact, these values are lower than the minimum value allowed by the default configuration, which is 26 inHg, so 880 mbar. But I have pressure as low as 862 mbar, which were removed when imported in database.

Would it make sense to lower the minimum value in the default weewx.conf file, or change it during configuration, by setting it from the specified altitude?

tkeffer, 2023-01-22:
Hello.

The issues list is for bug tracking and reporting.

The best place to get answers for support questions is the [weewx-user](https://groups.google.com/g/weewx-user) mail group.

closed 2023-01-22 by tkeffer

jocelynj, 2023-01-22:
Thanks - found a link to https://github.com/weewx/weewx/wiki/Barometer,-pressure,-and-altimeter on this mail group, which explains that I incorrectly configured the pression from the station with `barometer` instead of `pressure`.