weewx/weewx #845: wee_device for vantage doesn't set transmitter 1 properly

gohlhausen, 2023-02-15:
If I use `wee_device --set-transmitter-type=1,10`, it doesn't set the type.
It works on all other transmitters (2-8), but not 1.
For 1, It stays iss.

Using the davis software, I was able to configure 1 as none.
Running wee_device --info, I can see that it is set properly now.

Using Vantage driver version 3.5.2 (weewx.drivers.vantage)

tkeffer, 2023-02-16:
I tried this on one of my consoles and it worked fine. 

Could you please post the results of

    wee_device --info

Also, set `debug=1`, run `wee_device --set-transmitter-type1,10`, then post the log.

gohlhausen, 2023-02-17:
Last login: Thu Feb 16 18:51:27 2023 from 192.168.1.4
root@WeatherPleak:~# wee_device --info
Using configuration file /etc/weewx/weewx.conf
Using Vantage driver version 3.5.2 (weewx.drivers.vantage)
Unable to wake up console... sleeping
Unable to wake up console... retrying
Querying...
Davis Vantage EEPROM settings:

    CONSOLE TYPE:                   Vantage Pro2

    CONSOLE FIRMWARE:
      Date:                         Jun  3 2013
      Version:                      3.15

    CONSOLE SETTINGS:
      Archive interval:             60 (seconds)
      Altitude:                     1760 (foot)
      Wind cup type:                large
      Rain bucket type:             0.01 inches
      Rain year start:              1
      Onboard time:                 2023-02-17 08:34:06

    CONSOLE DISPLAY UNITS:
      Barometer:                    inHg
      Temperature:                  degree_F
      Rain:                         inch
      Wind:                         mile_per_hour

    CONSOLE STATION INFO:
      Latitude (onboard):           +33.5
      Longitude (onboard):          -99.7
      Use manual or auto DST?       AUTO
      DST setting:                  N/A
      Use GMT offset or zone code?  ZONE_CODE
      Time zone code:               6
      GMT offset:                   N/A
      Temperature logging:          LAST
      Retransmit channel:           OFF (0)

    TRANSMITTERS:
      Channel   Receive   Repeater  Type
         1      inactive    none    (N/A)
         2      inactive    none    (N/A)
         3      inactive    none    (N/A)
         4      inactive    none    (N/A)
         5      active      none    iss
         6      inactive    none    (N/A)
         7      inactive    none    (N/A)
         8      inactive    none    (N/A)

    RECEPTION STATS:
      Total packets received:       10292
      Total packets missed:         154
      Number of resynchronizations: 5
      Longest good stretch:         1468
      Number of CRC errors:         3

    BAROMETER CALIBRATION DATA:
      Current barometer reading:    30.585 inHg
      Altitude:                     1760 feet
      Dew point:                    19 F
      Virtual temperature:          34 F
      Humidity correction factor:   0.6
      Correction ratio:             1.069
      Correction constant:          +0.000 inHg
      Gain:                         0.000
      Offset:                       9.000

    OFFSETS:
[... 330 lines cut ...]
Unable to wake up console... sleeping
Unable to wake up console... retrying
    BAROMETER CALIBRATION DATA:
      Current barometer reading:    30.587 inHg
      Altitude:                     1760 feet
      Dew point:                    255 F
      Virtual temperature:          34 F
      Humidity correction factor:   2.7
      Correction ratio:             1.069
      Correction constant:          +0.000 inHg
      Gain:                         0.000
      Offset:                       9.000

Unable to wake up console... sleeping
Unable to wake up console... retrying
Unable to wake up console... sleeping
Unable to wake up console... retrying
    OFFSETS:
      Wind direction:               +0 deg
      Inside Temperature:           +0.0 F
      Inside Humidity:              +0 %
      Outside Temperature:          +0.0 F
      Outside Humidity:             +0 %

root@WeatherPleak:~#

gohlhausen, 2023-02-17:
I am also unable to set 5 to iss.
Attached is the syslog for a set=1,10 that I ran after enabling debug=1
[syslog.txt](https://github.com/weewx/weewx/files/10767918/syslog.txt)

THis is a Vantage Envoy 6316.  From what I can tell, it thinks it is a Vantage Pro2.  Firmware responds as 6312.

tkeffer, 2023-02-17:
You must stop `weewxd` before running `wee_device`. Otherwise, they compete for the same device.

gohlhausen, 2023-02-17:
If I set transmitter 1 to anything other than iss, it works.
I am then able to set 5 to iss.
 TRANSMITTERS:
      Channel   Receive   Repeater  Type
         1      active      none    leaf_soil
         2      inactive    none    (N/A)
         3      inactive    none    (N/A)
         4      inactive    none    (N/A)
         5      active      none    iss
         6      inactive    none    (N/A)
         7      inactive    none    (N/A)
         8      inactive    none    (N/A)

With the service stopped, it acts the same way.
I can set transmitter 1 to anything other than none.
If transmitter 1 is iss, I can't set the type of any other transmitter.
If I set transmitter 1 to 8, I can then set transmitter 5 to iss

    TRANSMITTERS:
      Channel   Receive   Repeater  Type
         1      inactive    none    iss
         2      inactive    none    (N/A)
         3      inactive    none    (N/A)
         4      inactive    none    (N/A)
         5      active      none    (N/A)
         6      inactive    none    (N/A)
         7      inactive    none    (N/A)
         8      inactive    none    (N/A)

    TRANSMITTERS:
      Channel   Receive   Repeater  Type
         1      active      none    leaf_soil
         2      inactive    none    (N/A)
         3      inactive    none    (N/A)
         4      inactive    none    (N/A)
         5      active      none    iss
         6      inactive    none    (N/A)
         7      inactive    none    (N/A)
         8      inactive    none    (N/A)

gohlhausen, 2023-02-17:
Feb 17 10:30:16 WeatherPleak systemd[1]: weewx.service: Deactivated successfully.
Feb 17 10:30:16 WeatherPleak systemd[1]: Stopped LSB: weewx weather system.
Feb 17 10:30:21 WeatherPleak wee_device[15226] DEBUG weewx.drivers.vantage: Driver version is 3.5.2
Feb 17 10:30:21 WeatherPleak wee_device[15226] DEBUG weewx.drivers.vantage: Option loop_request=1
Feb 17 10:30:21 WeatherPleak wee_device[15226] DEBUG weewx.drivers.vantage: Opened up serial port /dev/ttyUSB0; baud 19200; timeout 4.00
Feb 17 10:30:22 WeatherPleak wee_device[15226] DEBUG weewx.drivers.vantage: Successfully woke up Vantage console
Feb 17 10:30:22 WeatherPleak wee_device[15226] DEBUG weewx.drivers.vantage: Hardware type is 16
Feb 17 10:30:22 WeatherPleak wee_device[15226] DEBUG weewx.drivers.vantage: ISS ID is 1
Feb 17 10:30:22 WeatherPleak wee_device[15226] DEBUG weewx.drivers.vantage: Hardware name: Vantage Pro2
Feb 17 10:30:25 WeatherPleak wee_device[15226] DEBUG weewx.drivers.vantage: Successfully woke up Vantage console
Feb 17 10:30:25 WeatherPleak wee_device[15226] DEBUG weewx.drivers.vantage: ISS ID is 1
Feb 17 10:30:25 WeatherPleak wee_device[15226] INFO weewx.drivers.vantage: Transmitter type for channel 1 set to 0 (iss), repeater: none, active
Feb 17 10:30:29 WeatherPleak wee_device[15228] DEBUG weewx.drivers.vantage: Driver version is 3.5.2
Feb 17 10:30:29 WeatherPleak wee_device[15228] DEBUG weewx.drivers.vantage: Option loop_request=1
Feb 17 10:30:29 WeatherPleak wee_device[15228] DEBUG weewx.drivers.vantage: Opened up serial port /dev/ttyUSB0; baud 19200; timeout 4.00
Feb 17 10:30:30 WeatherPleak wee_device[15228] DEBUG weewx.drivers.vantage: Successfully woke up Vantage console
Feb 17 10:30:30 WeatherPleak wee_device[15228] DEBUG weewx.drivers.vantage: Hardware type is 16
Feb 17 10:30:30 WeatherPleak wee_device[15228] DEBUG weewx.drivers.vantage: ISS ID is 1
Feb 17 10:30:30 WeatherPleak wee_device[15228] DEBUG weewx.drivers.vantage: Hardware name: Vantage Pro2
Feb 17 10:30:30 WeatherPleak wee_device[15228] DEBUG weewx.drivers.vantage: Successfully woke up Vantage console
Feb 17 10:30:34 WeatherPleak wee_device[15228] message repeated 4 times: [ DEBUG weewx.drivers.vantage: Successfully woke up Vantage console]
Feb 17 10:30:39 WeatherPleak snapd[829]: storehelpers.go:769: cannot refresh: snap has no updates available: "bare", "chromium", "core18", "core20", "cups", "firefox", "gnome-3-38-2004", "gtk-common-themes", "hunspell-dictionaries-1-7-2004", "snap-store", "snapd"
Feb 17 10:30:39 WeatherPleak snapd[829]: autorefresh.go:551: auto-refresh: all snaps are up-to-date
Feb 17 10:30:41 WeatherPleak wee_device[15232] DEBUG weewx.drivers.vantage: Driver version is 3.5.2
Feb 17 10:30:41 WeatherPleak wee_device[15232] DEBUG weewx.drivers.vantage: Option loop_request=1
Feb 17 10:30:41 WeatherPleak wee_device[15232] DEBUG weewx.drivers.vantage: Opened up serial port /dev/ttyUSB0; baud 19200; timeout 4.00
Feb 17 10:30:42 WeatherPleak wee_device[15232] DEBUG weewx.drivers.vantage: Successfully woke up Vantage console
Feb 17 10:30:42 WeatherPleak wee_device[15232] DEBUG weewx.drivers.vantage: Hardware type is 16
Feb 17 10:30:42 WeatherPleak wee_device[15232] DEBUG weewx.drivers.vantage: ISS ID is 1
Feb 17 10:30:42 WeatherPleak wee_device[15232] DEBUG weewx.drivers.vantage: Hardware name: Vantage Pro2
Feb 17 10:30:45 WeatherPleak wee_device[15232] DEBUG weewx.drivers.vantage: Successfully woke up Vantage console
Feb 17 10:30:45 WeatherPleak wee_device[15232] DEBUG weewx.drivers.vantage: ISS ID is 1
Feb 17 10:30:45 WeatherPleak wee_device[15232] INFO weewx.drivers.vantage: Transmitter type for channel 1 set to 10 (none), repeater: none, inactive
Feb 17 10:30:47 WeatherPleak wee_device[15237] DEBUG weewx.drivers.vantage: Driver version is 3.5.2
Feb 17 10:30:47 WeatherPleak wee_device[15237] DEBUG weewx.drivers.vantage: Option loop_request=1
Feb 17 10:30:47 WeatherPleak wee_device[15237] DEBUG weewx.drivers.vantage: Opened up serial port /dev/ttyUSB0; baud 19200; timeout 4.00
Feb 17 10:30:48 WeatherPleak wee_device[15237] DEBUG weewx.drivers.vantage: Successfully woke up Vantage console
Feb 17 10:30:48 WeatherPleak wee_device[15237] DEBUG weewx.drivers.vantage: Hardware type is 16
Feb 17 10:30:48 WeatherPleak wee_device[15237] DEBUG weewx.drivers.vantage: ISS ID is 1
Feb 17 10:30:48 WeatherPleak wee_device[15237] DEBUG weewx.drivers.vantage: Hardware name: Vantage Pro2
Feb 17 10:30:48 WeatherPleak wee_device[15237] DEBUG weewx.drivers.vantage: Successfully woke up Vantage console
Feb 17 10:30:52 WeatherPleak wee_device[15237] message repeated 4 times: [ DEBUG weewx.drivers.vantage: Successfully woke up Vantage console]
Feb 17 10:32:29 WeatherPleak wee_device[15240] DEBUG weewx.drivers.vantage: Driver version is 3.5.2
Feb 17 10:32:29 WeatherPleak wee_device[15240] DEBUG weewx.drivers.vantage: Option loop_request=1
Feb 17 10:32:29 WeatherPleak wee_device[15240] DEBUG weewx.drivers.vantage: Opened up serial port /dev/ttyUSB0; baud 19200; timeout 4.00
Feb 17 10:32:29 WeatherPleak wee_device[15240] DEBUG weewx.drivers.vantage: Successfully woke up Vantage console
Feb 17 10:32:29 WeatherPleak wee_device[15240] DEBUG weewx.drivers.vantage: Hardware type is 16
Feb 17 10:32:29 WeatherPleak wee_device[15240] DEBUG weewx.drivers.vantage: ISS ID is 1
Feb 17 10:32:29 WeatherPleak wee_device[15240] DEBUG weewx.drivers.vantage: Hardware name: Vantage Pro2
[... cut ...]
Feb 17 10:33:59 WeatherPleak wee_device[15250] DEBUG weewx.drivers.vantage: Successfully woke up Vantage console
Feb 17 10:33:59 WeatherPleak wee_device[15250] DEBUG weewx.drivers.vantage: Hardware type is 16
Feb 17 10:33:59 WeatherPleak wee_device[15250] DEBUG weewx.drivers.vantage: ISS ID is 1
Feb 17 10:33:59 WeatherPleak wee_device[15250] DEBUG weewx.drivers.vantage: Hardware name: Vantage Pro2
Feb 17 10:34:00 WeatherPleak wee_device[15250] DEBUG weewx.drivers.vantage: Successfully woke up Vantage console

gohlhausen, 2023-02-17:
With the bogus sensor in transmitter 1, I am able to get packets from my iss on 5.
THis gets it working without having to move it to a PC.
There is still something hinky going on, but it is working for me now.
(I used an odroid C2 with ubuntu 20.4 and the ubuntu apt-get install method

    RECEPTION STATS:
      Total packets received:       129
      Total packets missed:         0
      Number of resynchronizations: 0
      Longest good stretch:         129
      Number of CRC errors:         0

tkeffer, 2023-02-17:
I really need to know what commands you are using so I can reproduce the problem. Your original problem was that

    wee_device --set-transmitter-type=1,10

did not work. Is this still the problem? If so, I am unable to reproduce it.

Stop `weewxd` the run the command again. Does it still not work?

gohlhausen, 2023-02-17:
THe problem still exists and it still happens when weewx is stopped.

This is where I start
    TRANSMITTERS:
      Channel   Receive   Repeater  Type
         1      inactive    none    iss
         2      inactive    none    (N/A)
         3      inactive    none    (N/A)
         4      inactive    none    (N/A)
         5      inactive    none    (N/A)
         6      inactive    none    (N/A)
         7      inactive    none    (N/A)
         8      inactive    none    (N/A)

These are the commands I did:
wee_device --set-transmitter-type=1,10 -y
wee_device --set-transmitter-type=5,0 -y
wee_device --info

When I do those commands, I get these transmitter values
    TRANSMITTERS:
      Channel   Receive   Repeater  Type
         1      inactive    none    iss
         2      inactive    none    (N/A)
         3      inactive    none    (N/A)
         4      inactive    none    (N/A)
         5      active      none    (N/A)
         6      inactive    none    (N/A)
         7      inactive    none    (N/A)
         8      inactive    none    (N/A)

5 isn't getting set to iss, but it getting set active
1 isn't getting set to none and it is still active.

If I set 1 to something else, I can set 5.

wee_device --set-transmitter-type=1,8 -y
wee_device --set-transmitter-type=5,0 -y
wee_device --info

   TRANSMITTERS:
      Channel   Receive   Repeater  Type
         1      active      none    leaf_soil
         2      inactive    none    (N/A)
         3      inactive    none    (N/A)
         4      inactive    none    (N/A)
         5      active      none    iss
         6      inactive    none    (N/A)
         7      inactive    none    (N/A)
         8      inactive    none    (N/A)

tkeffer, 2023-02-17:
Finally! I'm able to reproduce the results. Let me work on this.

chaunceygardiner, 2023-02-18:
IIRC, you can’t have two specified as iss—even if one is inactive.  That’s what I remember when I needed to specify an iss on a different channel than 1.  I *presume* it’s the console itself that doesn’t allow it.

tkeffer, 2023-02-18:
@chaunceygardiner The issue does not involve setting two ISS simulataneously. It involves trying to shut an ISS on channel 1 off *then* turn on another channel.

all: from what I can tell, if you shut off channel 1 by either pushing the buttons on the console, or by doing `--set-transmit=1,10`, it still shows up as an ISS, although inactive:

```
    TRANSMITTERS: 
      Channel   Receive   Repeater  Type
         1      inactive    none    iss 
         2      inactive    none    (N/A) 
         3      inactive    none    (N/A) 
         4      inactive    none    (N/A) 
         5      inactive    none    (N/A) 
         6      inactive    none    (N/A) 
         7      inactive    none    (N/A) 
         8      inactive    none    (N/A) 
```
If you set a different ISS channel, either through the console, or using `--set-transmitter=2,0` then it disappears, and the ISS channel appears in the new place, but inactive:

```
   TRANSMITTERS: 
      Channel   Receive   Repeater  Type
         1      inactive    none    (N/A) 
         2      inactive    none    iss 
         3      inactive    none    (N/A) 
```

If you disable the new channel with `--set-transmitter=2,10` channel 1 ISS reappears:

```
   TRANSMITTERS: 
      Channel   Receive   Repeater  Type
         1      inactive    none    iss 
         2      inactive    none    (N/A) 
         3      inactive    none    (N/A) 
```

I'm watched the right bits go into the proper EEPROM byte, but when you read them back out, they've changed. 

I'll give Davis a call on Monday. I've been through this drill with them before, and they are very responsive, but it can be very hard to convey a subtle problem like this.

tkeffer, 2023-03-10:
I have not been able to make much progress with Davis support. It's a difficult problem to reproduce because of all the handshaking that goes on between computer and console. 

Still, I think things are a little better with the V5 driver, should you want to give it a try. It uses a different strategy for retrieving and setting the station information EEPROM. If you are using Python 3.7 or greater, it should be a drop-in replacement. 

https://raw.githubusercontent.com/weewx/weewx/V5/bin/weewx/drivers/vantage.py

gohlhausen, 2023-03-11:
Thanks.  I’ll give it a try.

tkeffer, 2023-05-01:
I seem to have inadvertently fixed the problem while fixing issue #863.

To appear in V5.

closed 2023-05-01 by tkeffer