weewx/weewx #848: diff calculations are off

JDJoe, 2023-03-01:
Version: 4.10.1  

```
          [[[daydiff]]]
             [[[[diff]]]]
         
                 data_type = inTemp - outTemp
                 label = Inside - Outside 

```

Inside temperature: +22.1 C
Outside temperature: -3.2 C

Graph plots ~ 35 C

tkeffer, 2023-03-01:
My guess is that your database is in US units, so the actual math being done is something like `72 - 26`, or about `46`. That's not `35`, but close --- seeing the actual plot would be helpful.

In a normal plot that does not use SQL math, the unit of the result is easy to determine: just look up the observation type. However, in this case, the observation type is `inTemp-outTemp`, which cannot be looked up.

So, no unit conversion to Celsius can be done and the results stay the same: about `35`.

It is a bug, but not one with an easy fix. 

However, there is a straightforward work around. Try this:

``` ini
          [[[daydiff]]]
             [[[[diff]]]]
                 # Convert from US to Celsius
                 data_type = (inTemp - outTemp) / 1.8
                 label = Inside - Outside 
```

JDJoe, 2023-03-02:
Thanks. This seems to fix the problem for now. 

Question: Why is my db in F's and not C? 
I have everything configured to use C when I fist ran the setup. 
Weather station is WS90 (WittBoy).

tkeffer, 2023-03-02:
See http://www.weewx.com/docs/usersguide.htm#StdConvert

closed 2023-03-02 by tkeffer

JDJoe, 2023-03-05:
Hello. I have to reopen this. Division by 1.8 wont always work 

```
Outside Temperature 	-10.2°C
KH Temp 	-4.4°C

```
Graphs https://imgur.com/a/twoLr24

gjr80, 2023-03-06:
I've set up a VM running the simulator using US customary database units and displaying in Metric. Two plots are being generated; first plot shows `outTemp`, `inTemp` and `inTemp-outTemp` and second plot shows `outTemp`,`inTemp` and `(inTemp-outTemp)/1.8`.

The first plot displays incorrectly, `inTemp` and `outTemp` are both plotted correctly showing values in °C but `inTemp-outTemp` displays as values in °F. The plotted values are shown below, the unbracketed values are the actual values plotted, the bracketed values for `inTemp` and `outTemp` are the °F equivalents:
```
inTemp                  outTemp                (inTemp-outTemp)
-5.0000(-41.0000)       -35.9325(-96.6786)     55.6786
-5.0000(-41.0001)       -35.9079(-96.6343)     55.6342
-5.0003(-41.0005)       -35.8602(-96.5484)     55.5479
-5.0007(-41.0012)       -35.8132(-96.4638)     55.4625
-5.0013(-41.0024)       -35.7670(-96.3805)     55.3781
```

It's pretty clear from this data that when you plot in temperature units that are different to the database units and 'difference' temperatures are plotted as values using the database temperature units.

The second plot displays correctly, `inTemp` and `outTemp` are both plotted correctly showing values in °C as is `(inTemp-outTemp)/1.8`. The plotted values are shown below, again the unbracketed values are the actual values plotteds, the bracketed values for `inTemp` and `outTemp` are the °F equivalents:
```
inTemp                  outTemp                (inTemp-outTemp)
-5.0000(-41.0000)       -35.9325(-96.6786)     30.9325
-5.0000(-41.0001)       -35.9079(-96.6343)     30.9079
-5.0003(-41.0005)       -35.8602(-96.5484)     30.8599
-5.0007(-41.0012)       -35.8132(-96.4638)     30.8125
-5.0013(-41.0024)       -35.7670(-96.3805)     30.7656
```

I also set the simulator to provide temperatures above zero, and as expected the '/1.8 difference' plot displayed correctly and 'non/1.8 difference' plot did not.

I think it's exactly as Tom called it; it's a bug but it's not going to be easy to fix. Adding some suitable maths to the plot definition is a workaround.

JDJoe, 2023-03-06:
out -10.2 C = 13.64 F

KH -4.4 C =  24.08 F

24.08 and 13.64 difference is 10.44 and dividing this by 1.8 should plot 5.8 on the chart but as you can clearly see, it is not.

JDJoe, 2023-03-06:
Looks like I made a mistake :) 
My calculation for daily used wrong sensor - inTemp. Should have been extraTemp3.  

I also took a look at the dadabase and all temperatures are recorded F.  That sucks.

tkeffer, 2023-03-06:
Thanks, Gary.

Joe: see http://www.weewx.com/docs/customizing.htm#Changing_the_unit_system