weewx/weewx #849: wind.vecdir aggregation and null values for wind direction

Jterrettaz, 2023-03-02:
I observed that the **wind.vecdir** aggregation type return 90 degrees  if all archive records  within the aggregation period have a null windDir value.

I am running Weewx V 4.10.2 and my version of sqlite3 does not support math function.

With a Vantage Pro weather station, the windDir is null if windSpeed = 0 during the archive period.
So for exemple, the tag **$hour.wind.vecdir.raw** return 90 if all the windDir during the last hour were null.  I would expect to have a null value instead.

For exemple, here are charts of non aggregated wind direction values showing periods with null windDir:
23-02-2023:
[image]
25-02-2023:
[image]

in the following weekly chart, data are aggregated (60 min interval), and the 90 degree value corresponding to the periods with null windDir is clearly visible  : 
[image]

tkeffer, 2023-03-09:
I'm unable to replicate this, so I wanted to check with you that I've got the circumstances OK.

For the last 90 minutes of data, I set `windSpeed=0` and `windDir=0`.

Then I ran wee_reports to evaluate 

```
<p>\$hour.wind.vecdir.raw = $hour.wind.vecdir ($hour.wind.vecdir.raw) </p>
```

I got the results I would expect:
```
$hour.wind.vecdir.raw = N/A ()
```

Did I miss something?

tkeffer, 2023-03-09:
My apologies. I got my response wrong. What I meant to say is that I set `windDir` to `null`. Here's the exact sqlite statement I used:

``` sql
update archive set windSpeed=0.0, windDir=null where dateTime >= 1678316820;
```

Jterrettaz, 2023-03-09:
OK.  
Since my sqlite3 does not support math function, if I modify the function  weeewx.xtypes.get_wind_aggregate_long as follows , by adding a nowindDir variable ans some related tests: 

        .....
        xsum = 0.0
        ysum = 0.0
        sumtime = 0.0
        nowindDir=True
        for row in db_manager.genSql(sql_stmt):
           if row[1] is not None:
               sumtime += row[0]
               if row[2] is not None:
                   nowindDir = False
                   xsum += row[0] * row[1] * math.cos(math.radians(90.0 - row[2]))
                   ysum += row[0] * row[1] * math.sin(math.radians(90.0 - row[2]))
       
        if not sumtime or nowindDir:
           value = None
        elif aggregate_type == 'vecdir':
        .....

my problem is resolved.
In that case , the tag $hour($hours_ago=xxx).wind.vecdir.raw is returning  "" (empty string), when all the windDir are null within the aggregation interval.  Otherwise I will have the "90" value returned by the tag .

Would be happy to understand if and in that case why this problem could be specific to my settings

tkeffer, 2023-03-09:
I've forced sqlite3 not to recognize the math functions, and now I am able to replicate your results.

The problem is that wind direction is being calculated as

``` python
            deg = 90.0 - math.degrees((math.atan2(ysum, xsum)))
```

If `windDir` is null the whole time, then both `xsum` and `ysum` will be zero, which causes `atan2` to return zero.

To fix, change the line
```python
        if not sumtime:
            value = None
```
to read
``` python
        if not sumtime or (xsum == 0.0 and ysum == 0.0):
            value = None
```

Commit https://github.com/weewx/weewx/commit/57921f544a07d8285d8cd7adda456195745ef6d9, to appear in V5.

Thanks for your patience!

closed 2023-03-09 by tkeffer