weewx/weewx #852: $current.dateTime always outputs UTC

USA-RedDragon, 2023-03-05:
Hello,

I am using WeeWX along with MQTT (https://github.com/USA-RedDragon/weewxMQTT) to ingress data from a weather station. I tried placing ISO timestamps with a local timezone and Unix epoch timestamps in the `dateTime` data, but the output never changes from UTC.

Running `timedatectl` shows that the timezone is set correctly on the machine:

```
reddragon@ki5vmf-server:~$ timedatectl
               Local time: Sun 2023-03-05 15:07:06 CST
           Universal time: Sun 2023-03-05 21:07:06 UTC
                 RTC time: Sun 2023-03-05 21:07:06
                Time zone: America/Chicago (CST, -0600)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
```

The data payload WeeWX receives: 

```
{"outTempBatteryStatus": 0, "outTemp": 26.7, "outHumidity": 26, "windDir": 260, "windSpeed": 4.5, "windGust": 5.9, "rain": 5.7, "UV": 3.9, "dateTime": 1678050483, "heatindex": 26.7, "windchill": 26.7, "dewpoint": 5.6, "frostpoint": 1.9, "inTempBatteryStatus": 0, "inTemp": 21.6, "inHumidity": 38}
```

The log shows the local timezone:

```
Mar 05 15:09:41 ki5vmf-server python3[10679]: weewx[10679] INFO weewx.manager: Added record 2023-03-05 21:09:36 CST (1678072176) to database 'weewx.sdb'
Mar 05 15:09:41 ki5vmf-server python3[10679]: weewx[10679] INFO weewx.manager: Added record 2023-03-05 21:09:36 CST (1678072176) to daily summary in 'weewx.sdb'
Mar 05 15:09:41 ki5vmf-server python3[10679]: weewx[10679] DEBUG weewx.reportengine: Running reports for latest time in the database.
Mar 05 15:09:41 ki5vmf-server python3[10679]: weewx[10679] DEBUG weewx.reportengine: Running report 'SeasonsReport'
Mar 05 15:09:41 ki5vmf-server python3[10679]: weewx[10679] DEBUG weewx.reportengine: Found configuration file /etc/weewx/skins/Seasons/skin.conf for report 'SeasonsReport'
Mar 05 15:09:41 ki5vmf-server python3[10679]: weewx[10679] DEBUG weewx.cheetahgenerator: Using search list ['weewx.cheetahgenerator.Almanac', 'weewx.cheetahgenerator.Current', 'weewx.cheetahgenerator.DisplayOptions', 'weewx.cheetahgenerator.Extras', 'weewx.cheetahgenerator.Gettext', 'weewx.cheetahgenerator.JSONHelpers', 'weewx.cheetahgenerator.PlotInfo', 'weewx.cheetahgenerator.SkinInfo', 'weewx.cheetahgenerator.Station', 'weewx.cheetahgenerator.Stats', 'weewx.cheetahgenerator.UnitInfo']
Mar 05 15:09:41 ki5vmf-server python3[10679]: weewx[10679] DEBUG weewx.manager: Daily summary version is 4.0
Mar 05 15:09:41 ki5vmf-server python3[10679]: weewx[10679] INFO weewx.cheetahgenerator: Generated 8 files for report SeasonsReport in 0.47 seconds
Mar 05 15:09:41 ki5vmf-server python3[10679]: weewx[10679] DEBUG weewx.manager: Daily summary version is 4.0
Mar 05 15:09:42 ki5vmf-server python3[10679]: weewx[10679] INFO weewx.imagegenerator: Generated 30 images for report SeasonsReport in 1.23 seconds
Mar 05 15:09:42 ki5vmf-server python3[10679]: weewx[10679] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx
Mar 05 15:09:42 ki5vmf-server python3[10679]: weewx[10679] DEBUG weewx.reportengine: Report 'SmartphoneReport' not enabled. Skipping.
Mar 05 15:09:42 ki5vmf-server python3[10679]: weewx[10679] DEBUG weewx.reportengine: Running report 'MobileReport'
Mar 05 15:09:42 ki5vmf-server python3[10679]: weewx[10679] DEBUG weewx.reportengine: Found configuration file /etc/weewx/skins/Mobile/skin.conf for report 'MobileReport'
Mar 05 15:09:42 ki5vmf-server python3[10679]: weewx[10679] DEBUG weewx.cheetahgenerator: Using search list ['weewx.cheetahgenerator.Almanac', 'weewx.cheetahgenerator.Current', 'weewx.cheetahgenerator.DisplayOptions', 'weewx.cheetahgenerator.Extras', 'weewx.cheetahgenerator.Gettext', 'weewx.cheetahgenerator.JSONHelpers', 'weewx.cheetahgenerator.PlotInfo', 'weewx.cheetahgenerator.SkinInfo', 'weewx.cheetahgenerator.Station', 'weewx.cheetahgenerator.Stats', 'weewx.cheetahgenerator.UnitInfo']
Mar 05 15:09:42 ki5vmf-server python3[10679]: weewx[10679] DEBUG weewx.manager: Daily summary version is 4.0
Mar 05 15:09:42 ki5vmf-server python3[10679]: weewx[10679] INFO weewx.cheetahgenerator: Generated 1 files for report MobileReport in 0.01 seconds
Mar 05 15:09:42 ki5vmf-server python3[10679]: weewx[10679] DEBUG weewx.manager: Daily summary version is 4.0
Mar 05 15:09:43 ki5vmf-server python3[10679]: weewx[10679] INFO weewx.imagegenerator: Generated 4 images for report MobileReport in 0.04 seconds
Mar 05 15:09:43 ki5vmf-server python3[10679]: weewx[10679] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx/mobile
Mar 05 15:09:43 ki5vmf-server python3[10679]: weewx[10679] DEBUG weewx.reportengine: Report 'StandardReport' not enabled. Skipping.
Mar 05 15:09:43 ki5vmf-server python3[10679]: weewx[10679] DEBUG weewx.reportengine: Report 'FTP' not enabled. Skipping.
Mar 05 15:09:43 ki5vmf-server python3[10679]: weewx[10679] DEBUG weewx.reportengine: Report 'RSYNC' not enabled. Skipping.
```

tkeffer, 2023-03-06:
Looks like something along the data chain is adding 6 hours to the timestamp.

Any particular reason why you suspect WeeWX rather than the MQTT driver?

USA-RedDragon, 2023-03-06:
Yep, dumb user here. Definitely the driver. It was overriding my timestamps

closed 2023-03-06 by USA-RedDragon