weewx/weewx #853: All time data in statistics

cincuranet, 2023-03-09:
Is there a reason why `$alltime` is not used in [statistics](https://github.com/weewx/weewx/blob/master/skins/Seasons/statistics.inc#L7) in _Seasons_ skin? For the separate page (not the index widget) it makes sense to me without any obvious downsides.

Would a PR be welcome?

cincuranet, 2023-03-11:
Would it make sense to you to have it under a flag in config file (disabled being default)?

tkeffer, 2023-03-11:
I think the flag would be more complicated than the necessary changes to `statistics.inc`. It's  really only two lines that need to be changed:

``` html
*** statistics.inc	2023-03-10 09:12:41.262442506 -0800
--- /home/weewx/skins/Seasons/statistics.inc	2023-03-11 05:29:01.367089674 -0800
***************
*** 4,10 ****
  #errorCatcher Echo
  #encoding UTF-8
  
! #set $time_tags = [$day, $week, $month, $year, $rainyear]
  
  ## Get the list of observations from the configuration file, otherwise fallback
  ## to a very rudimentary set of observations.
--- 4,10 ----
  #errorCatcher Echo
  #encoding UTF-8
  
! #set $time_tags = [$day, $week, $month, $year, $rainyear, $alltime]
  
  ## Get the list of observations from the configuration file, otherwise fallback
  ## to a very rudimentary set of observations.
***************
*** 31,36 ****
--- 31,37 ----
            <th class="hilo_month">$gettext("Month")</th>
            <th class="hilo_year">$gettext("Year")</th>
            <th class="hilo_rainyear">$gettext("Rain Year")</th>
+           <th class="hilo_year">$gettext("All time")</th>
          </tr>
  
  #for $x in $observations
```

What I would suggest is writing a very short wiki article about how to add `$alltime` statistics. Put it under the "How to" guides.

cincuranet, 2023-03-20:
Sure, I already changed my template. I was thinking more around average user and making it easier for them.