weewx/weewx #861: Partial deadlock: Hanging in futex()?

spacelama, 2023-03-26:
For a week now, but some weeks after my last upgrade from weewx 4.10.1 to 4.10.2,  weewx has been hanging for 20 minutes at a time, sometimes, after receiving data from my ecowitt GW1000.  `strace -f` reveals it hanging after these calls (the poll being the usual polling with 1 second timeout while it's waiting for data from the GW1000):

```
[pid 4096830] poll([{fd=4, events=POLLIN}], 1, 500) = 0 (Timeout)
[pid 4096830] poll([{fd=4, events=POLLIN}], 1, 500) = 0 (Timeout)
[pid 4096830] poll([{fd=4, events=POLLIN}], 1, 500) = 0 (Timeout)
[pid 4096830] poll([{fd=4, events=POLLIN}], 1, 500) = 0 (Timeout)
[pid 4096830] poll([{fd=4, events=POLLIN}], 1, 500) = 0 (Timeout)
[pid 4096830] poll([{fd=4, events=POLLIN}], 1, 500) = 0 (Timeout)
[pid 4096830] poll([{fd=4, events=POLLIN}], 1, 500) = 0 (Timeout)
[pid 4096830] poll([{fd=4, events=POLLIN}], 1, 500) = 0 (Timeout)
[pid 4096830] poll([{fd=4, events=POLLIN}], 1, 500) = 0 (Timeout)
[pid 4096830] poll([{fd=4, events=POLLIN}], 1, 500) = 0 (Timeout)
[pid 4096830] poll([{fd=4, events=POLLIN}], 1, 500) = 0 (Timeout)
[pid 4096830] poll([{fd=4, events=POLLIN}], 1, 500) = 0 (Timeout)
[pid 4096830] poll([{fd=4, events=POLLIN}], 1, 500) = 1 ([{fd=4, revents=POLLIN}])
[pid 4096830] poll([{fd=4, events=POLLIN}], 1, 20000) = 1 ([{fd=4, revents=POLLIN}])
[pid 4096830] accept4(4, {sa_family=AF_INET, sin_port=htons(23183), sin_addr=inet_addr("192.168.1.99")}, [16], SOCK_CLOEXEC) = 6
[pid 4096830] getsockname(6, {sa_family=AF_INET, sin_port=htons(8000), sin_addr=inet_addr("192.168.1.17")}, [128->16]) = 0
[pid 4096830] ioctl(6, FIONBIO, [1])    = 0
[pid 4096830] poll([{fd=6, events=POLLIN}], 1, 20000) = 1 ([{fd=6, revents=POLLIN}])
[pid 4096830] recvfrom(6, "POST /weatherdata/report/ HTTP/1"..., 8192, 0, NULL, NULL) = 880
[pid 4096830] getpid()                  = 4096814
[pid 4096830] futex(0x2100610, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, NULL, FUTEX_BITSET_MATCH_ANY
```

The last few lines of weewx.log show:

```
Mar 26 21:21:40 dirac weewx[4096814] DEBUG user.interceptor: ignored parameter monthlyrainin=0.693
Mar 26 21:21:40 dirac weewx[4096814] DEBUG user.interceptor: ignored parameter yearlyrainin=2.984
Mar 26 21:21:40 dirac weewx[4096814] DEBUG user.interceptor: ignored parameter freq=433M
Mar 26 21:21:40 dirac weewx[4096814] DEBUG user.interceptor: ignored parameter model=GW1000'
Mar 26 21:21:40 dirac weewx[4096814] DEBUG user.interceptor: raw packet: {'dateTime': 1679826099, 'usUnits': 1, 'temperature_in': 68.72, 'humidity_in': 68.0, 'pressure': 29.648, 'temperature_out': 60.98, 'humidity_out': 84.0, 'wind_dir': 193.0, 'wind_speed': 0.0, 'wind_gust': 0.0, 'solar_radiation': 0.0, 'uv': 0.0, 'rain_rate': 0.0, 'rain_total': 2.984, 'temperature_1': 68.0, 'humidity_1': 67.0, 'temperature_2': 69.44, 'humidity_2': 62.0, 'temperature_3': 38.84, 'humidity_3': 35.0, 'temperature_4': 53.78, 'humidity_4': 54.0, 'temperature_5': 12.02, 'humidity_5': 43.0, 'temperature_6': 42.8, 'humidity_6': 13.0, 'soil_moisture_1': 29.0, 'soil_moisture_3': 35.0, 'battery': 0.0, 'battery_1': 0.0, 'battery_2': 0.0, 'battery_3': 0.0, 'battery_4': 0.0, 'battery_5': 0.0, 'battery_6': 0.0, 'soil_battery_1': 1.6, 'soil_battery_3': 1.4, 'rain': 0.0}
Mar 26 21:21:40 dirac weewx[4096814] DEBUG user.interceptor: mapped packet: {'dateTime': 1679826099, 'usUnits': 1, 'pressure': 29.648, 'outHumidity': 84.0, 'inHumidity': 68.0, 'outTemp': 60.98, 'inTemp': 68.72, 'windSpeed': 0.0, 'windGust': 0.0, 'windDir': 193.0, 'radiation': 0.0, 'rain': 0.0, 'rainRate': 0.0, 'UV': 0.0, 'txBatteryStatus': 0.0, 'extraTemp1': 68.0, 'extraTemp2': 69.44, 'extraTemp3': 38.84, 'extraHumid1': 67.0, 'extraHumid2': 62.0, 'soilMoist1': 29.0, 'soilMoist3': 35.0, 'extraTemp4': 53.78, 'extraTemp5': 12.02, 'extraTemp6': 42.8, 'extraHumid3': 35.0, 'extraHumid4': 54.0, 'extraHumid5': 43.0, 'extraHumid6': 13.0}
Mar 26 21:23:52 dirac weewx[4096814] INFO weewx.manager: Added record 2023-03-26 21:21:00 AEDT (1679826060) to database 'weewx.sdb'
Mar 26 21:26:52 dirac weewx[4096814] INFO weewx.manager: Added record 2023-03-26 21:21:00 AEDT (1679826060) to daily summary in 'weewx.sdb'
Mar 26 21:29:53 dirac weewx[4096814] DEBUG weewx.reportengine: Running reports for latest time in the database.
Mar 26 21:32:53 dirac weewx[4096814] DEBUG weewx.reportengine: Running report 'SeasonsReport'
```

Running a tcpdump shows that the ecowitt is sending new data in the meantime every minute or so.

After some time around 20 minutes, weewx resumes from the futex() call
and carries on as if nothing happened:

```
1272307 <... sendto resumed>)           = 90
1272307 futex(0x2100610, FUTEX_WAKE_PRIVATE, 1) = 1
4096814 <... futex resumed>)            = 0
4096814 futex(0x93a56c, FUTEX_WAIT_BITSET_PRIVATE, 0, {tv_sec=3036626, tv_nsec=527168657}, FUTEX_BITSET_MATCH_ANY <unfinished ...>
1272307 getpid()                        = 4096814
1272307 futex(0x93a56c, FUTEX_WAKE_PRIVATE, 1) = 1
4096814 <... futex resumed>)            = 0
1272307 futex(0x2100610, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, NULL, FUTEX_BITSET_MATCH_ANY <unfinished ...>
4096814 futex(0x93a570, FUTEX_WAKE_PRIVATE, 1) = 0
4096814 sendto(3, "<15>weewx[4096814] DEBUG user.in"..., 55, 0, NULL, 0) = 55
4096814 futex(0x2100610, FUTEX_WAKE_PRIVATE, 1 <unfinished ...>
4096830 <... futex resumed>)            = 0
4096814 <... futex resumed>)            = 1
4096830 futex(0x93a568, FUTEX_WAIT_BITSET_PRIVATE, 0, {tv_sec=3036626, tv_nsec=527541851}, FUTEX_BITSET_MATCH_ANY <unfinished ...>
4096814 futex(0x93a568, FUTEX_WAKE_PRIVATE, 1 <unfinished ...>
4096830 <... futex resumed>)            = -1 EAGAIN (Resource temporarily unavailable)
4096814 <... futex resumed>)            = 0
4096830 futex(0x93a570, FUTEX_WAIT_PRIVATE, 2, NULL <unfinished ...>
```

with weewx.log resuming after the above with:

```
[... 9 lines cut ...]
weewx.log is showing the pauses aren't always happening at the same place at least as far as weewx.log is concerned.  Here's an earlier one with timestamps:

```
Mar 24 05:12:45 dirac weewx[1228887] DEBUG user.interceptor: ignored parameter dailyrainin=0.000
Mar 24 05:12:45 dirac weewx[1228887] DEBUG user.interceptor: ignored parameter weeklyrainin=0.244
Mar 24 05:12:45 dirac weewx[1228887] DEBUG user.interceptor: ignored parameter monthlyrainin=0.693
Mar 24 05:12:45 dirac weewx[1228887] DEBUG user.interceptor: ignored parameter yearlyrainin=2.984
Mar 24 05:12:45 dirac weewx[1228887] DEBUG user.interceptor: ignored parameter freq=433M
Mar 24 05:12:45 dirac weewx[1228887] DEBUG user.interceptor: ignored parameter model=GW1000'
Mar 24 05:12:45 dirac weewx[1228887] DEBUG user.interceptor: raw packet: {'dateTime': 1679595165, 'usUnits': 1, 'temperature_in': 71.78, 'humidity_in': 68.0, 'pressure': 29.781, 'temperature_out': 62.06
, 'humidity_out': 90.0, 'wind_dir': 197.0, 'wind_speed': 0.0, 'wind_gust': 0.0, 'solar_radiation': 0.0, 'uv': 0.0, 'rain_rate': 0.0, 'rain_total': 2.984, 'temperature_1': 69.8, 'humidity_1': 66.0, 'temp
erature_2': 71.78, 'humidity_2': 64.0, 'temperature_3': 40.28, 'humidity_3': 54.0, 'temperature_4': 54.86, 'humidity_4': 54.0, 'temperature_5': 9.68, 'humidity_5': 46.0, 'temperature_6': 41.9, 'humidity
_6': 13.0, 'soil_moisture_1': 29.0, 'soil_moisture_3': 31.0, 'battery': 1.0, 'battery_1': 0.0, 'battery_2': 0.0, 'battery_3': 0.0, 'battery_4': 0.0, 'battery_5': 0.0, 'battery_6': 0.0, 'soil_battery_1':
 1.6, 'soil_battery_3': 1.4, 'rain': 0.0}
Mar 24 05:12:45 dirac weewx[1228887] DEBUG user.interceptor: mapped packet: {'dateTime': 1679595165, 'usUnits': 1, 'pressure': 29.781, 'outHumidity': 90.0, 'inHumidity': 68.0, 'outTemp': 62.06, 'inTemp'
: 71.78, 'windSpeed': 0.0, 'windGust': 0.0, 'windDir': 197.0, 'radiation': 0.0, 'rain': 0.0, 'rainRate': 0.0, 'UV': 0.0, 'txBatteryStatus': 1.0, 'extraTemp1': 69.8, 'extraTemp2': 71.78, 'extraTemp3': 40
.28, 'extraHumid1': 66.0, 'extraHumid2': 64.0, 'soilMoist1': 29.0, 'soilMoist3': 31.0, 'extraTemp4': 54.86, 'extraTemp5': 9.68, 'extraTemp6': 41.9, 'extraHumid3': 54.0, 'extraHumid4': 54.0, 'extraHumid5
': 46.0, 'extraHumid6': 13.0}
Mar 24 05:15:01 dirac weewx[1228887] DEBUG user.interceptor: empty queue
Mar 24 05:21:02 dirac weewx[1228887] DEBUG user.interceptor: empty queue
Mar 24 05:24:02 dirac weewx[1228887] DEBUG user.interceptor: empty queue
Mar 24 05:33:03 dirac weewx[1228887] DEBUG user.interceptor: empty queue
Mar 24 05:33:03 dirac weewx[1228887] DEBUG user.interceptor: empty queue
Mar 24 05:33:03 dirac weewx[1228887] DEBUG user.interceptor: empty queue
```

gjr80, 2023-03-26:
I can't help you analyse `strace` or `tcpdump` output, but have you considered using the [Ecowitt Gateway Device API based driver](https://github.com/gjr80/weewx-gw1000) rather than the interceptor driver? When the GW1000 was first supported by WeeWX the interceptor driver was the only driver option, but as the gateway device API was released the API based driver was developed and is now the recommended driver for Ecowitt gateway devices. I think you will find it easier to use, has a greater vocabulary of device/sensor data plus it doesn't need the gateway device to be set to upload to any external service/host. The API based driver requires network connectivity with the gateway device on the local network segment, so if the device is remote or you have network reliability issues the API based driver may not work.

tkeffer, 2023-03-28:
Neither the GW1000, nor the interceptor driver, are part of WeeWX, so you're unlikely to get a fix here.

following Gary's suggestion would be your best bet.

sorry.