weewx/weewx #867: Set locale when using internationalized skins

tkeffer, 2023-05-05:
Right now, we depend on the user setting the `LOCALE` environment variable before running reports. This works, but it means all reports use the same locale.

Alternatively, we could optionally set the locale within the program before running each report.

``` python
import locale
...
locale.setlocale(locale.LC_ALL, 'es_ES')
```

Unfortunately, the locale used in `setlocale()` requires a country code as well as a language code, so we can't use option `lang` in `skin.conf` (which is only a language code). So, we need to either add a country code to `lang`, or we need another option `locale`.

tkeffer, 2024-05-02:
Implemented in commit https://github.com/weewx/weewx/commit/bbaf514fe8c3d1692cc759620e7ea00d66ef0225

closed 2024-05-02 by tkeffer

mKainzbauer, 2024-05-02:
Yes! 😀

tkeffer, 2024-05-02:
Glad you like it!

See [_Changing languages_](https://github.com/weewx/weewx/blob/development/docs_src/custom/custom-reports.md#changing-languages) for how it works.