weewx/weewx #868: weewx upgrade testing matrix for 4-to-5

matthewwall, 2023-05-07:
this is intended to track the testing of upgrade paths from weewx 4 to weewx 5.  please append to this issue with each case that you test.  include the initial system/python/weewx configuration and the ending system/python/weewx configuration.

this is the list of upgrade paths that should work out-of-the-box:

general cases:
* [ ] weewx v4 with python2 to weewx v5
* [ ] weewx v4 with python3 to weewx v5
* [ ] weewx v5 to later v5
  * [ ] v5aN to v5aM
  * [ ] v5b to v5
 
specific tests
* [ ] debian 9, python2, weewx 4 to debian 11, python3, weewx 5
* [ ] centos7, python2, weewx 4 to rocky8, python3, weewx 5

tkeffer, 2023-05-07:
Tested the first in your list: upgrading from v4 with Python 2 to V5 with Python 3. 

The user will have to edit `sources.list.d/weewx.list` and change this

    deb [arch=all] http:/weewx.com/apt/python2 squeeze main

to this

    deb [arch=all] https:/weewx.com/apt/python3 stable main

We should mention that in the Upgrade Guide.

tkeffer, 2023-05-07:
Also tested the second on the list: Upgrading v4 to v5, both using Python 3, on a virgin copy of Deb 10.13. No problems.

matthewwall, 2023-12-29:
this issue has been superseded by the v5 testing matrix wiki page

https://github.com/weewx/weewx/wiki/Version-5-testing

closed 2023-12-29 by matthewwall