weewx/weewx #869: how to remove a simulator station from the weewx stations map

kjames79, 2023-05-17:
I have a proper station now that is registered to the map at https://weather.kevinheaven.net but a simulator station I made when I was first trying to work with weewx is still there and unless I remove the "simulators" option on the stations map you are unable to see my proper station. How do I put a request in to have the simulator station removed? It's been on there for over a month. Thanks

tkeffer, 2023-05-25:
No response.

closed 2023-05-25 by tkeffer

kjames79, 2023-05-26:
> All of the "kevinheaven" entries I am seeing are of station type "SDR".
>
> Are you still having this problem?

yes, when the stations aren't being filtered and you go to the stations map
in its default state where you are able to see all of the stations an old
simulator station I must have published before building and publishing my
actual station is placed over top of my actual station, making it
impossible to see my actual station. I'd like to find out how I can have
that one removed. is there someone that I need to contact?

here is what comes up when no filter is applied
[image: snip1.JPG]

and when the filter is applied you can see the actual station. I doubt many
people are visiting the stations map and applying filters. I'd like only my
actual station to be there and not hidden behind the non-existent simulator
station
[image: snip2.JPG]

kjames79, 2023-05-26:
I should add that initially I really didn't know what I was doing and the link for the old simulator station is a link to another of my stations on weatherunderground. Not sure if this helps at all... It might be why it hasn't just cycled out since it's a live link

tkeffer, 2023-05-26:
You're not giving me a lot of information here. What's the offending `station_url`??

I see a bunch of entries to https://www.wunderground.com/dashboard/pws/KPAOAKDA3, all of which are type `Simulator`, so I'm going to assume that's it and delete them.

Let me know if I got lucky on my guess.

kjames79, 2023-05-26:
Yes that was it! Thank you so much! Sorry for the trouble. Really
appreciate the help. Loving the Weewx software btw. I've wanted something
like this for a long time.

kjames79, 2023-05-26:
Last thing, I'm still getting an old bad link that is no longer active.
This one needs to be removed: https://wx.kevinheaven.net/weewx/

The one that stays is https://weather.kevinheaven.net

Sorry Thanks again

tkeffer, 2023-05-27:
Done.

I will note that a `station_url` is no longer used after 30 days. If you wait long enough, the station map will correct itself.

kjames79, 2023-05-27:
good to know. thank you