weewx/weewx #876: extractor noop

roe-dl, 2023-07-09:
WeeWX is hard to crash, but I managed to crash it. I simply set the extractor to `noop` for an observation type that is saved to the database. The error message was `too many values to unpack (expected 4)`. Unfortunately, I did not find out where this error came from in `manager.py` or `accum.py`.

There was an user extension involved, but the issue is not depending on that user extension. It is only about using the `noop` extractor where it should not.

```
Jul  9 14:55:17 Lokal weewx[482791] INFO weewx.manager: Added record 2023-07-09 14:55:00 CEST (1688907300) to database 'weewx.sdb'
Jul  9 14:55:17 Lokal weewx[482791] INFO weewx.manager: Added record 2023-07-09 14:55:00 CEST (1688907300) to daily summary in 'weewx.sdb'
Jul  9 14:55:17 Lokal weewx[482791] INFO weewx.engine: Main loop exiting. Shutting engine down.
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__: Caught unrecoverable exception:
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****  too many values to unpack (expected 4)
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****  Traceback (most recent call last):
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 214, in run
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      self.dispatchEvent(weewx.Event(weewx.CHECK_LOOP, packet=packet))
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 245, in dispatchEvent
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      callback(event)
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 634, in check_loop
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      raise BreakLoop
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****  weewx.engine.BreakLoop
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****  
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****  During handling of the above exception, another exception occurred:
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****  
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****  Traceback (most recent call last):
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 650, in post_loop
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      self._catchup(self.engine.console.genArchiveRecords)
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 697, in _catchup
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      for record in generator(lastgood_ts):
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/drivers/__init__.py", line 30, in genArchiveRecords
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      raise NotImplementedError("Method 'genArchiveRecords' not implemented")
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****  NotImplementedError: Method 'genArchiveRecords' not implemented
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****  
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****  During handling of the above exception, another exception occurred:
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****  
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****  Traceback (most recent call last):
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewxd", line 154, in main
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      engine.run()
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 221, in run
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      self.dispatchEvent(weewx.Event(weewx.POST_LOOP))
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 245, in dispatchEvent
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      callback(event)
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 652, in post_loop
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      self._software_catchup()
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 715, in _software_catchup
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      self.engine.dispatchEvent(weewx.Event(weewx.NEW_ARCHIVE_RECORD,
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/engine.py", line 245, in dispatchEvent
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      callback(event)
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/user/precipmeter.py", line 2919, in new_archive_record
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      self.dbm_new_archive_record(event.record)
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/user/precipmeter.py", line 2964, in dbm_new_archive_record
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      self.dbm.addRecord(record,
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/manager.py", line 403, in addRecord
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      self._updateHiLo(accumulator, cursor)
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/manager.py", line 1189, in _updateHiLo
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      _stats_dict = self._get_day_summary(_sod_ts, cursor)
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/manager.py", line 1576, in _get_day_summary
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      _day_accum.set_stats(_day_key, _stats_tuple)
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/accum.py", line 489, in set_stats
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      self[obs_type].setStats(stats_tuple)
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****    File "/usr/share/weewx/weewx/accum.py", line 370, in setStats
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****      self.first, self.firsttime, self.last, self.lasttime = stats_tuple \
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****  ValueError: too many values to unpack (expected 4)
Jul  9 14:55:21 Lokal weewx[482791] CRITICAL __main__:     ****  Exiting.
```

Excerpt from `precipmeter.py`:

```
...
class PrecipArchive(StdService):

    ...

    def dbm_init(self, engine, binding, binding_found):
        self.accumulator = None
        self.old_accumulator = None
        self.dbm = None
        ...
        self.dbm = engine.db_binder.get_manager(data_binding=binding,
                                                     initialize=True)
        if self.dbm:
            loginf("Using binding '%s' to database '%s'" % (binding,self.dbm.database_name))
            # Back fill the daily summaries.
            _nrecs, _ndays = self.dbm.backfill_day_summary()
        else:
            loginf("no database access")

    ...

    def new_archive_record(self, event):
        """ process archive record """
        if self.dbm:
            self.dbm_new_archive_record(event.record)

    ...

    def dbm_new_archive_record(self, record):
        if self.dbm:
            self.dbm.addRecord(record,
                           accumulator=self.old_accumulator,
                           log_success=self.log_success,
                           log_failure=self.log_failure)

    ...
```

Of course, after that experience there is an try-except clause around the call of dbm_new_archive_record.

tkeffer, 2023-07-11:
I tried setting the extractor for `consBatteryVoltage` to `noop`, and it worked fine. I'm sorry, but I cannot replicate this without more information.

Was it the extractor or adder that was `noop`?

Can you come up with an example that does not use your custom service?

roe-dl, 2023-07-11:
It was the extractor. 

I will try to find a to demonstrate the issue in a pure WeeWX configuration.

roe-dl, 2023-07-12:
The crash happened at the end of the archive interval while WeeWX was about to store a new archive record to the database. That's why I thought the extractor was the problem. But it was not. It was a little bit different.

So, how to reproduce the crash? I set up pure WeeWX with the simulator driver. No extensions. Then I started WeeWX and let it run one archive cycle. After that I added the accumulator:

```
[Accumulator]
    [[outTemp]]
        accumulator = firstlast
        adder = noop
        extractor = noop
```

Now I started WeeWX again. **And now it crashed at the end of the archive cycle.**

So the real reason for the crash was: I did not only change the extractor to `noop` but also the accumulator from `scalar` to `firstlast`. I was not aware of this additional change when I created this issue.

Now, when it came to update the daily summary, WeeWX read the record from the daily summary table in the database. That table represented a `scalar` accumulator. And when WeeWX tried to assign that record to the `firstlast` accumulator according to the changed configuration, it crashed.

I added a try-except clause in `accum.py` in line 489:

```
    def set_stats(self, obs_type, stats_tuple):

        self._init_type(obs_type)
        try:
            self[obs_type].setStats(stats_tuple)
        except ValueError as e:
            log.error("accum set_stats obs_type='%s' stats_tuple='%s' %s %s" % (obs_type,stats_tuple,e.__class__.__name__,e))
```

and I got the following error message instead of the crash:

```
Jul 12 06:40:14 Lokal weewx[10214] ERROR weewx.accum: accum set_stats obs_type='outTemp' stats_tuple='(32.59736413552349, 1689136499, 32.67858297022247, 1689136386, 32.637878863872146, 1, 9791.363659161643, 300)' ValueError too many values to unpack (expected 4)
Jul 12 06:40:14 Lokal weewx[10214] ERROR weewx.manager: Replace failed for database weewxtest.sdb: table archive_day_outTemp has 9 columns but 5 values were supplied
```

tkeffer, 2023-07-13:
Problem turned out to be with the `firstlast` accumulator, not `noop`.

Swap out your copy of `accum.py` and see if it works for you.

closed 2023-07-13 by tkeffer

roe-dl, 2023-07-14:
Checked. No log messages regarding accumulators or daily summaries any more.