weewx/weewx #877: Cannot display XType in current section of Seasons skin

bellrichm, 2023-07-11:
The current section (current.inc) of Seasons uses has_data to determine if an observation has data. has_data checks if the observation is a column in the database. If it is not, it returns False. This means that to display an XType, it must be backed by the database.
I was originally seeing this in a custom XType. I then removed appTemp from the database and has_data now returns False for it.

tkeffer, 2023-07-27:
I haven't forgotten about this. It's just that I haven't thought of an easy fix without running the full xtypes query, which would be quite expensive.

The right way to do it is to add function `exists()` and `has_data()` to the xtypes interface, in a similar manner to the way `get_scalar()`, `get_aggregate()`, etc. are done. But, it's way too late to be changing the xtypes interface.

As an interim solution, the class `XType` could have implementations of `exists()` and `has_data()` that do the full query. Subclasses could then offer more efficient versions.

I'll keep noodling on it.

bellrichm, 2023-07-27:
Tom, thanks for the update. No worries, there are work arounds.
If I had a good idea, I would have submitted a pull request. Maybe these ideas might spark some thing…
Currently, I changed I `has_data` in `Manager` to 
```

    def has_data(self, obs_type, timespan):
        """Checks whether the observation type exists in the database and whether it has any
        data.

        Args:
            obs_type(str): The observation type to check for existence.
            timespan (tuple): A 2-way tuple with the start and stop time to be checked for data.

        Returns:
            bool: True if the type is in the schema, and has some data within the given timespan.
                Otherwise, return False.
        """
        return bool(weewx.xtypes.get_aggregate(obs_type, timespan, 'not_null', self)[0])
```
Since `Manager` knows about xtypes, I wonder if `exists` could somehow check if the type is also an xtype? Something like this
```
def exists(self, obs_type):
        """Checks whether the observation type exists in the database.

        Args:
            obs_type(str): The observation type to check for existence.

        Returns:
            bool: True if the observation type is in the database schema. False otherwise.
        """

        # Check to see if this is a valid observation type:
        return obs_type in self.obskeys or obs_type in self.xtype_obs
```
I’m still dipping my toe in xtypes, so thanks for your patience.

Again, thanks so much for WeeWX.

bellrichm, 2023-07-27:
I believe that `has_data` is handled by a xtype implementing the aggregate_type `not_null`. I had to implement this to get the xtype to plot via imagegenerator.
So a new method `exists()` needs to be implemented by each xtype. I put together a small prototype in a pull request, #882

tkeffer, 2023-08-30:
Turned out not to be as hard as I thought. Commit https://github.com/weewx/weewx/commit/b259d3dd91d74252fa058837f3f4cb6ea157f651

I'll upload a new version to pypi later today.

closed 2023-08-30 by tkeffer