weewx/weewx #885: Allow extension installer to install from pypi

tkeffer, 2023-08-24:
See https://groups.google.com/g/weewx-user/c/pYH7f2xzSq4/m/WIHdjMCsAQAJ

matthewwall, 2024-02-11:
the description should probably be "allow extensions to specify python (and other?) dependencies, and allow the extension installer to install those dependencies".

for the case where weewx is running within a python virtual environment (either using a pip install or using a deb/rpm install), then allowing the extension installer to install dependencies is *usually* a fairly safe thing to do.

however, it can get ugly very quickly.

for example:

* if two different extensions require different versions of the same package, how do you track that and which one has priority?
* if weewx is *not* running in a virtual environment, and the python it is using is the system python or a python that is not managed using pip (e.g., a system alternative), then installing anything using pip will probably break something (assuming the user has permissions to even do so, which may protect against this failure mode).

letting extensions specify their dependencies, and letting extension installer install them, would be a massively good thing.  but we must ensure that we define the scope and test the permissions so that we don't unleash a monster.

tkeffer, 2024-02-11:
I'm not as worried about those things as the _integration_ problems. 

Pip can install the extensions --- the problems you mention are typical of any pip install. It's the next step I worry about: how do you make the changes to `weewx.conf`. How do you install the skins.

I can see `weectl` growing into the monster you worry about.

tkeffer, 2024-02-11:
Not a priority. I'm going to remove this issue. It can be restored later if one of us gets a brilliant idea.

closed 2024-02-11 by tkeffer