weewx/weewx #887: weewx v5 restart fails on Ethernet connection - Too quick?

chrisale, 2023-08-27:
Running weewx v5b10 pip install on Debian.  Connecting to Vantage Pro 2 via Ethernet-Serial bridge.

WeeWx consistently fails to restart.  I believe it is simply acting too quickly for the Ethernet bridge, if it paused for 1-2 seconds before starting the system I believe it might resolve this issue.

Logs below show normal shutdown after restart command issued, failed start, then successful start after reissuing restart command 6 seconds later.

Aug 27 09:26:19 footiMac weewx[638331] INFO __main__: Received signal TERM (15).
Aug 27 09:26:19 footiMac weewx[638331] INFO weewx.engine: Main loop exiting. Shutting engine down.
Aug 27 09:26:19 footiMac weewx[638331] INFO weewx.engine: Shutting down StdReport thread
Aug 27 09:26:19 footiMac weewx[638331] INFO __main__: Terminating weewx version 5.0.0b10
Aug 27 09:26:19 footiMac systemd[1]: weewx.service: Succeeded.
Aug 27 09:26:19 footiMac systemd[1]: weewx.service: Consumed 5min 59.866s CPU time.
Aug 27 09:26:19 footiMac weewx[653062] INFO __main__: Initializing weewx version 5.0.0b10
Aug 27 09:26:19 footiMac weewx[653062] INFO __main__: Using Python 3.9.2 (default, Feb 28 2021, 17:03:44) #012[GCC 10.2.1 20210110]
Aug 27 09:26:19 footiMac weewx[653062] INFO __main__: Located at /home/pi/weewx-venv/bin/python3
Aug 27 09:26:19 footiMac weewx[653062] INFO __main__: Platform Linux-5.10.0-25-amd64-x86_64-with-glibc2.31
Aug 27 09:26:19 footiMac weewx[653062] INFO __main__: Locale is 'en_CA.UTF-8'
Aug 27 09:26:19 footiMac weewx[653062] INFO __main__: Entry path: /home/pi/weewx-venv/lib/python3.9/site-packages/weewxd.py
Aug 27 09:26:19 footiMac weewx[653062] INFO __main__: Using configuration file /home/pi/weewx-data/weewx.conf
Aug 27 09:26:19 footiMac weewx[653062] INFO __main__: Debug is 0
Aug 27 09:26:19 footiMac weewx[653062] INFO weewx.engine: Loading station type Vantage (weewx.drivers.vantage)
Aug 27 09:26:19 footiMac weewx[653062] ERROR weewx.drivers.vantage: Socket error while opening port 4660 to ethernet host 192.168.1.101.
Aug 27 09:26:19 footiMac weewx[653062] ERROR weewx.engine: Import of driver failed: [Errno 111] Connection refused (<class 'weewx.WeeWxIOError'>)
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****  Traceback (most recent call last):
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****    File "/home/pi/weewx-venv/lib/python3.9/site-packages/weewx/drivers/vantage.py", line 354, in openPort
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****      self.socket.connect((self.host, self.port))
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****  ConnectionRefusedError: [Errno 111] Connection refused
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****  
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****  During handling of the above exception, another exception occurred:
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****  
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****  Traceback (most recent call last):
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****    File "/home/pi/weewx-venv/lib/python3.9/site-packages/weewx/engine.py", line 116, in setupStation
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****      self.console = loader_function(config_dict, self)
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****    File "/home/pi/weewx-venv/lib/python3.9/site-packages/weewx/drivers/vantage.py", line 32, in loader
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****      return VantageService(engine, config_dict)
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****    File "/home/pi/weewx-venv/lib/python3.9/site-packages/weewx/drivers/vantage.py", line 2046, in __init__
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****      Vantage.__init__(self, **config_dict[DRIVER_NAME])
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****    File "/home/pi/weewx-venv/lib/python3.9/site-packages/weewx/drivers/vantage.py", line 536, in __init__
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****      self.port.openPort()
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****    File "/home/pi/weewx-venv/lib/python3.9/site-packages/weewx/drivers/vantage.py", line 359, in openPort
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****      raise weewx.WeeWxIOError(ex)
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL weewx.engine:     ****  weewx.WeeWxIOError: [Errno 111] Connection refused
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL __main__: Unable to load driver: [Errno 111] Connection refused
Aug 27 09:26:19 footiMac weewx[653062] CRITICAL __main__:     ****  Exiting...
Aug 27 09:26:19 footiMac systemd[1]: weewx.service: Main process exited, code=exited, status=4/NOPERMISSION
Aug 27 09:26:19 footiMac systemd[1]: weewx.service: Failed with result 'exit-code'.
Aug 27 09:26:25 footiMac weewx[653110] INFO __main__: Initializing weewx version 5.0.0b10
Aug 27 09:26:25 footiMac weewx[653110] INFO __main__: Using Python 3.9.2 (default, Feb 28 2021, 17:03:44) #012[GCC 10.2.1 20210110]
Aug 27 09:26:25 footiMac weewx[653110] INFO __main__: Located at /home/pi/weewx-venv/bin/python3
Aug 27 09:26:25 footiMac weewx[653110] INFO __main__: Platform Linux-5.10.0-25-amd64-x86_64-with-glibc2.31
Aug 27 09:26:25 footiMac weewx[653110] INFO __main__: Locale is 'en_CA.UTF-8'
Aug 27 09:26:25 footiMac weewx[653110] INFO __main__: Entry path: /home/pi/weewx-venv/lib/python3.9/site-packages/weewxd.py
Aug 27 09:26:25 footiMac weewx[653110] INFO __main__: Using configuration file /home/pi/weewx-data/weewx.conf
Aug 27 09:26:25 footiMac weewx[653110] INFO __main__: Debug is 0
Aug 27 09:26:25 footiMac weewx[653110] INFO weewx.engine: Loading station type Vantage (weewx.drivers.vantage)
Aug 27 09:26:30 footiMac weewx[653110] INFO user.purple: Service version is 3.3.
Aug 27 09:26:30 footiMac weewx[653110] INFO user.purple

tkeffer, 2023-08-27:
Have you tried option [`tcp_send_delay`](http://www.weewx.com/docs/latest/usersguide.htm#[Vantage])? 

There is also an undocumented option `command_delay`. This controls how long to wait after sending a command to the Vantage. Default is 0.5 seconds.

chrisale, 2023-08-28:
Have tried with both tcp_send_delay and command_delay set to 2 and while it does delay the start of communication on a startup from a stopped, it still fails to restart with the same connection refusal after a restart command from an active state.

tkeffer, 2023-08-28:
I didn't quite follow that, but no matter. I don't have an ethernet version of the logger at my disposal so it's hard for me to experiment. Perhaps you could experiment with where you think the delay should be?

chrisale, 2023-08-28:
Sorry I re-read my last comment and it was about as clear as mud. 

Adding a delay does not have any effect on the error when restarting the service.  All it seems to do is lengthen the time between every single operation while weewx is running. I tried it at 5s and 10s (for both tcp_send_delay and command_delay.

There does not seem to be any delay observed when initially starting the service regardless of those settings, and even if a 5 second delay had worked, the impact on the responsiveness of the system as a whole is quite large.

A 'delay-on-startup' value that you could set at 10 or 15 seconds might work.

tkeffer, 2023-08-28:
Can you give one a try? For example, try changing this (found around line 352)

```
            self.socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
            self.socket.settimeout(self.timeout)
            self.socket.connect((self.host, self.port))
```

to this

```
            self.socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
            self.socket.settimeout(self.timeout)
            self.socket.connect((self.host, self.port))
            time.sleep(5.0)
```

Experiment with sleeping for various amounts of time. If it works, we can parameterize it with a sensible default.

chrisale, 2023-08-29:
I have found something that works.

I inserted a 5s sleep (plus a message) in weewxd.py right after the initial log messaging:
around line 108
    log.info("Debug is %s", weewx.debug)
    log.info("Pausing for Ethernet Serial Sanity")
    time.sleep(5.0)

Results in log entry:
Aug 28 20:31:27 footiMac weewx[1101312] INFO __main__: Debug is 0
Aug 28 20:31:27 footiMac weewx[1101312] INFO __main__: Pausing for Ethernet Serial Sanity
Aug 28 20:31:32 footiMac weewx[1101312] INFO weewx.engine: Loading station type Vantage (weewx.drivers.vantage)
Aug 28 20:31:36 footiMac weewx[1101312] INFO user.purple: Service version is 3.3.

And successful restart!

It sleeps on both a cold start and a restart but that's fine if it's configurable.

tkeffer, 2023-08-29:
That early in the startup of `weewxd`, there is nothing asynchronous going on, so I'm not seeing why sleeping would make any difference. All that has happened is that the configuration file has been read and parsed. The device driver hasn't even been loaded, yet alone run.

I'm reluctant to make everyone's `weewxd` sleep for an unknown reason. If this works for you, you could put the sleep in `user.extensions`.

chrisale, 2023-08-29:
I see in the weewxd startup code there are catchers for critical errors.  Could we catch the:
"Aug 27 09:26:19 footiMac weewx[653062] CRITICAL main: Unable to load driver: [Errno 111] Connection refused"

error and then allow it to pause for 60secs rather than exit? That would most likely solve it.

tkeffer, 2023-08-29:
Option [`loop_on_init`](https://www.weewx.com/docs/5.0/reference/weewx-options/general/#loop_on_init) does exactly that. Did it not work for you?

chrisale, 2023-08-29:
no

tkeffer, 2023-08-29:
@duvalljm Are you using the Vantage IP logger?

@chrisale Can you be a bit more specific?

duvalljm, 2023-08-29:
First, thank you for taking the time.  However, understand that my experience goes back to the early MS DOS days before anyone had heard of internet and everything outside the Windows world is slow going for me.  So, if it seems hopeless, just walk away.  
Vantage IP logger:  I assume (possibly incorrectly) that this is reference to a Davis Vantage station.  So, no.  I am using Weatherflow Tempest.
More specific: The behavior I have observed seems to me to be related to weewx and not station specific but again, If I knew the answer I wouldn't be fishing for ideas.
Weewx on Raspberry Pi 4b pulling data from Tempest station and feeding to PWSWeather.  All working fine until our area lost internet connection (ISP issue).  Internet down for several minutes and I noticed that even after it came back, data was no longer being fed to PWS.  I checked the log (sudo tail -f /var/log/syslog) and saw zero activity related to weewx.  Unsure how to proceed, I issued command to stop weewx (sudo /etc/init.d/weewx stop).  Response was that weewx was not running.  Issued command to start weewx (sudo /etc/init.d/weewx start).  Observed the expected log entries and data again began showing up at PWS.  Power to the pi was not interrupted and it ran continuously throughout.  Same thing happened again the next day all as previously described.  The only negative event was the drop of internet connection for several minutes.  The only thing I can conclude is that if the internet connection drops, weewx stops running.

tkeffer, 2023-08-30:
@duvalljm This dicussion is about the Vantage driver. The Tempest driver is supplied by a 3rd party. How it responds to a LAN failure is entirely up to how the driver was written. A WAN failure will cause uploads to fail, but they don't crash weewxd. I just tested this, and it worked perfectly.

duvalljm, 2023-08-30:
Thank you for your insight and taking the extra step to test.  I will go that direction and see where it leads.

tkeffer, 2023-09-04:
@chrisale Anything new on this?

chrisale, 2023-09-04:
Hi Tom,
No, I updated to b11 this morning (which removed my temporary modification) and it exhibited the same behaviour when restarting. In the file below you can see the log when I issued the restart command at 11:05:52. It successfully stopped, then it restarts and fails with a Socket Error, Connection refused, and crash.
I then restart it successfully at 11:06:35.
[weewxdebub.txt](https://github.com/weewx/weewx/files/12515865/weewxdebub.txt)

tkeffer, 2023-09-04:
I was hoping for more information on why `loop_on_init` did not work for you.

chrisale, 2023-09-04:
Sorry Tom, I'm not sure how to get any more info on why it's not working.  Turning on Debug and restarting again didn't provide any more detail in the logs (attached).  Suggestions?

[weewxdebub.txt](https://github.com/weewx/weewx/files/12515925/weewxdebub.txt)

tkeffer, 2023-09-04:
It does not appear to me that `loop_on_init` is enabled.

chrisale, 2023-09-04:
Ah. I see now. That option isn't in my weewx.conf.
I've added it in, set it to true, and it caught the exception when I restarted. 
It waited 60 seconds and tried again successfully.
Looks like problem solved.
May I suggest the ability to set the wait period? I suspect in my instance (it might be different for others) a waiting time of 10 seconds would be sufficient.

tkeffer, 2023-09-04:
Good idea. I've introduced option `retry_wait` --- how long to wait before a retry.

Commit https://github.com/weewx/weewx/commit/a4ec368a19b1ce6b3f905571b29f9f9857b4d03b

To appear in V5

closed 2023-09-04 by tkeffer