weewx/weewx #890: v5.x - weewx.cheetahgenerator SeasonsReport runs almost 10x longer than in v4.x

AhrBee, 2023-09-19:
Wondering if my expectation of performance to be nearly the same between versions is reasonable, or am I simply doing something ignorant. In summary: the `weewx.cheetahgenerator` step for the `Seasons` report is now taking almost 10x longer to generate on the same hardware and environment: about 1.5 seconds in v4 and about 12 seconds in v5. Nothing else running on the host. 

While this seems like nit-picking in the scheme of things, and not what I would term a 'problem', yet I see the significantly longer runtime and am curious as to why so?

The only changes to the `Seasons` report from the as-delivered versions are renamed labels, `y_label_side=both`, and a radiation calculation. I am using the `user.gw1000` driver for data collection, and publishing to `mqtt` in both versions.  No modifications to as-delivered skins or other configuration. 

For testing, I did a clean installation of Debian 12 and installed fresh each version of `weewx` and the driver, then copied over the archive file `weewx.sdb` (`wwewx` stopped on the source before copy), then did a `drop-daily` and `rebuild-daily` action to database. `pragma integrity_check` came back OK. 

Link to weewx debug file at the end of this. **NOTE:** I did upgrade to 5.0.0b11, same results (longer `weewx.cheetahgenerator`  runtime)

Environment for each version:
```
HP t630 Thin Client AMD Embedded G-Series GX-420GI Radeon R7E current scheduling governor: schedutil
8GiB System Memory SODIMM DDR4 Synchronous Unbuffered (Unregistered) 1866 MHz (0.5 ns)
Disk - 128GB TS128GMTS430S 

Linux 6.1.0-12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.52-1 (2023-09-07) x86_64 GNU/Linux

Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux

pip install in venv (in version 5.x)

Station type GW1000 (user.gw1000) GatewayDriver: version is 0.5.0b5
record_generation = software

archive interval is 60 seconds, over 600k records
```

<hr>

**weewx version 5.0.0b10:**
```
INFO weewx.cheetahgenerator: Generated 8 files for report SeasonsReport in 11.64 seconds
INFO weewx.imagegenerator: Generated 17 images for report SeasonsReport in 0.71 seconds

# as root: pip list
Package            Version
------------------ ---------
certifi            2022.9.24
chardet            5.1.0
charset-normalizer 3.0.1
ephem              4.1.4
httplib2           0.20.4
idna               3.3
pip                23.0.1
pycurl             7.45.2
pyparsing          3.0.9
PySimpleSOAP       1.16.2
python-apt         2.6.0
python-debian      0.1.49
python-debianbts   4.0.1
reportbug          12.0.0
requests           2.28.1
setuptools         66.1.1
six                1.16.0
urllib3            1.26.12
wheel              0.38.4

# as weewx (VENV): pip list
Package    Version
---------- --------
configobj  5.0.8
CT3        3.3.2
ephem      4.1.4
paho-mqtt  1.6.1
Pillow     10.0.0
pip        23.0.1
PyMySQL    1.1.0
pyserial   3.5
pyusb      1.2.1
setuptools 66.1.1
six        1.16.0
ssh-audit  3.0.0
weewx      5.0.0b11

```

**weewx version 4.10.2**
```
INFO weewx.cheetahgenerator: Generated 8 files for report SeasonsReport in 1.34 seconds
INFO weewx.imagegenerator: Generated 17 images for report SeasonsReport in 0.66 seconds

Package            Version
------------------ -----------
certifi            2022.9.24
chardet            5.1.0
charset-normalizer 3.0.1
configobj          5.0.8
CT3                3.3.1
ephem              4.1.4
httplib2           0.20.4
idna               3.3
mysqlclient        1.4.6
olefile            0.46
paho-mqtt          1.6.1
Pillow             9.4.0
pip                23.0.1
pycurl             7.45.2
pyparsing          3.0.9
pyserial           3.5
PySimpleSOAP       1.16.2
python-apt         2.6.0
python-debian      0.1.49
python-debianbts   4.0.1
pyusb              1.2.1.post2
reportbug          12.0.0
requests           2.28.1
setuptools         66.1.1
six                1.16.0
urllib3            1.26.12
wheel              0.38.4
```

[weewx_debug_20230919.txt](https://github.com/weewx/weewx/files/12665915/weewx_debug_20230919.txt)

tkeffer, 2023-09-19:
Thanks for the report!

I tried it on my machine and got similar answers for v4.10.2 and v5.0.0b11.

|Version | Files | time |
|--------|-------|------|
|4.10.2  |212    |9.25s |
|5.0.0b11|212    |9.42s |

This is on a reasonably fast NUC with a SSD.

Some questions:
- Which version of Python? I tried both v3.7 and v3.11 and got similar results.
- Does the machine have a Solid-State Disk?
- Did you use `wee_reports` or `weewxd`?

AhrBee, 2023-09-20:
- Python v.3.11.2 in both cases
- Solid-State Disk - yes of recent vintage, hardly used. 
- I let `weewxd` do the work

I installed 5.0.0b11 on a laptop with SSD (Intel(R) Core(TM) i5-3210M CPU @ 2.50GHz current scheduling governor: schedutil), Python 3.11.2, same kernel as the other tests, and got the same LONGER run results.

Must be me doing something in the installation process (pre-requisites and weewx/driver installs) because I built the Debian 12 OS exactly the same from the same media on freshly-wiped disk (ext4 filesystem, everything in one partition) following the exact same steps in every test case. 

I do have significantly fewer files that are generated - eight (8). 

Oh well, another one for the *Journal of Irreproducible Results* - no sense trying to troubleshoot something that can't be reproduced elsewhere.

tkeffer, 2023-09-20:
Could you try it using `wee_reports`? It's more of a "pure" test of report generation.

The difference in the number of files is probably because I ran `wee_reports` against a large database with a lot of years and records: 16 years, 1.7M records. That would result in a lot of NOAA files. 

I tried it against a much smaller database and got similar results:

|Version  | Files | time |
|---------|-------|------|
|v4.10.2  | 8     | .31s |
|v5.0.0b11| 8     | .35s |

AhrBee, 2023-09-20:
Using the command `time wee_reports` under the different `weewx` versions using the same archive file:

Version | Time (s) |
---| --- |
v4.10.2 | 1.8 |
v5.0.0b11 | 13.9 |

tkeffer, 2023-09-20:
Not so fast. For some reason, now I'm able to reproduce your results:

|Version  |Files | time |
|---------|------|------|
|v4.10.2  | 212  | 1.41s|
|v5.0.0b11| 212  | 9.48s|

I'm not sure what I'm doing differently. 

I will note that using the C version of `NameMapper` can make a huge difference. I believe I'm using it for both test runs.

tkeffer, 2023-09-20:
Found the problem. To see whether there is any data in a timespan, the Cheetah generator was always consulting the xtypes system, even when the type was in the database. Because xtypes calls are relatively slow, this killed performance. 

Now I get

|Version  | Files | time |
|---------|-------|------|
|v4.10.2  |212    |1.41s |
|v5.0.0b12|212    |1.53s |

I'll upload v5.0.0b12 later today and you can give it a try.

closed 2023-09-20 by tkeffer

AhrBee, 2023-09-20:
Performance of the new version v5.0.0b12 now matches very closely with v4.10.2 - THANK YOU for the fix and so quickly!