weewx/weewx #892: Upgrade Google Analytics handling to use GA4 rather than deprecated UA tags

jonandel, 2023-09-28:
A year or so ago, GOogle decided to deprecate the old Universal Analytics tages and implement a new GA4 tagging system.

This changes the script in the header for these new tages from the old UA one.

Standard Weewx functionality inserts the header if uncommeting the GoogleAnalyticsID tag in the skins - but appears to be based upon the old UA tagging.

For the new GA4 ta the header needs to be this:
<html>
<body>

<head>
<script async src="https://www.googletagmanager.com/gtag/js?id=G-ABCDEFGHI"></script>
  | <script>
  | window.dataLayer = window.dataLayer \|\| [];
  | function gtag(){dataLayer.push(arguments);}
  | gtag('js', new Date());
  |  
  | gtag('config', 'G-ABCDEFGHI');
  | </script>

</body>
</html>

Where the GA tag is "G-ABCDEFGHI" in above..

tkeffer, 2023-09-28:
Thanks for pointing this out!

Following your suggestion, the file `analytics.inc` would look something like this?

```

## If a Google Analytics GA4 code has been specified, include it.

#if 'googleAnalyticsId' in $Extras
    <script async src="https://www.googletagmanager.com/gtag/js?id=$Extras.googleAnalyticsId"></script>
    <script>
      window.dataLayer = window.dataLayer || [];
      function gtag(){dataLayer.push(arguments);}
      gtag('js', new Date());

      gtag('config', '$Extras.googleAnalyticsId');
    </script>
#end if
```

Perhaps you could give it a try?

jonandel, 2023-09-28:
Hi Tom,  Yes - appears to work.

I note Google suggests putting the script in the Header - and Weewx puts in the page body..and at the end.. ( at least for Seasons).....but Im not sure if thats significant ...  I can see the Analytics is registering the hits... I will play with it for a day or so and get bck to you...

tkeffer, 2023-09-28:
Fixed in commit https://github.com/weewx/weewx/commit/2d7d7a1553c5687aac1dd3c4c4b5590e115392a9

I also moved the script to the `<head>` region.

I don't use Google analytics, so I'll be dependent on others to check it out!

closed 2023-09-28 by tkeffer

jonandel, 2023-09-29:
Thanks Tom, the only other thing I did on my local copy was included the analytics.inc in the other page templates...e.g tabular.html.tmpl as using these excludes the analytics code....but I'm still playing with Google at the moment to see the impact ...
I've only done this in Seasons so far.
I also note Google says they're deprecating UA from July this year, so others may still be using ..maybe worth noting in the docs...